openwrt/asu #896: Missing overview endpoint after rewrite to FastAPI

dannil, 2024-07-20:
As observed by a few users after the 23.05.4 release, it seems that the old overview endpoint didn't get migrated, which at least breaks the LuCI app on 23.05.3 and earlier. Looking at http://sysupgrade.openwrt.org/docs, `/api/overview` was replaced with `/api/v1/overview` but there's no redirect from the old one to the new one. I think it's due to c10687b cause I can't find the `/api/overview` endpoint in that commit.

There may be more missing, this is just the one I found quickly.

For reference: 
*  `GET /api/overview`: https://forum.openwrt.org/t/attended-sysupgrade-auc-error/204521 (does this break `auc` as well or does it already use the versioned path?)
    * [Old URL in in the LuCI app](https://github.com/openwrt/luci/blob/b3ea37cf890b8fc7d99581127367decb859707d4/applications/luci-app-attendedsysupgrade/htdocs/luci-static/resources/view/attendedsysupgrade/overview.js#L402)

patrakov, 2024-07-20:
`auc -y` works on 23.05.3.

kirenida, 2024-07-20:
I'm on 23.05.2 - r23630-842932a63d

and I get this error when I try to search for updates:

```
Error connecting to upgrade server

Could not reach API at "https://sysupgrade.openwrt.org/api/overview?1721470470154". Please try again later.

{"detail":"Not Found"}
```

langerma, 2024-07-20:
same here: {"detail":"Not Found"}

popy2k14, 2024-07-20:
Same here: {"detail":"Not Found"}
nanoPi R4S URL: https://sysupgrade.openwrt.org/api/overview?1721472938585

atomGit, 2024-07-20:
OpenWrt 23.05.4, r24012-d8dd03c46f

```
Error connecting to upgrade server

Could not reach API at "https://sysupgrade.openwrt.org/api/overview?1721483070122".
```

`auc -r` and `auc -c` both work

popy2k14, 2024-07-20:
thanks worked with `auc` but had to remove liblua, see here: https://forum.openwrt.org/t/openwrt-23-05-4-service-release/204514/26?u=popywrt

dannil, 2024-07-20:
Some more observations: the API used to be mounted [on /api](https://github.com/openwrt/asu/commit/c10687bd5ac5aef0163af44a9942e76f053fa9d6#diff-f72c97eb2750c245340b39390a4fec2e3394a8ac1ab7aa91f497149182c09928L8) which then [/overview was bound to](https://github.com/openwrt/asu/commit/c10687bd5ac5aef0163af44a9942e76f053fa9d6#diff-f72c97eb2750c245340b39390a4fec2e3394a8ac1ab7aa91f497149182c09928L61), resulting in the effective endpoint of `/api/overview`. With the change that [the API router gets bound to /api/v1 instead](https://github.com/openwrt/asu/commit/c10687bd5ac5aef0163af44a9942e76f053fa9d6#diff-491c5ee0e7d6af1894d859d3dd8a9ab458af02c12a91ccc64ee8a444497ee49aR22), this results in a new effective path of `/api/v1/overview`, so my initial suspicions seems to be correct.

I saw that https://github.com/openwrt/asu/commit/019c0b7ec1b63a32f91e5302bbc53d22f8eb62ee adds the `/overview` endpoint after the rewrite, but if I'm understanding the code correctly this results in a path of https://sysupgrade.openwrt.org/overview (which doesn't seem to be deployed?). It's not impossible that this commit was supposed to add back `/api/overview` but missed the `v1` part but I don't think so since the code comment says "very legacy".

aparcar, 2024-07-20:
Fixed via a redirect in Caddy (the web server), should work again?

C-512L, 2024-07-20:
> Fixed via a redirect in Caddy (the web server), should work again?

Can confirm, both `luci-app-attendedsysupgrade` and https://firmware-selector.openwrt.org/ work fine.

Lanchon, 2024-07-20:
confirmed working. with this fix, downstream projects will have issues, but i guess @dannil can close this now.

udanfacy22, 2024-07-21:
`Luci-app-attendedsysupgrade` appears to be working. Running it in 23.05.4 now gives an expected message about me already being on the latest firmware, instead of the `{"detail":"Not Found"}` error it was giving me yesterday.

`auc` on the other hand, appears to me like it may still be broken (unless I'm mistaken). I haven't used `auc` prior to yesterday, so I'm not too familiar with it. I run it with no arguments, and this is the output I get:

    auc/0.3.2-1
    Server:    https://sysupgrade.openwrt.org
    Running:   23.05.4 r24012-d8dd03c46f on bcm27xx/bcm2711 (raspberrypi,4-model-b)
    Invalid argument (22)

"Invalid argument (22)" looks like an error, and it also isn't giving any kind of message about me already being on the latest release.

I haven't yet tried temporarily downgrading to 25.03.3 and then checking for updates to see what would happen. Would you like me to do that?

popy2k14, 2024-07-21:
Also working here now in luci.
Thx

closed 2024-07-21 by dannil