openwrt/asu #930: update: 'versions' list contains branch name in overview.json

efahl, 2024-08-06:
This appeared today.  Look in https://sysupgrade.openwrt.org/json/v1/overview.json and the branches["*"].versions  lists all contain the branch name.  The https://downloads.openwrt.org/.versions.json contents look fine, so maybe it's coming from the settings via `get_branch`???
```
...
  "branches": {
    "21.02": {
      "path": "releases/{version}",
      "enabled": true,
      "path_packages": "DEPRECATED",
      "name": "21.02",
      "versions": [
        "21.02.7",
        "21.02-SNAPSHOT",
        "21.02"             <<<<<========  Should not be here
      ],
...
```

aparcar, 2024-08-06:
Removed, sorry script issue on my end.

closed 2024-08-06 by aparcar

efahl, 2024-08-06:
👍🏼  I figured it was something like that when the release snapshots appeared at the same time.