openwrt/asu #931: QualcommAX stuck on r27052-4e6212e62f

Soromeister, 2024-08-06:
> HTTP/1.1 500 Internal Server Error
> Access-Control-Allow-Credentials: true
> Access-Control-Allow-Headers: *
> Access-Control-Allow-Methods: POST, GET, OPTIONS
> Access-Control-Allow-Origin: *
> Access-Control-Expose-Headers: Authorization
> Connection: close
> Content-Length: 7306
> Content-Type: application/json
> Date: Tue, 06 Aug 2024 18:19:02 GMT
> Server: Caddy
> Vary: Origin

```
{
    "detail": "Error: Received incorrect version r27052-4e6212e62f (requested r27082-b733b6acb5)",
    "request": {
        "distro": "openwrt",
        "version": "SNAPSHOT",
        "version_code": "r27082-b733b6acb5",
        "target": "qualcommax/ipq807x",
        "packages": [
            "ath10k-firmware-qca9887",
            "ath11k-firmware-ipq8074",
            "ath11k-firmware-qcn9074",
            "base-files",
            "busybox",
            "ca-bundle",
            "coreutils",
            "curl",
            "dnsmasq",
            "dropbear",
            "e2fsprogs",
            "firewall4",
            "gre",
            "htop",
            "ipq-wifi-xiaomi_ax9000",
            "iw",
            "iwinfo",
            "jq",
            "kernel",
            "kmod-ath10k-ct",
            "kmod-ath11k-ahb",
            "kmod-ath11k-pci",
            "kmod-fs-ext4",
            "kmod-gpio-button-hotplug",
            "kmod-ieee802154-6lowpan",
            "kmod-leds-gpio",
            "kmod-mac802154",
            "kmod-mppe",
            "kmod-phy-aquantia",
            "kmod-qca-nss-dp",
            "kmod-sit",
            "kmod-udptunnel6",
            "kmod-usb-dwc3",
            "kmod-usb-dwc3-qcom",
            "kmod-usb3",
            "libc",
            "libustream-wolfssl",
            "libwolfssl",
            "logd",
            "losetup",
            "luci",
            "luci-app-attendedsysupgrade",
            "luci-app-irqbalance",
            "mtd",
            "ncat-full",
            "odhcp6c",
            "odhcpd-ipv6only",
            "ppp",
            "ppp-mod-pppoe",
            "procd",
            "procd-seccomp",
            "procd-ujail",
            "uboot-envtools",
            "uci",
            "urandom-seed",
            "urngd",
            "wget-ssl",
            "wpad-wolfssl"
        ],
        "profile": "xiaomi_ax9000",
        "packages_versions": {},
        "defaults": null,
        "client": null,
        "rootfs_size_mb": null,
        "diff_packages": true,
        "repositories": {},
        "repository_keys": []
    },
    "stdout": "Current Target: "qualcommax/ipq807x"\nCurrent Architecture: "aarch64"\nCurrent Revision: "r27052-4e6212e62f"\nDefault Packages: base-files ca-bundle dropbear fstools libc libgcc libustream-mbedtls logd mtd netifd uci uclient-fetch urandom-seed urngd kmod-usb3 kmod-usb-dwc3 kmod-usb-dwc3-qcom kmod-leds-gpio kmod-gpio-button-hotplug kmod-qca-nss-dp kmod-ath11k-ahb wpad-basic-mbedtls uboot-envtools e2fsprogs kmod-fs-ext4 losetup kmod-phy-aquantia ath11k-firmware-ipq8074 opkg busybox procd procd-ujail procd-seccomp dnsmasq firewall4 nftables kmod-nft-offload odhcp6c odhcpd-ipv6only ppp ppp-mod-pppoe\nAvailable Profiles:\n\narcadyan_aw1000:\n Arcadyan AW1000\n Packages: ipq-wifi-arcadyan_aw1000 kmod-spi-gpio kmod-gpio-nxp-74hc164 kmod-usb-serial-option uqmi\n hasImageMetadata: 1\n SupportedDevices: arcadyan,aw1000\nasus_rt-ax89x:\n Asus RT-AX89X\n Packages: kmod-hwmon-gpiofan ipq-wifi-asus_rt-ax89x\n hasImageMetadata: 1\n SupportedDevices: asus,rt-ax89x\nbuffalo_wxr-5950ax12:\n Buffalo WXR-5950AX12\n Packages: ipq-wifi-buffalo_wxr-5950ax12\n hasImageMetadata: 1\n SupportedDevices: buffalo,wxr-5950ax12\ncmcc_rm2-6:\n CMCC RM2-6\n Packages: ipq-wifi-cmcc_rm2-6 kmod-hwmon-gpiofan\n hasImageMetadata: 1\n SupportedDevices: cmcc,rm2-6\ncompex_wpq873:\n Compex WPQ873\n Packages: ipq-wifi-compex_wpq873\n hasImageMetadata: 1\n SupportedDevices: compex,wpq873\ndynalink_dl-wrx36:\n Dynalink DL-WRX36\n Packages: ipq-wifi-dynalink_dl-wrx36\n hasImageMetadata: 1\n SupportedDevices: dynalink,dl-wrx36\nedgecore_eap102:\n Edgecore EAP102\n Packages: ipq-wifi-edgecore_eap102\n hasImageMetadata: 1\n SupportedDevices: edgecore,eap102\nedimax_cax1800:\n Edimax CAX1800\n Packages: ipq-wifi-edimax_cax1800\n hasImageMetadata: 1\n SupportedDevices: edimax,cax1800\nlinksys_mx4200v1:\n Linksys MX4200 v1\n Packages: kmod-leds-pca963x ipq-wifi-linksys_mx4200 kmod-bluetooth\n hasImageMetadata: 1\n SupportedDevices: linksys,mx4200v1\nlinksys_mx4200v2:\n Linksys MX4200 v2\n Packages: kmod-leds-pca963x ipq-wifi-linksys_mx4200 kmod-bluetooth\n hasImageMetadata: 1\n SupportedDevices: linksys,mx4200v2\nlinksys_mx5300:\n Linksys MX5300\n Packages: kmod-leds-pca963x kmod-rtc-ds1307 ipq-wifi-linksys_mx5300 kmod-ath10k-ct ath10k-firmware-qca9984-ct\n hasImageMetadata: 1\n SupportedDevices: linksys,mx5300\nlinksys_mx8500:\n Linksys MX8500\n Packages: kmod-leds-pca963x ipq-wifi-linksys_mx8500 kmod-ath11k-pci ath11k-firmware-qcn9074 kmod-bluetooth\n hasImageMetadata: 1\n SupportedDevices: linksys,mx8500\nnetgear_rax120v2:\n Netgear RAX120v2\n Packages: ipq-wifi-netgear_rax120v2 kmod-spi-gpio kmod-spi-bitbang kmod-gpio-nxp-74hc164 kmod-hwmon-g762\n hasImageMetadata: 1\n SupportedDevices: netgear,rax120v2\nnetgear_sxr80:\n Netgear SXR80\n Packages: ipq-wifi-netgear_sxk80\n hasImageMetadata: 1\n SupportedDevices: netgear,sxr80\nnetgear_sxs80:\n Netgear SXS80\n Packages: ipq-wifi-netgear_sxk80\n hasImageMetadata: 1\n SupportedDevices: netgear,sxs80\nnetgear_wax218:\n Netgear WAX218\n Packages: kmod-spi-gpio kmod-spi-bitbang kmod-gpio-nxp-74hc164 ipq-wifi-netgear_wax218\n hasImageMetadata: 1\n SupportedDevices: netgear,wax218\nnetgear_wax620:\n Netgear WAX620\n Packages: kmod-spi-gpio kmod-gpio-nxp-74hc164 ipq-wifi-netgear_wax620\n hasImageMetadata: 1\n SupportedDevices: netgear,wax620\nnetgear_wax630:\n Netgear WAX630\n Packages: kmod-spi-gpio ipq-wifi-netgear_wax630\n hasImageMetadata: 1\n SupportedDevices: netgear,wax630\nprpl_haze:\n prpl Foundation Haze\n Packages: ath11k-firmware-qcn9074 ipq-wifi-prpl_haze kmod-ath11k-pci mkf2fs f2fsck kmod-fs-f2fs kmod-leds-lp5562\n hasImageMetadata: 1\n SupportedDevices: prpl,haze\nqnap_301w:\n QNAP 301w\n Packages: ipq-wifi-qnap_301w\n hasImageMetadata: 1\n SupportedDevices: qnap,301w\nredmi_ax6:\n Redmi AX6\n Packages: ipq-wifi-redmi_ax6\n hasImageMetadata: 1\n SupportedDevices: redmi,ax6\nspectrum_sax1v1k:\n Spectrum SAX1V1K\n Packages: ipq-wifi-spectrum_sax1v1k\n hasImageMetadata: 1\n SupportedDevices: spectrum,sax1v1k\nxiaomi_ax3600:\n Xiaomi AX3600\n Packages: ipq-wifi-xiaomi_ax3600 kmod-ath10k-ct-smallbuffers ath10k-firmware-qca9887-ct\n hasImageMetadata: 1\n SupportedDevices: xiaomi,ax3600\nxiaomi_ax9000:\n Xiaomi AX9000\n Packages: ipq-wifi-xiaomi_ax9000 kmod-ath11k-pci ath11k-firmware-qcn9074 kmod-ath10k-ct ath10k-firmware-qca9887-ct\n hasImageMetadata: 1\n SupportedDevices: xiaomi,ax9000\nyuncore_ax880:\n Yuncore AX880\n Packages: ipq-wifi-yuncore_ax880\n hasImageMetadata: 1\n SupportedDevices: yuncore,ax880\nzbtlink_zbt-z800ax:\n Zbtlink ZBT-Z800AX\n Packages: ipq-wifi-zbtlink_zbt-z800ax\n hasImageMetadata: 1\n SupportedDevices: zbtlink,zbt-z800ax\nzte_mf269:\n ZTE MF269\n Packages: ipq-wifi-zte_mf269\n hasImageMetadata: 1\n SupportedDevices: zte,mf269\nzyxel_nbg7815:\n ZYXEL NBG7815\n Packages: ipq-wifi-zyxel_nbg7815 kmod-ath11k-pci kmod-bluetooth kmod-hwmon-tmp103\n hasImageMetadata: 1\n SupportedDevices: zyxel,nbg7815\n",
    "stderr": "",
    "status": 500,
    "error": "Traceback (most recent call last):\n File "/usr/local/lib/python3.12/site-packages/rq/worker.py", line 1430, in perform_job\n rv = job.perform()\n ^^^^^^^^^^^^^\n File "/usr/local/lib/python3.12/site-packages/rq/job.py", line 1280, in perform\n self._result = self._execute()\n ^^^^^^^^^^^^^^^\n File "/usr/local/lib/python3.12/site-packages/rq/job.py", line 1317, in _execute\n result = self.func(*self.args, **self.kwargs)\n ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^\n File "/app/asu/build.py", line 76, in build\n report_error(\n File "/app/asu/util.py", line 324, in report_error\n raise\nRuntimeError: No active exception to reraise\n",
    "enqueued_at": "2024-08-06T18:18:52.998575",
    "request_hash": "32588d99001a79e83a0db37c2756de0a"
}
```

aparcar, 2024-08-07:
I kicked the docker container creator again, please give it an hour and check again

Soromeister, 2024-08-07:
Thank you @aparcar, this works now.

closed 2024-08-07 by Soromeister