openwrt/asu #934: asu failed to download kmod-mii on NanoPi R4S

grzesiczek1, 2024-08-08:
"url": "https://sysupgrade.openwrt.org",
    "revision": "r23809-234f1a2efa",
    "advanced_mode": "0",
    "sha256_unsigned": "",
    "branch": "23.05",
    "efi": null,
    "profile": "friendlyarm,nanopi-r4s",
    "target": "rockchip/armv8",
    "version": "23.05.4",

STDERR:
Generate local signing keys...
Generate local certificate...
Package list missing or not up-to-date, generating it.

Building package index...
Downloading https://downloads.openwrt.org/releases/23.05.4/targets/rockchip/armv8/packages/Packages.gz
Updated list of available packages in /builder/build_dir/target-aarch64_generic_musl/root-rockchip/../../../../builder/dl/openwrt_core
Downloading https://downloads.openwrt.org/releases/23.05.4/targets/rockchip/armv8/packages/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/23.05.4/packages/aarch64_generic/base/Packages.gz
Updated list of available packages in /builder/build_dir/target-aarch64_generic_musl/root-rockchip/../../../../builder/dl/openwrt_base
Downloading https://downloads.openwrt.org/releases/23.05.4/packages/aarch64_generic/base/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/23.05.4/packages/aarch64_generic/luci/Packages.gz
Updated list of available packages in /builder/build_dir/target-aarch64_generic_musl/root-rockchip/../../../../builder/dl/openwrt_luci
Downloading https://downloads.openwrt.org/releases/23.05.4/packages/aarch64_generic/luci/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/23.05.4/packages/aarch64_generic/packages/Packages.gz
Updated list of available packages in /builder/build_dir/target-aarch64_generic_musl/root-rockchip/../../../../builder/dl/openwrt_packages
Downloading https://downloads.openwrt.org/releases/23.05.4/packages/aarch64_generic/packages/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/23.05.4/packages/aarch64_generic/routing/Packages.gz
Updated list of available packages in /builder/build_dir/target-aarch64_generic_musl/root-rockchip/../../../../builder/dl/openwrt_routing
Downloading https://downloads.openwrt.org/releases/23.05.4/packages/aarch64_generic/routing/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/23.05.4/packages/aarch64_generic/telephony/Packages.gz
Updated list of available packages in /builder/build_dir/target-aarch64_generic_musl/root-rockchip/../../../../builder/dl/openwrt_telephony
Downloading https://downloads.openwrt.org/releases/23.05.4/packages/aarch64_generic/telephony/Packages.sig
Signature check passed.
Downloading file:packages/Packages
Updated list of available packages in /builder/build_dir/target-aarch64_generic_musl/root-rockchip/../../../../builder/dl/imagebuilder
Downloading file:packages/Packages.sig
Signature check passed.
Collected errors:
 * opkg_download: Failed to download https://downloads.openwrt.org/releases/23.05.4/targets/rockchip/armv8/packages/kmod-mii_5.15.162-1_aarch64_generic.ipk, wget returned 4.
 * opkg_download: Check your network settings and connectivity.

 * opkg_install_pkg: Failed to download kmod-mii. Perhaps you need to run 'opkg update'?
 * opkg_install_cmd: Cannot install package kmod-r8169.
make[2]: *** [Makefile:189: package_install] Error 255
make[1]: *** [Makefile:144: _call_image] Error 2
make: *** [Makefile:262: image] Error 2

grzesiczek1, 2024-08-08:
Opening link directly downloads right package 
https://downloads.openwrt.org/releases/23.05.4/targets/rockchip/armv8/packages/kmod-mii_5.15.162-1_aarch64_generic.ipk

erik434, 2024-08-11:
Did you ever get this resolved? I'm seeing something similar and wondering if there's something wrong with my requests or with the asu server.

grzesiczek1, 2024-08-11:
I did not get it resolved via UI but it worked via CLI. Since I upgraded my device via command line I have no environment to prove UI is working correctly.

erik434, 2024-08-11:
Thanks for the tip! Just running "auc" on the command line did the job perfectly. I wonder what's going on with the luci interface?? Very strange.