openwrt/asu #935: Pod not starting correctly.

pcartwright81, 2024-08-09:
Currently using Fedora CoreOS, but I have tried a few other distros without success.  Also root & rootless does not seem to matter.
Network connectivity seems to be working with other pods/containers.

The socket is activated using these instructions.
https://github.com/containers/podman/blob/main/docs/tutorials/socket_activation.md

Even after 30 minutes the below logs are all I see:
```
core@fedoracoreos:~/asu$ curl localhost:8000
curl: (56) Recv failure: Connection reset by peer

core@fedoracoreos:~/asu$ ls $XDG_RUNTIME_DIR/podman/podman.sock
/run/user/1000/podman/podman.sock

core@fedoracoreos:~/asu$ cat .env
PUBLIC_PATH=/var/home/core/asu/public
CONTAINER_SOCK=/run/user/1000/podman/podman.sock

core@fedoracoreos:~/asu$ podman ps
CONTAINER ID  IMAGE                                      COMMAND               CREATED         STATUS         PORTS                     NAMES
59b850bc3a85  docker.io/redis/redis-stack-server:latest  /entrypoint.sh        18 minutes ago  Up 18 minutes  6379/tcp                  asu_redis_1
b5789dce8fa3  localhost/asu:latest                       /bin/sh -c uvicor...  18 minutes ago  Up 18 minutes  127.0.0.1:8000->8000/tcp  asu_server_1
dfc25164bf40  localhost/asu:latest                       rqworker --with-s...  18 minutes ago  Up 18 minutes                            asu_worker_1

core@fedoracoreos:~/asu$ podman logs -t asu_server_1
024-08-09T01:44:31.608473000Z INFO:     Started server process [2]
2024-08-09T01:44:31.611308000Z INFO:     Waiting for application startup.
2024-08-09T01:44:31.611654000Z INFO:root:ASU server starting up
2024-08-09T01:44:31.611716000Z INFO:     Application startup complete.
2024-08-09T01:44:31.632659000Z INFO:     Uvicorn running on http://127.0.0.1:8000 (Press CTRL+C to quit)

core@fedoracoreos:~/asu$ podman logs -t  asu_worker_1
2024-08-09T01:44:30.281151000Z 01:44:30 Worker rq:worker:7a7c33530d7048fd95a00b1e2f480515 started with PID 1, version 1.16.2
2024-08-09T01:44:30.284328000Z 01:44:30 Subscribing to channel rq:pubsub:7a7c33530d7048fd95a00b1e2f480515
2024-08-09T01:44:30.284430000Z 01:44:30 *** Listening on default...
2024-08-09T01:44:30.290837000Z 01:44:30 Scheduler for default started with PID 3
2024-08-09T01:44:30.291664000Z 01:44:30 Cleaning registries for queue: default
2024-08-09T01:58:00.331967000Z 01:58:00 Cleaning registries for queue: default

core@fedoracoreos:~/asu$ podman network ls
NETWORK ID    NAME         DRIVER
083d3a9b8790  asu_default  bridge
2f259bab93aa  podman       bridge
```
Redis log
[redis.log](https://github.com/user-attachments/files/16554985/redis.log)

Network inspect
[083d3a9b8790.json](https://github.com/user-attachments/files/16554990/083d3a9b8790.json)

asu_server_1
[asu_server_1.json](https://github.com/user-attachments/files/16555000/asu_server_1.json)

asu_worker_1
[asu_worker_1.json](https://github.com/user-attachments/files/16555002/asu_worker_1.json)

aparcar, 2024-08-09:
Sorry for the trouble, fixed via 810c60b4351b

pcartwright81, 2024-08-09:
Works great now.  Thanks!

Had to run the below commands to make it functional:
podman pod stop pod_asu
podman pod rm pod_asu
podman system prune -a

closed 2024-08-09 by pcartwright81