openwrt/asu #938: Error: Impossible package selection -> libustream-ssl

fow0ryl, 2024-08-10:
Hello,
this is the second time I'm trying attended sysupgrade. But it fails on my NanoPi R2s and my Archer C7 V5 with the same error ...

{
    "url": "https://sysupgrade.openwrt.org",
    "revision": "r20028-43d71ad93e",
    "advanced_mode": "1",
    "branch": "22.03",
    "efi": null,
    "profile": "friendlyarm,nanopi-r2s",
    "target": "rockchip/armv8",
    "version": "23.05.4",
    "packages": [
        "base-files",
        "block-mount",
        "busybox",
        "ca-bundle",
        "cgi-io",
        "collectd-mod-thermal",
        "comgt",
        "dmesg",
        "dnsmasq",
        "dropbear",
        "dropbearconvert",
        "e2fsprogs",
        "fdisk",
        "firewall",
        "fstools",
        "fwtool",
        "getrandom",
        "gpioctl-sysfs",
        "hub-ctrl",
        "iconv",
        "ip6tables-nft",
        "iperf3",
        "iptables-nft",
        "jshn",
        "jsonfilter",
        "kernel",
        "kmod-fs-exfat",
        "kmod-fs-ext4",
        "kmod-fs-vfat",
        "kmod-gpio-button-hotplug",
        "kmod-ip6tables",
        "kmod-ipt-conntrack",
        "kmod-ipt-core",
        "kmod-ipt-nat",
        "kmod-ipt-offload",
        "kmod-lib-crc-ccitt",
        "kmod-lp",
        "kmod-nf-conntrack",
        "kmod-nf-conntrack6",
        "kmod-nf-flow",
        "kmod-nf-ipt",
        "kmod-nf-ipt6",
        "kmod-nf-nat",
        "kmod-nf-reject",
        "kmod-nf-reject6",
        "kmod-ppp",
        "kmod-pppoe",
        "kmod-pppox",
        "kmod-slhc",
        "kmod-usb-net-cdc-ether",
        "kmod-usb-net-rtl8152",
        "kmod-usb-printer",
        "kmod-usb-serial-option",
        "kmod-usb-storage",
        "kmod-usb2",
        "kmod-usb3",
        "libblkid",
        "libblobmsg-json",
        "libc",
        "libcomerr",
        "libext2fs",
        "libf2fs",
        "libip4tc",
        "libip6tc",
        "libiwinfo",
        "libiwinfo-data",
        "libiwinfo-lua",
        "libjson-c",
        "libjson-script",
        "liblua",
        "liblucihttp",
        "liblucihttp-lua",
        "libnl-tiny",
        "libsmartcols",
        "libss",
        "libubox",
        "libubus",
        "libubus-lua",
        "libuci",
        "libuclient",
        "libustream-wolfssl",
        "libuuid",
        "libwolfssl",
        "libxtables",
        "logd",
        "lua",
        "luci",
        "luci-app-adblock",
        "luci-app-attendedsysupgrade",
        "luci-app-commands",
        "luci-app-ddns",
        "luci-app-firewall",
        "luci-app-opkg",
        "luci-app-p910nd",
        "luci-app-squid",
        "luci-app-statistics",
        "luci-app-uhttpd",
        "luci-app-wireguard",
        "luci-app-wol",
        "luci-base",
        "luci-lib-base",
        "luci-lib-ip",
        "luci-lib-jsonc",
        "luci-lib-nixio",
        "luci-mod-admin-full",
        "luci-mod-network",
        "luci-mod-status",
        "luci-mod-system",
        "luci-proto-ipv6",
        "luci-proto-ppp",
        "luci-ssl",
        "luci-theme-bootstrap",
        "mkf2fs",
        "mosquitto-client-nossl",
        "mosquitto-ssl",
        "msmtp",
        "mtd",
        "mtr-json",
        "nano",
        "netifd",
        "odhcp6c",
        "odhcpd-ipv6only",
        "openssh-sftp-server",
        "openwrt-keyring",
        "opkg",
        "partx-utils",
        "patch",
        "picocom",
        "ppp",
        "ppp-mod-pppoe",
        "procd",
        "px5g-wolfssl",
        "qrencode",
        "rpcd",
        "rpcd-mod-file",
        "rpcd-mod-iwinfo",
        "rpcd-mod-luci",
        "rpcd-mod-rrdns",
        "smstools3",
        "socat",
        "squid-mod-cachemgr",
        "sshfs",
        "tcpdump",
        "uboot-envtools",
        "ubox",
        "ubus",
        "ubusd",
        "uci",
        "uclient-fetch",
        "uhttpd",
        "uhttpd-mod-ubus",
        "urandom-seed",
        "urngd",
        "usb-modeswitch",
        "usbutils",
        "usign",
        "zstd"
    ],
    "diff_packages": true,
    "filesystem": "ext4",
    "client": "luci/git-22.210.54153-69231a7"
}

STDERR:

Generate local signing keys...
Generate local certificate...
Package list missing or not up-to-date, generating it.

Building package index...
Downloading https://downloads.openwrt.org/releases/23.05.4/targets/rockchip/armv8/packages/Packages.gz
Updated list of available packages in /builder/build_dir/target-aarch64_generic_musl/root-rockchip/../../../../builder/dl/openwrt_core
Downloading https://downloads.openwrt.org/releases/23.05.4/targets/rockchip/armv8/packages/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/23.05.4/packages/aarch64_generic/base/Packages.gz
Updated list of available packages in /builder/build_dir/target-aarch64_generic_musl/root-rockchip/../../../../builder/dl/openwrt_base
Downloading https://downloads.openwrt.org/releases/23.05.4/packages/aarch64_generic/base/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/23.05.4/packages/aarch64_generic/luci/Packages.gz
Updated list of available packages in /builder/build_dir/target-aarch64_generic_musl/root-rockchip/../../../../builder/dl/openwrt_luci
Downloading https://downloads.openwrt.org/releases/23.05.4/packages/aarch64_generic/luci/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/23.05.4/packages/aarch64_generic/packages/Packages.gz
Updated list of available packages in /builder/build_dir/target-aarch64_generic_musl/root-rockchip/../../../../builder/dl/openwrt_packages
Downloading https://downloads.openwrt.org/releases/23.05.4/packages/aarch64_generic/packages/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/23.05.4/packages/aarch64_generic/routing/Packages.gz
Updated list of available packages in /builder/build_dir/target-aarch64_generic_musl/root-rockchip/../../../../builder/dl/openwrt_routing
Downloading https://downloads.openwrt.org/releases/23.05.4/packages/aarch64_generic/routing/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/23.05.4/packages/aarch64_generic/telephony/Packages.gz
Updated list of available packages in /builder/build_dir/target-aarch64_generic_musl/root-rockchip/../../../../builder/dl/openwrt_telephony
Downloading https://downloads.openwrt.org/releases/23.05.4/packages/aarch64_generic/telephony/Packages.sig
Signature check passed.
Downloading file:packages/Packages
Updated list of available packages in /builder/build_dir/target-aarch64_generic_musl/root-rockchip/../../../../builder/dl/imagebuilder
Downloading file:packages/Packages.sig
Signature check passed.
Collected errors:
 * check_data_file_clashes: Package libustream-mbedtls20201210 wants to install file /builder/build_dir/target-aarch64_generic_musl/root-rockchip/lib/libustream-ssl.so
	But that file is already provided by package  * libustream-wolfssl20201210
 * opkg_install_cmd: Cannot install package luci-ssl.
make[2]: *** [Makefile:189: package_install] Error 255
make[1]: *** [Makefile:154: _call_manifest] Error 2
make: *** [Makefile:274: manifest] Error 2

akostadinov, 2024-09-02:
I hit the same issue. Here's my full output
```
{
    "url": "https://sysupgrade.openwrt.org",
    "revision": "r19803-9a599fee93",
    "advanced_mode": "1",
    "branch": "22.03",
    "efi": null,
    "request_hash": "989ba32e0d7abdb1c88af4723cae0a51",
    "profile": "tplink,c2600",
    "target": "ipq806x/generic",
    "version": "23.05.4",
    "packages": [
        "ath10k-firmware-qca99x0-ct",
        "base-files",
        "block-mount",
        "busybox",
        "ca-bundle",
        "cgi-io",
        "dnsmasq",
        "dropbear",
        "firewall4",
        "fstools",
        "fwtool",
        "getrandom",
        "hostapd-common",
        "iw",
        "iwinfo",
        "jansson",
        "jshn",
        "jsonfilter",
        "kernel",
        "kmod-ata-ahci",
        "kmod-ata-ahci-platform",
        "kmod-ata-core",
        "kmod-ath",
        "kmod-ath10k-ct",
        "kmod-cfg80211",
        "kmod-crypto-aead",
        "kmod-crypto-ccm",
        "kmod-crypto-cmac",
        "kmod-crypto-crc32c",
        "kmod-crypto-ctr",
        "kmod-crypto-gcm",
        "kmod-crypto-gf128",
        "kmod-crypto-ghash",
        "kmod-crypto-hash",
        "kmod-crypto-hmac",
        "kmod-crypto-manager",
        "kmod-crypto-null",
        "kmod-crypto-rng",
        "kmod-crypto-seqiv",
        "kmod-crypto-sha256",
        "kmod-fs-vfat",
        "kmod-gpio-button-hotplug",
        "kmod-hwmon-core",
        "kmod-leds-gpio",
        "kmod-lib-crc-ccitt",
        "kmod-lib-crc32c",
        "kmod-mac80211",
        "kmod-nf-conntrack",
        "kmod-nf-conntrack6",
        "kmod-nf-flow",
        "kmod-nf-log",
        "kmod-nf-log6",
        "kmod-nf-nat",
        "kmod-nf-reject",
        "kmod-nf-reject6",
        "kmod-nfnetlink",
        "kmod-nft-core",
[... 126 lines cut ...]
Signature check passed.
Downloading https://downloads.openwrt.org/releases/23.05.4/packages/arm_cortex-a15_neon-vfpv4/packages/Packages.gz
Updated list of available packages in /builder/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/root-ipq806x/../../../../builder/dl/openwrt_packages
Downloading https://downloads.openwrt.org/releases/23.05.4/packages/arm_cortex-a15_neon-vfpv4/packages/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/23.05.4/packages/arm_cortex-a15_neon-vfpv4/routing/Packages.gz
Updated list of available packages in /builder/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/root-ipq806x/../../../../builder/dl/openwrt_routing
Downloading https://downloads.openwrt.org/releases/23.05.4/packages/arm_cortex-a15_neon-vfpv4/routing/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/23.05.4/packages/arm_cortex-a15_neon-vfpv4/telephony/Packages.gz
Updated list of available packages in /builder/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/root-ipq806x/../../../../builder/dl/openwrt_telephony
Downloading https://downloads.openwrt.org/releases/23.05.4/packages/arm_cortex-a15_neon-vfpv4/telephony/Packages.sig
Signature check passed.
Downloading file:packages/Packages
Updated list of available packages in /builder/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/root-ipq806x/../../../../builder/dl/imagebuilder
Downloading file:packages/Packages.sig
Signature check passed.
Collected errors:
 * check_data_file_clashes: Package libustream-mbedtls20201210 wants to install file /builder/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/root-ipq806x/lib/libustream-ssl.so
	But that file is already provided by package  * libustream-wolfssl20201210
 * opkg_install_cmd: Cannot install package luci-ssl.
make[2]: *** [Makefile:189: package_install] Error 255
make[1]: *** [Makefile:154: _call_manifest] Error 2
make: *** [Makefile:274: manifest] Error 2
```

akostadinov, 2024-09-02:
Seems to be previously reported in #647