openwrt/asu #939: opkg_download: Failed to download, wget returned 4

erik434, 2024-08-11:
Here's what seems to be the most relevant part:
```
 * opkg_download: Failed to download https://downloads.openwrt.org/releases/23.05.4/packages/aarch64_cortex-a53/packages/coreutils-sleep_9.3-1_aarch64_cortex-a53.ipk, wget returned 4.
 * opkg_download: Check your network settings and connectivity.

 * opkg_install_pkg: Failed to download coreutils-sleep. Perhaps you need to run 'opkg update'?
 * opkg_install_cmd: Cannot install package coreutils-sleep.
```

I can download the file just fine if I try downloading that URL locally. Is there something wrong with the asu server? Or is there something I can try on my end that would help here? Full details below in case they're helpful.

[Request Data](https://pastebin.com/NRR1nXPt)

[STDOUT](https://pastebin.com/6Q3rgjjq)

[STDERR](https://pastebin.com/bQ8wfgac)

erik434, 2024-08-11:
FWIW, if anyone else runs into the same kind of issue, I had success running "auc" on the command line, instead of triggering the update via the luci web interface. No idea why it would make a difference - you'd think that in both cases they would be doing the same thing with the asu server.

dangowrt, 2024-08-11:
The main difference is that the LuCI interface uses your browser to communicate with the asu server -- in this way you can update even devices which are not directly connected to the Internet themselves.
`auc` in contrast establishes direct https connection from the device itself to the asu server.