openwrt/asu #949: FriendlyARM NanoPi R5C: Probably errorneous too low storage size restriction.

dreirund, 2024-08-22:
Ahoj,

I want to use the [OpenWrt firmware selector](https://firmware-selector.openwrt.org/?version=SNAPSHOT&target=rockchip%2Farmv8&id=friendlyarm_nanopi-r5c) to build a customised snapshot image for [FriendlyARM NanoPi R5C](https://openwrt.org/toh/friendlyarm/nanopi_r5c).

This device can boot from microSD or from internal flash storage. [Internal storage is 32&thinsp;GiB](https://wiki.friendlyelec.com/wiki/index.php/NanoPi_R5C#Hardware_Spec), microSD can be even larger.

When I try to build with the following pre-selected packages:  
```
adb adb-enablemodem alsa-lib alsa-ucm-conf alsa-utils alsa-utils-seq alsa-utils-tests ar aserver ath6k-firmware ath9k-htc-firmware attr avahi-dbus-daemon badblocks base-files bash bind-dig bind-libs block-mount blockd bluez-libs bluld bmx7 bmx7-iwinfo bmx7-json bmx7-sms bmx7-table bmx7-topology bmx7-tun bmx7-uci-config boost brcmfmac-firmware-usb bridge btrfs-progs busybox ca-bundle carl9170-firmware cfdisk cgdisk cgi-io chat chattr collectd collectd-mod-cpu collectd-mod-csv collectd-mod-curl collectd-mod-df collectd-mod-dhcpleases collectd-mod-disk collectd-mod-dns collectd-mod-email collectd-mod-entropy collectd-mod-ethstat collectd-mod-exec collectd-mod-filecount collectd-mod-fscache collectd-mod-interface collectd-mod-ipstatistics collectd-mod-iptables collectd-mod-irq collectd-mod-iwinfo collectd-mod-load collectd-mod-logfile collectd-mod-match-empty-counter collectd-mod-match-hashed collectd-mod-match-regex collectd-mod-match-timediff collectd-mod-match-value collectd-mod-memory collectd-mod-netlink collectd-mod-network collectd-mod-olsrd collectd-mod-openvpn collectd-mod-ping collectd-mod-processes collectd-mod-protocols collectd-mod-rrdtool collectd-mod-sensors collectd-mod-smart collectd-mod-snmp collectd-mod-snmp6 collectd-mod-sqm collectd-mod-swap collectd-mod-table collectd-mod-tail collectd-mod-tail-csv collectd-mod-tcpconns collectd-mod-thermal collectd-mod-threshold collectd-mod-unixsock collectd-mod-uptime collectd-mod-users collectd-mod-vmem collectd-mod-wireless comgt comgt-directip coreutils coreutils-chroot coreutils-sha1sum curl cypress-firmware-4373-usb dbus dbus-utils dcwapd ddns-scripts ddns-scripts-services dkjson dmesg dnsmasq-full dosfstools dropbear dumpe2fs e2freefrag e2fsprogs e4crypt edgeport-firmware espeak etherwake ethtool-full exfat-fsck exfat-mkfs f2fs-tools f2fsck faad2 fatresize fbtest fdk-aac ffprobe firewall4 fstools fwtool gammu gdisk getrandom glib2 gpioctl-sysfs gpiod-tools gre grep gsm-utils gsmlib hfsfsck hidapi hostapd-common hostapd-utils hplip-common hplip-sane ip-bridge ip-full ip6tables-mod-nat ip6tables-nft ip6tables-zz-legacy ipip iptables-mod-conntrack-extra iptables-mod-iface iptables-mod-ipopt iptables-mod-iprange iptables-mod-led iptables-mod-nat-extra iptables-mod-physdev iptables-nft iptables-zz-legacy iputils-tracepath iw-full iwinfo jansson4 jshn jsoncpp jsonfilter kernel kmod-gpio-button-hotplug kmod-nft-offload kmod-r8169 kmod-rtw88-8822ce lame-lib lcd4linux-custom lcdproc-clients lcdproc-drivers lcdproc-server libc libgcc linux-atm lm-sensors lm-sensors-detect logd losetup lsattr lsblk lscpu lslocks lsns lua lua-bencode lua-sha2 luasocket luci luci-app-bmx7 luci-app-commands luci-app-dcwapd luci-app-ddns luci-app-firewall luci-app-ksmbd luci-app-ledtrig-rssi luci-app-ledtrig-switch luci-app-ledtrig-usbport luci-app-minidlna luci-app-mjpg-streamer luci-app-mwan3 luci-app-nlbwmon luci-app-olsr luci-app-olsr-services luci-app-olsr-viz luci-app-openvpn luci-app-opkg luci-app-p910nd luci-app-sqm luci-app-squid luci-app-statistics luci-app-travelmate luci-app-uhttpd luci-app-upnp luci-app-vnstat2 luci-app-watchcat luci-app-wifischedule luci-app-wol luci-app-xinetd luci-base luci-compat luci-lib-base luci-lib-httpclient luci-lib-httpprotoutils luci-lib-ip luci-lib-ipkg luci-lib-json luci-lib-jsonc luci-lib-nixio luci-lib-px5g luci-mod-admin-full luci-mod-network luci-mod-status luci-mod-system luci-proto-3g luci-proto-bonding luci-proto-gre luci-proto-hnet luci-proto-ipip luci-proto-ipv6 luci-proto-modemmanager luci-proto-ncm luci-proto-openconnect luci-proto-openfortivpn luci-proto-ppp luci-proto-pppossh luci-proto-qmi luci-proto-relay luci-proto-sstp luci-proto-vpnc luci-proto-vxlan luci-proto-wireguard luci-proto-xfrm luci-ssl-openssl luci-theme-bootstrap luci-theme-material luci-theme-openwrt lvm2 lynx lz4 lzmadec lzmainfo maccalc macchanger madplay mbim-utils mc mc-skins minidlna miniupnpc miniupnpd-nftables mjpg-streamer mjpg-streamer-input-file mjpg-streamer-input-uvc mkf2fs mmc-utils moc modemmanager mpc mrmctl msmtp msmtp-queue mt7601u-firmware mt7622bt-firmware mtd mwan3 nail nano-full ncat-full netifd nft-qos nftables-json nlbwmon ntfs-3g ntfs-3g-utils odhcp6c odhcpd-ipv6only olsrd olsrd-mod-arprefresh olsrd-mod-bmf olsrd-mod-dot-draw olsrd-mod-dyn-gw olsrd-mod-dyn-gw-plain olsrd-mod-filtergw olsrd-mod-httpinfo olsrd-mod-jsoninfo olsrd-mod-mdns olsrd-mod-nameservice olsrd-mod-netjson olsrd-mod-p2pd olsrd-mod-pgraph olsrd-mod-pud olsrd-mod-quagga olsrd-mod-secure olsrd-mod-sgwdynspeed olsrd-mod-txtinfo olsrd-mod-watchdog olsrd-utils openconnect openfortivpn openssh-sftp-server openssl-util openvpn-openssl openwrt-keyring opkg p11-kit p54-usb-firmware p910nd partx-utils pciids pciutils perl perlbase-base perlbase-bytes perlbase-class perlbase-config perlbase-cwd perlbase-errno perlbase-essential perlbase-fcntl perlbase-file perlbase-filehandle perlbase-i18n perlbase-integer perlbase-io perlbase-list perlbase-locale perlbase-params perlbase-posix perlbase-re perlbase-scalar perlbase-selectsaver perlbase-socket perlbase-symbol perlbase-tie perlbase-unicore perlbase-utf8 perlbase-xsloader picocom pigz portaudio ppp ppp-mod-passwordfd ppp-mod-pppoa ppp-mod-pppoe ppp-mod-pptp ppp-mod-radius pppdump pppoe-discovery pppossh procd procd-seccomp procd-ujail proto-bonding pulseaudio-daemon pulseaudio-profiles pulseaudio-tools pv python3 python3-asyncio python3-base python3-cgi python3-cgitb python3-codecs python3-ctypes python3-dbm python3-decimal python3-distutils python3-email python3-light python3-logging python3-lzma python3-multiprocessing python3-ncurses python3-openssl python3-pydoc python3-readline python3-sqlite3 python3-unittest python3-urllib python3-uuid python3-xml qmi-utils r8152-firmware r8169-firmware relayd resize2fs resolveip rpcd rpcd-mod-file rpcd-mod-iwinfo rpcd-mod-luci rpcd-mod-rpcsys rpcd-mod-rrdns rpcd-mod-ucode rrdtool1 rs9113-firmware rssileds rt2800-usb-firmware rtl8188eu-firmware rtl8192ce-firmware rtl8192cu-firmware rtl8192de-firmware rtl8192eu-firmware rtl8192se-firmware rtl8723au-firmware rtl8723bu-firmware rtl8821ae-firmware rtl8822be-firmware rtl8822ce-firmware sane-daemon sane-net sane-v4l serdisplib setterm shine sox spi-tools spidev-test sqm-scripts squashfs-tools-mksquashfs squashfs-tools-unsquashfs squid squid-mod-cachemgr ss sstp-client strace swap-utils sysfsutils tc-full tcpdump terminfo tmux travelmate triggerhappy tune2fs uboot-envtools ubox ubus ubusd uci uclient-fetch ucode ucode-mod-fs ucode-mod-ubus ucode-mod-uci uhttpd uhttpd-mod-lua uhttpd-mod-ubus uhttpd-mod-ucode umbim unrar upmpdcli uqmi urandom-seed urngd usb-modeswitch usbids usign uuidgen vnstat2 vnstati2 vpnc vpnc-scripts vxlan watchcat wget-ssl wifischedule wifitoggle wireguard-tools wireless-regdb wireless-tools wpad wsdd2 wwan xfrm xfs-fsck xinetd xtables-legacy xtables-nft xz xz-utils xzdec xzdiff xzgrep xzless xzmore yajl yggdrasil zlib zram-swap zstd
```

the build fails with

> *Error: Selected packages exceed device storage*

Further output from the firmware selector web interface:

* STDERR:  
  ```
  Generate local signing keys...
  Generate local certificate...
  Package list missing or not up-to-date, generating it.

  Building package index...
  Downloading https://downloads.openwrt.org/snapshots/targets/rockchip/armv8/packages/Packages.gz
  Updated list of available packages in /builder/build_dir/target-aarch64_generic_musl/root-rockchip/../../../../builder/dl/openwrt_core
  Downloading https://downloads.openwrt.org/snapshots/targets/rockchip/armv8/packages/Packages.sig
  Signature check passed.
  Downloading https://downloads.openwrt.org/snapshots/packages/aarch64_generic/base/Packages.gz
  Updated list of available packages in /builder/build_dir/target-aarch64_generic_musl/root-rockchip/../../../../builder/dl/openwrt_base
  Downloading https://downloads.openwrt.org/snapshots/packages/aarch64_generic/base/Packages.sig
[... cut ...]
  [...]
  Creating filesystem with parameters:
      Size: 109051904
  [...]
  ```  
  which seems to be ~1&thinsp;GiB, smaller than 32&thinsp;GiB.

  *(I came to the conclusion "~1&thinsp;GiB", since when I request a build just with `luci` specified as package, I end up with image files that, after `gunzip`, are ca. 168&thinsp;MiB in size and the number in STDOUT reads `Size: 16777216`, which gives a factor of ca. 100&hairsp;000. Applying this to `Size: 109051904` gives ~1&thinsp;GiB.)*

Regards!

ristomatti, 2024-08-29:
I just got the same error but for NanoPI R2S which boots only from microSD.

ristomatti, 2024-08-29:
It looks like the size is limited on purpose (unless I'm looking at the wrong file):

https://github.com/openwrt/asu/blob/92ee38499d24c08869739a2b2a26ae86c8c56d1c/asu/config.py#L17

dreirund, 2024-08-29:
> It looks like the size is limited on purpose (unless I'm looking at
> the wrong file):
> 
> https://github.com/openwrt/asu/blob/92ee38499d24c08869739a2b2a26ae86c8c56d1c/asu/config.py#L17

Then the purpose is probably errorneous too general restrictive(?).

ristomatti, 2024-08-29:
Considering the firmware selector is free to use but certainly not free to host, I'd say it'd be an understandable bandwidth/cost saving measure. It is still somewhat frustrating. I was trying to include `ffmpeg` to be able to re-encode an RTSP stream I'm proxying using MediaMTX but it pulled in too many dependencies to fit. I then spent ~2 hours trying to get the Podman version running locally but the instructions seem to be missing something. Having gone through the `menuconfig` route once ~3 years ago, I remember it was cumbersome enough to setup that I'll rather just come up with another way around the limitation...

dreirund, 2024-08-30:
> Considering the firmware selector is free to use but certainly not
> free to host, I'd say it'd be an understandable bandwidth/cost saving
> measure.

OK that is!, for that reason the error message is confusing.

Regards for your find of the 1024 MiB limit; I seem not to be too far
away from it, so I can tweak.

Regards!

ristomatti, 2024-08-30:
I found a static build of `ffmpeg` that'd likely work but it's ~50M. It won't fit even on my other R2S into which I've only added `mediamtx`'s 35M binary.. only 32M left :facepalm:.

Using an EXT4 installation would make this a non-issue but the two R2S's are responsible of linking a remote (300km away) site to my home network, so I'd find it too risky. Especially since the remote site has recurring short power outages.

Perhaps I might have to reconsider checking if creating an image with `menuconfig` has become simpler... :thinking: