openwrt/asu #961: sysupgrade and custom build requests failing

lorand-horvath, 2024-09-07:
@aparcar @efahl ASU and firmware-selector fail to build for a few days now, it seems the sysupgrade server is down or its API is somehow broken.
Please see reports:
https://github.com/openwrt/openwrt/issues/15192#issuecomment-2331246644
https://forum.openwrt.org/t/the-openwrt-firmware-selector/81721/874
https://forum.openwrt.org/t/the-openwrt-firmware-selector/81721/885

efahl, 2024-09-07:
Yeah, the front end API server there was completely dead for a while (503 errors fetching stuff from json/v1), then it came back.  Still dead with any build requests are just returning `Internal Server Error'.

hainguyen-cmd, 2024-09-08:
Until now, ASU still have problem! Waiting for @aparcar @efahl !!

efahl, 2024-09-08:
Sorry, not much I can do, I only work on the clients (and sometimes submit PRs to ASU).  Paul is the only one I know who can actually log into the servers.

sqliuchang, 2024-09-09:
[https://sysupgrade.openwrt.org/api/v1/build](https://sysupgrade.openwrt.org/api/v1/build) is still not working.

sqliuchang, 2024-09-09:
It works now.

hainguyen-cmd, 2024-09-09:
Still got error:
`Downloading file:packages/Packages
Updated list of available packages in /builder/build_dir/target-mipsel_24kc_musl/root-ramips/../../../../builder/dl/imagebuilder
Downloading file:packages/Packages.sig
Signature check passed.
Collected errors:
 * opkg_install_cmd: Cannot install package luci-app-wireguard.
make[2]: *** [Makefile:220: package_install] Error 255
make[1]: *** [Makefile:161: _call_manifest] Error 2
make: *** [Makefile:322: manifest] Error 2`

efahl, 2024-09-09:
> opkg_install_cmd: Cannot install package luci-app-wireguard.

That's a package issue, nothing to do with ASU server.  You can check on the download server to see if it's there, and try again later.

hainguyen-cmd, 2024-09-09:
> > opkg_install_cmd: Cannot install package luci-app-wireguard.
> 
> That's a package issue, nothing to do with ASU server. You can check on the download server to see if it's there, and try again later.

Thanks so much! I have checked again. Everything work fine!