openwrt/asu #987: stats page broken?

henriklb, 2024-09-23:
https://sysupgrade.openwrt.org/stats/d/cdtwsm4clr9xcb appears to be broken. 

Got it from [here ](https://sysupgrade.openwrt.org/): View the Server Stats: [Server Stats](https://sysupgrade.openwrt.org/stats)

henriklb, 2024-12-04:
Appears to be broken still, months later

aparcar, 2024-12-04:
Yea I'm busy doing the basic maintenance. Are you interested in taking on the challenge?

henriklb, 2024-12-04:
Thank you for acknowledging that you are aware of the issue. My intention was to be helpful by reporting this. 

If I may, I suggest removing the references to the stats page entirely. Or perhaps add an informative note on the page to let users know that the maintainers know that it is down, to avoid confusion