openwrt/asu #989: Unsupported target: ath79/generic

damianperera, 2024-09-24:
### Request Data

```json
{
    "url": "https://sysupgrade.openwrt.org",
    "revision": "r24012-d8dd03c46f",
    "advanced_mode": "0",
    "sha256_unsigned": "",
    "branch": "23.05",
    "efi": null,
    "profile": "tplink,tl-wdr3600-v1",
    "target": "ath79/generic",
    "version": "23.05.5",
    "packages": [
        "attendedsysupgrade-common",
        "base-files",
        "busybox",
        "ca-bundle",
        "cgi-io",
        "dnsmasq",
        "dropbear",
        "firewall4",
        "fstools",
        "fwtool",
        "getrandom",
        "iw",
        "iwinfo",
        "jansson",
        "jshn",
        "jsonfilter",
        "kernel",
        "kmod-ath",
        "kmod-ath9k",
        "kmod-ath9k-common",
        "kmod-cfg80211",
        "kmod-crypto-acompress",
        "kmod-crypto-aead",
        "kmod-crypto-ccm",
        "kmod-crypto-cmac",
        "kmod-crypto-crc32c",
        "kmod-crypto-ctr",
        "kmod-crypto-gcm",
        "kmod-crypto-gf128",
        "kmod-crypto-ghash",
        "kmod-crypto-hash",
        "kmod-crypto-hmac",
        "kmod-crypto-manager",
        "kmod-crypto-null",
        "kmod-crypto-rng",
        "kmod-crypto-seqiv",
        "kmod-crypto-sha512",
        "kmod-gpio-button-hotplug",
        "kmod-lib-crc-ccitt",
        "kmod-lib-crc32c",
        "kmod-lib-lzo",
        "kmod-mac80211",
        "kmod-nf-conntrack",
        "kmod-nf-conntrack6",
        "kmod-nf-flow",
        "kmod-nf-log",
        "kmod-nf-log6",
        "kmod-nf-nat",
        "kmod-nf-reject",
        "kmod-nf-reject6",
        "kmod-nfnetlink",
        "kmod-nft-core",
        "kmod-nft-fib",
        "kmod-nft-nat",
        "kmod-nft-offload",
        "kmod-ppp",
        "kmod-pppoe",
        "kmod-pppox",
        "kmod-random-core",
        "kmod-slhc",
        "kmod-usb-ledtrig-usbport",
        "kmod-usb2",
        "libblobmsg-json",
        "libc",
        "libiwinfo",
        "libiwinfo-data",
        "libjson-c",
        "libjson-script",
        "liblucihttp",
        "liblucihttp-ucode",
        "libmbedtls",
        "libmnl",
        "libnftnl",
        "libnl-tiny",
        "libubox",
        "libubus",
        "libuci",
        "libuclient",
        "libucode",
        "libustream-mbedtls",
        "logd",
        "luci",
        "luci-app-attendedsysupgrade",
        "luci-app-firewall",
        "luci-app-opkg",
        "luci-base",
        "luci-compat",
        "luci-lib-ipkg",
        "luci-light",
        "luci-mod-admin-full",
        "luci-mod-network",
        "luci-mod-status",
        "luci-mod-system",
        "luci-proto-ipv6",
        "luci-proto-ppp",
        "luci-ssl",
        "luci-theme-bootstrap",
        "mtd",
        "netifd",
        "nftables-json",
        "odhcp6c",
        "odhcpd-ipv6only",
        "openwrt-keyring",
        "opkg",
        "ppp",
        "ppp-mod-pppoe",
        "procd",
        "procd-seccomp",
        "procd-ujail",
        "px5g-mbedtls",
        "rpcd",
        "rpcd-mod-file",
        "rpcd-mod-iwinfo",
        "rpcd-mod-luci",
        "rpcd-mod-rrdns",
        "rpcd-mod-ucode",
        "swconfig",
        "uboot-envtools",
        "ubox",
        "ubus",
        "ubusd",
        "uci",
        "uclient-fetch",
        "ucode",
        "ucode-mod-fs",
        "ucode-mod-html",
        "ucode-mod-math",
        "ucode-mod-nl80211",
        "ucode-mod-rtnl",
        "ucode-mod-ubus",
        "ucode-mod-uci",
        "ucode-mod-uloop",
        "uhttpd",
        "uhttpd-mod-ubus",
        "urandom-seed",
        "urngd",
        "usign",
        "wireless-regdb",
        "wpad-basic-mbedtls"
    ],
    "diff_packages": true,
    "filesystem": "squashfs",
    "client": "luci/git-23.339.51123-138595a"
}
```

huafu, 2024-09-24:
Same here with `ath79/nand`:
```
{
    "url": "https://sysupgrade.openwrt.org",
    "revision": "r24012-d8dd03c46f",
    "advanced_mode": "0",
    "sha256_unsigned": "",
    "branch": "23.05",
    "efi": null,
    "profile": "glinet,gl-ar750s-nor-nand",
    "target": "ath79/nand",
    "version": "23.05.5",
    "packages": [
        "adb",
        "ath10k-firmware-qca9887-ct",
        "base-files",
        "bash",
        "block-mount",
        "btop",
        "busybox",
        "ca-bundle",
        "cgi-io",
        "coreutils-sort",
        "coreutils-timeout",
        "dnsmasq-full",
        "dropbear",
        "firewall4",
        "fstools",
        "fwtool",
        "gawk",
        "getrandom",
        "gpsd",
        "gpsd-clients",
        "gpsd-utils",
        "grep",
        "hostapd-common",
        "ipset",
        "iw",
        "iwinfo",
        "jansson",
        "jq",
        "jshn",
        "jsonfilter",
        "kernel",
        "kmod-ath",
        "kmod-ath10k-ct",
        "kmod-ath9k",
        "kmod-ath9k-common",
        "kmod-cfg80211",
        "kmod-crypto-acompress",
        "kmod-crypto-aead",
        "kmod-crypto-ccm",
        "kmod-crypto-cmac",
        "kmod-crypto-crc32c",
        "kmod-crypto-ctr",
        "kmod-crypto-gcm",
        "kmod-crypto-gf128",
        "kmod-crypto-ghash",
        "kmod-crypto-hash",
        "kmod-crypto-hmac",
        "kmod-crypto-manager",
        "kmod-crypto-null",
        "kmod-crypto-rng",
        "kmod-crypto-seqiv",
        "kmod-crypto-sha512",
        "kmod-gpio-button-hotplug",
        "kmod-lib-crc-ccitt",
        "kmod-lib-crc32c",
        "kmod-lib-lzo",
        "kmod-mac80211",
        "kmod-nf-conntrack",
        "kmod-nf-conntrack6",
        "kmod-nf-flow",
        "kmod-nf-log",
        "kmod-nf-log6",
        "kmod-nf-nat",
        "kmod-nf-reject",
        "kmod-nf-reject6",
        "kmod-nfnetlink",
        "kmod-nft-core",
        "kmod-nft-fib",
        "kmod-nft-nat",
        "kmod-nft-offload",
        "kmod-nls-utf8",
        "kmod-ppp",
        "kmod-pppoe",
        "kmod-pppox",
        "kmod-random-core",
        "kmod-slhc",
        "kmod-usb-acm",
        "kmod-usb-storage",
        "kmod-usb2",
        "libblobmsg-json",
        "libc",
        "libiwinfo",
        "libiwinfo-data",
        "libjson-c",
        "libjson-script",
        "liblucihttp",
        "liblucihttp-ucode",
        "libmbedtls",
        "libmnl",
        "libnetfilter-conntrack",
        "libnettle",
        "libnftnl",
        "libnl-tiny",
        "libubox",
        "libubus",
        "libuci",
        "libuclient",
        "libucode",
        "libustream-mbedtls",
        "logd",
        "logrotate",
        "luci",
        "luci-app-attendedsysupgrade",
        "luci-app-firewall",
        "luci-app-opkg",
        "luci-app-statistics",
        "luci-app-vnstat2",
        "luci-base",
        "luci-light",
        "luci-mod-admin-full",
        "luci-mod-dashboard",
        "luci-mod-network",
        "luci-mod-status",
        "luci-mod-system",
        "luci-proto-ipv6",
        "luci-proto-ppp",
        "luci-proto-wireguard",
        "luci-ssl",
        "luci-theme-bootstrap",
        "mtd",
        "nano",
        "netifd",
        "nftables-json",
        "ntp-utils",
        "ntpd",
        "odhcp6c",
        "odhcpd-ipv6only",
        "openwrt-keyring",
        "opkg",
        "ppp",
        "ppp-mod-pppoe",
        "procd",
        "procd-seccomp",
        "procd-ujail",
        "procps-ng-ps",
        "procps-ng-top",
        "px5g-mbedtls",
        "rpcd",
        "rpcd-mod-file",
        "rpcd-mod-iwinfo",
        "rpcd-mod-luci",
        "rpcd-mod-rrdns",
        "rpcd-mod-ucode",
        "sed",
        "swconfig",
        "ubi-utils",
        "uboot-envtools",
        "ubox",
        "ubus",
        "ubusd",
        "uci",
        "uclient-fetch",
        "ucode",
        "ucode-mod-fs",
        "ucode-mod-html",
        "ucode-mod-math",
        "ucode-mod-nl80211",
        "ucode-mod-rtnl",
        "ucode-mod-ubus",
        "ucode-mod-uci",
        "ucode-mod-uloop",
        "uhttpd",
        "uhttpd-mod-ubus",
        "uhubctl",
        "urandom-seed",
        "urngd",
        "usbutils",
        "usign",
        "wireless-regdb",
        "wpad-basic-mbedtls",
        "zoneinfo-all"
    ],
    "diff_packages": true,
    "filesystem": "squashfs",
    "client": "luci/git-23.339.51123-138595a"
}
```

aparcar, 2024-09-24:
Yea please hang in there

aparcar, 2024-09-27:
It's there now.