openwrt/asu #1003: Attended sysupgrade failed - avm,fritzbox-7530 - 23.05.4 to 23.05.5

anikolop, 2024-09-30:
```
Request Data:
{
    "url": "https://sysupgrade.openwrt.org",
    "revision": "r24012-d8dd03c46f",
    "advanced_mode": "0",
    "sha256_unsigned": "",
    "branch": "23.05",
    "efi": null,
    "profile": "avm,fritzbox-7530",
    "target": "ipq40xx/generic",
    "version": "23.05.5",
    "packages": [
        "ath10k-board-qca4019",
        "ath10k-firmware-qca4019-ct",
        "base-files",
        "busybox",
        "ca-bundle",
        "cgi-io",
        "curl",
        "ddns-scripts-noip",
        "dnsmasq",
        "dropbear",
        "firewall4",
        "fritz-caldata",
        "fritz-tffs-nand",
        "fstools",
        "fwtool",
        "getrandom",
        "hostapd-common",
        "iw",
        "iwinfo",
        "jansson",
        "jshn",
        "jsonfilter",
        "kernel",
        "kmod-ath",
        "kmod-ath10k-ct",
        "kmod-cfg80211",
        "kmod-crypto-acompress",
        "kmod-crypto-aead",
        "kmod-crypto-ccm",
        "kmod-crypto-cmac",
        "kmod-crypto-crc32c",
        "kmod-crypto-ctr",
        "kmod-crypto-gcm",
        "kmod-crypto-gf128",
        "kmod-crypto-ghash",
        "kmod-crypto-hash",
        "kmod-crypto-hmac",
        "kmod-crypto-manager",
        "kmod-crypto-null",
        "kmod-crypto-rng",
        "kmod-crypto-seqiv",
        "kmod-crypto-sha512",
        "kmod-gpio-button-hotplug",
        "kmod-hwmon-core",
        "kmod-leds-gpio",
        "kmod-lib-crc-ccitt",
        "kmod-lib-crc32c",
        "kmod-lib-lzo",
        "kmod-mac80211",
        "kmod-nf-conntrack",
        "kmod-nf-conntrack6",
        "kmod-nf-flow",
        "kmod-nf-log",
        "kmod-nf-log6",
        "kmod-nf-nat",
        "kmod-nf-reject",
        "kmod-nf-reject6",
        "kmod-nfnetlink",
        "kmod-nft-core",
        "kmod-nft-fib",
        "kmod-nft-nat",
        "kmod-nft-offload",
        "kmod-nls-base",
        "kmod-ppp",
        "kmod-pppoe",
        "kmod-pppox",
        "kmod-slhc",
        "kmod-usb-core",
        "kmod-usb-dwc3",
        "kmod-usb-dwc3-qcom",
        "kmod-usb-net-rndis",
        "kmod-usb-xhci-hcd",
        "kmod-usb3",
        "libblobmsg-json",
        "libc",
        "libcurl",
        "libiwinfo",
        "libiwinfo-data",
        "libjson-c",
        "libjson-script",
        "liblua",
        "liblucihttp",
        "liblucihttp-ucode",
        "libmbedtls",
        "libmnl",
        "libnftnl",
        "libnl-tiny",
        "libubox",
        "libubus",
        "libuci",
        "libuclient",
        "libucode",
        "libustream-mbedtls",
        "logd",
        "ltq-vdsl-vr11-app",
        "lua",
        "luci",
        "luci-app-attendedsysupgrade",
        "luci-app-ddns",
        "luci-app-firewall",
        "luci-app-openvpn",
        "luci-app-opkg",
        "luci-app-wol",
        "luci-base",
        "luci-light",
        "luci-mod-admin-full",
        "luci-mod-network",
        "luci-mod-status",
        "luci-mod-system",
        "luci-proto-ipv6",
        "luci-proto-ppp",
        "luci-proto-wireguard",
        "luci-ssl",
        "luci-theme-bootstrap",
        "mtd",
        "netifd",
        "nftables-json",
        "odhcp6c",
        "odhcpd-ipv6only",
        "openvpn-openssl",
        "openwrt-keyring",
        "opkg",
        "ppp",
        "ppp-mod-pppoe",
        "procd",
        "procd-seccomp",
        "procd-ujail",
        "px5g-mbedtls",
        "qrencode",
        "rpcd",
        "rpcd-mod-file",
        "rpcd-mod-iwinfo",
        "rpcd-mod-luci",
        "rpcd-mod-rrdns",
        "rpcd-mod-ucode",
        "ubi-utils",
        "uboot-envtools",
        "ubox",
        "ubus",
        "ubusd",
        "uci",
        "uclient-fetch",
        "ucode",
        "ucode-mod-fs",
        "ucode-mod-html",
        "ucode-mod-math",
        "ucode-mod-nl80211",
        "ucode-mod-rtnl",
        "ucode-mod-ubus",
        "ucode-mod-uci",
        "ucode-mod-uloop",
        "uhttpd",
        "uhttpd-mod-ubus",
        "urandom-seed",
        "urngd",
        "usign",
        "wireguard-tools",
        "wireless-regdb",
        "wpad-basic-mbedtls"
    ],
    "diff_packages": true,
    "filesystem": "squashfs",
    "client": "luci/git-23.339.51123-138595a"
}
```

```
STD ERR:
Generate local signing keys...
Generate local certificate...
Package list missing or not up-to-date, generating it.

Building package index...
Downloading https://downloads.openwrt.org/releases/23.05.5/targets/ipq40xx/generic/packages/Packages.gz
Updated list of available packages in /builder/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/root-ipq40xx/../../../../builder/dl/openwrt_core
Downloading https://downloads.openwrt.org/releases/23.05.5/targets/ipq40xx/generic/packages/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/23.05.5/packages/arm_cortex-a7_neon-vfpv4/base/Packages.gz
Updated list of available packages in /builder/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/root-ipq40xx/../../../../builder/dl/openwrt_base
Downloading https://downloads.openwrt.org/releases/23.05.5/packages/arm_cortex-a7_neon-vfpv4/base/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/23.05.5/packages/arm_cortex-a7_neon-vfpv4/luci/Packages.gz
Updated list of available packages in /builder/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/root-ipq40xx/../../../../builder/dl/openwrt_luci
Downloading https://downloads.openwrt.org/releases/23.05.5/packages/arm_cortex-a7_neon-vfpv4/luci/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/23.05.5/packages/arm_cortex-a7_neon-vfpv4/packages/Packages.gz
Updated list of available packages in /builder/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/root-ipq40xx/../../../../builder/dl/openwrt_packages
Downloading https://downloads.openwrt.org/releases/23.05.5/packages/arm_cortex-a7_neon-vfpv4/packages/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/23.05.5/packages/arm_cortex-a7_neon-vfpv4/routing/Packages.gz
Updated list of available packages in /builder/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/root-ipq40xx/../../../../builder/dl/openwrt_routing
Downloading https://downloads.openwrt.org/releases/23.05.5/packages/arm_cortex-a7_neon-vfpv4/routing/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/23.05.5/packages/arm_cortex-a7_neon-vfpv4/telephony/Packages.gz
Updated list of available packages in /builder/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/root-ipq40xx/../../../../builder/dl/openwrt_telephony
Downloading https://downloads.openwrt.org/releases/23.05.5/packages/arm_cortex-a7_neon-vfpv4/telephony/Packages.sig
Signature check passed.
Downloading file:packages/Packages
Updated list of available packages in /builder/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/root-ipq40xx/../../../../builder/dl/imagebuilder
Downloading file:packages/Packages.sig
Signature check passed.
Collected errors:
 * opkg_install_pkg: Checksum or size mismatch for package busybox. Either the opkg or the package index are corrupt. Try 'opkg update'.
 * opkg_install_cmd: Cannot install package busybox.
 * opkg_install_pkg: Checksum or size mismatch for package libopenssl3. Either the opkg or the package index are corrupt. Try 'opkg update'.
 * opkg_install_cmd: Cannot install package openvpn-openssl.
 * opkg_install_pkg: Checksum or size mismatch for package busybox. Either the opkg or the package index are corrupt. Try 'opkg update'.
 * opkg_install_cmd: Cannot install package busybox.
make[2]: *** [Makefile:189: package_install] Error 255
make[1]: *** [Makefile:154: _call_manifest] Error 2
make: *** [Makefile:274: manifest] Error 2
```

Brateham, 2024-09-30:
I have a similar problem

aparcar, 2024-09-30:
I just tested the very same request mentioned above without an error...?

Brateham, 2024-09-30:
Collected errors:
 * opkg_install_cmd: Cannot install package badvpn.
make[2]: *** [Makefile:189: package_install] Error 255
make[1]: *** [Makefile:154: _call_manifest] Error 2
make: *** [Makefile:274: manifest] Error 2

Brateham, 2024-09-30:
Partially figured out for my Building images for ramips - TP-Link Archer C6U v1. after removing the package   badvpn The update process has started, but unfortunately it did not reach 100%, a message about the following error appeared.
Error building the firmware image
Server response: Error: No JSON file found
`Number of uids 1
	unknown (0)
Number of gids 1
	unknown (0)
Skip JSON creation for non existing file /builder/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/tmp/openwrt-23.05.5-026bf2e19c93-ramips-mt7621-tplink_archer-c6u-v1-squashfs-sysupgrade.bin
Skip JSON creation for non existing file /builder/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/tmp/openwrt-23.05.5-026bf2e19c93-ramips-mt7621-tplink_archer-c6u-v1-squashfs-factory.bin
JSON info file script could not find any JSON files for target

Calculating checksums...`

`5/packages/mipsel_24kc/telephony/Packages.sig
Signature check passed.
Downloading file:packages/Packages
Updated list of available packages in /builder/build_dir/target-mipsel_24kc_musl/root-ramips/../../../../builder/dl/imagebuilder
Downloading file:packages/Packages.sig
Signature check passed.
Pseudo file "dev" exists in source filesystem "/builder/build_dir/target-mipsel_24kc_musl/root-ramips/dev".
Ignoring, exclude it (-e/-ef) to override.
33133+1 records in
33133+1 records out
16964340 bytes (17 MB, 16 MiB) copied, 0.0421237 s, 403 MB/s
file-system partition too big (more than 13053119 bytes): Success
stat: cannot statx '/builder/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/tmp/openwrt-23.05.5-026bf2e19c93-ramips-mt7621-tplink_archer-c6u-v1-squashfs-sysupgrade.bin': No such file or directory
bash: line 1: [: 16121856: unary operator expected
bash: line 1: 8: Bad file descriptor
Failed to open firmware file
sha256sum: /builder/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/tmp/openwrt-23.05.5-026bf2e19c93-ramips-mt7621-tplink_archer-c6u-v1-squashfs-sysupgrade.bin: No such file or directory
cp: cannot stat '/builder/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/tmp/openwrt-23.05.5-026bf2e19c93-ramips-mt7621-tplink_archer-c6u-v1-squashfs-sysupgrade.bin': No such file or directory
33133+1 records in
33133+1 records out
16964340 bytes (17 MB, 16 MiB) copied, 0.037136 s, 457 MB/s
file-system partition too big (more than 13041664 bytes): Success
cp: cannot stat '/builder/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/tmp/openwrt-23.05.5-026bf2e19c93-ramips-mt7621-tplink_archer-c6u-v1-squashfs-factory.bin': No such file or directory`

anikolop, 2024-10-01:
Tried today and it seems that the issue is fixed.

closed 2024-10-01 by anikolop

Brateham, 2024-10-01:
Great, I'll try the update today