openwrt/asu #1028: Server response: Error: Selected packages exceed device storage

lampra1, 2024-10-09:
Currently running: 22.03.0 - r19685-512e76967f
[Powered by LuCI openwrt-22.03 branch (git-22.245.77528-487e58a)](https://github.com/openwrt/luci) / OpenWrt 22.03.0 r19685-512e76967f
Error building the firmware image

Server response: Error: Selected packages exceed device storage
[Please report the error message and request](https://github.com/openwrt/asu/issues)

Request Data:

{
    "url": "https://sysupgrade.openwrt.org",
    "revision": "r19685-512e76967f",
    "advanced_mode": "1",
    "branch": "22.03",
    "efi": null,
    "profile": "cubietech,cubietruck",
    "target": "sunxi/cortexa7",
    "version": "22.03.7",
    "packages": [
        "arp-scan",
        "arp-scan-database",
        "base-files",
        "bash",
        "blkid",
        "block-mount",
        "bluez-daemon",
        "bluez-libs",
        "bluez-utils",
        "busybox",
        "ca-bundle",
        "cgi-io",
        "curl",
        "ddns-scripts",
        "dnsmasq",
        "dockerd",
        "dropbear",
        "e2fsprogs",
        "fdisk",
        "firewall4",
        "fstools",
        "fwtool",
        "getrandom",
        "htop",
        "jansson",
        "jshn",
        "jsonfilter",
        "kernel",
        "kmod-ata-sunxi",
        "kmod-bluetooth",
        "kmod-brcmfmac",
        "kmod-crypto-crc32c",
        "kmod-crypto-hash",
        "kmod-lib-crc-ccitt",
        "kmod-lib-crc32c",
        "kmod-nf-conntrack",
        "kmod-nf-conntrack6",
        "kmod-nf-flow",
        "kmod-nf-log",
        "kmod-nf-log6",
        "kmod-nf-nat",
        "kmod-nf-reject",
        "kmod-nf-reject6",
        "kmod-nfnetlink",
        "kmod-nft-core",
        "kmod-nft-fib",
        "kmod-nft-nat",
        "kmod-nft-offload",
        "kmod-ppp",
        "kmod-pppoe",
        "kmod-pppox",
[... 1206 lines cut ...]
Downloading https://downloads.openwrt.org/releases/22.03.7/packages/arm_cortex-a7_neon-vfpv4/luci/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/22.03.7/packages/arm_cortex-a7_neon-vfpv4/packages/Packages.gz
Updated list of available packages in /builder/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/root-sunxi/../../../../builder/dl/openwrt_packages
Downloading https://downloads.openwrt.org/releases/22.03.7/packages/arm_cortex-a7_neon-vfpv4/packages/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/22.03.7/packages/arm_cortex-a7_neon-vfpv4/routing/Packages.gz
Updated list of available packages in /builder/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/root-sunxi/../../../../builder/dl/openwrt_routing
Downloading https://downloads.openwrt.org/releases/22.03.7/packages/arm_cortex-a7_neon-vfpv4/routing/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/22.03.7/packages/arm_cortex-a7_neon-vfpv4/telephony/Packages.gz
Updated list of available packages in /builder/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/root-sunxi/../../../../builder/dl/openwrt_telephony
Downloading https://downloads.openwrt.org/releases/22.03.7/packages/arm_cortex-a7_neon-vfpv4/telephony/Packages.sig
Signature check passed.
Downloading file:packages/Packages
Updated list of available packages in /builder/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/root-sunxi/../../../../builder/dl/imagebuilder
Downloading file:packages/Packages.sig
Signature check passed.
Pseudo file "/dev" exists in source filesystem "/builder/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/root-sunxi/dev".
Ignoring, exclude it (-e/-ef) to override.
error: ext4_allocate_best_fit_partial: failed to allocate 1816 blocks, out of space?
make[3]: *** [/builder/include/image.mk:339: /builder/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-sunxi_cortexa7/root.ext4] Error 1
make[2]: *** [Makefile:191: build_image] Error 2
make[1]: *** [Makefile:129: _call_image] Error 2
make: *** [Makefile:245: image] Error 2

lordmortis, 2024-10-26:
I'm getting a similar error, i've attached the log 
[sysupgrade error.txt](https://github.com/user-attachments/files/17528861/sysupgrade.error.txt)

DooMMasteR, 2024-10-31:
I am getting the same error for any image I try to create for a nanopi-r6s.

lordmortis, 2024-11-16:
Still getting this one