openwrt/asu #1059: Download Link Opens Binary File in Browser Instead of Prompting Download (Custom Images)

0nelight, 2024-10-26:
Download-Links to freshly build Images via the Browser-Interface of the public [Firmware-Selector-Server](https://firmware-selector.openwrt.org/) on the [public ASU](https://sysupgrade.openwrt.org/) seem to not being correctly advertised as `Content-Disposition: attachment` from the Webserver running on ASU.

This leads to the Issue where, when clicking on the Download-Button, instead of Firefox opening the Download-File-Dialog, it interprets the File as a Webpage and tries to render it. Workaround is clicking "Save Target as".

I am not sure if I should put this here or at firmware-selector-openwrt-org - therefore I created a Ticket there too.

0nelight, 2024-10-26:
Please follow discussion here until there is a Decision on where this Issue belongs:
https://github.com/openwrt/firmware-selector-openwrt-org/issues/53#issuecomment-2439583322

Maybe also related [to this](https://github.com/openwrt/asu/issues/644).

AliveDevil, 2024-10-26:
Context chain:
- Set up staticfiles: https://github.com/openwrt/asu/blob/main/asu/main.py#L39
- staticfiles from fastio is forwarded to starlette: https://github.com/fastapi/fastapi/blob/master/fastapi/staticfiles.py
- starlette returns FileResponse: https://github.com/encode/starlette/blob/master/starlette/staticfiles.py#L178
- Which, if MediaType is None (which it possibly always is), guesses the content type: https://github.com/encode/starlette/blob/master/starlette/responses.py#L301-L302
- Which is a bundled component: https://docs.python.org/3/library/mimetypes.html

All that is to say: ASU should probably register `mimetypes.add_type("application/octet-stream", ".img.gz")`.

AliveDevil, 2024-10-27:
Check the godbolt output: Python can guess encoding and file type separately.
Any .X.Y file is checked against the encoding list of file extensions (i.e. gz, xz, bz2) - this is ignored and dropped by starlette.
So a .jpg.bz2 would be `(application/jpeg, BZip)`.

Starlette only cares about the guessed type: in this case .img evaluates to None, which is then falled back to text/plain.

This could be fixed by editing the mime.types file, or just fixed in application code, which would be less error-prone for users.

0nelight, 2024-10-27:
So this would mean we have to do mimetypes.add_type for other File Extensions too. 

I queried the firmware names and found the following non-duplicated list:

| Base Name                                  | Extension          |
|:-------------------------------------------|:-------------------|
| netgear_wndap620-squashfs-factory.img      | img                |
| netgear_wndr4700-device-tree.dtb           | dtb                |
| wd_mybooklive-ext4-factory.img.gz          | img; gz            |
| atmel_at91sam9263ek-ext4-root.ubi.gz       | ubi; gz            |
| atmel_at91sam9263ek-fit-zImage.itb         | itb                |
| atmel_at91sam9263ek-squashfs-root.ubi      | ubi                |
| atmel_at91sam9g15ek-uImage                 |                    |
| engenius_epg5000-squashfs-factory.dlf      | dlf                |
| standard-squashfs.trx                      | trx                |
| netgear_r6200-v1-squashfs.chk              | chk                |
| luxul_abr-4500-squashfs.lxl                | lxl                |
| gateworks_ventana-squashfs-img.gz          | gz                 |
| toradex_apalis-recovery.scr                | scr                |
| google_wifi-initramfs-fit-zImage.itb.vboot | itb; vboot         |
| netgear_wax218-web-ui-factory.fit          | fit                |
| ctera_c200-v1-initramfs-factory.firm       | firm               |
| iom_iconnect-1.1-initramfs-uImage          | 1-initramfs-uImage |
| vmlinux-initramfs.elf                      | elf                |
| bananapi_bpi-r3-emmc-bl31-uboot.fip        | fip                |
| wevo_air-duo-initramfs.upload              | upload             |
| netgear_gs750e-squashfs-factory.bix        | bix                |

So because anyway ASU provides Firmware-Files always it should always set `content-type: application/octet-stream`.
But because the File Extensions are not a limited set, setting it with `mimetypes.add_type()` would be not reliable.

aparcar, 2024-10-28:
Hi, I love how you analyze the issue and come up with ideas. I'm currently a bit focused to spent my OpenWrt time on the APK migration so don't really do stuff for ASU. If you need anything specific, please ping me.

aparcar, 2024-10-28:
This is the current Caddyfile, maybe someone got an idea how to improve it?

```
(cors) {
	@cors_preflight method OPTIONS

	header {
		Access-Control-Allow-Origin "{header.origin}"
		Vary Origin
		Access-Control-Expose-Headers "Authorization"
		Access-Control-Allow-Credentials "true"
	}

	handle @cors_preflight {
		header {
			Access-Control-Allow-Methods "GET, POST, PUT, PATCH, DELETE"
			Access-Control-Max-Age "3600"
		}
		respond "" 204
	}
}

{
	auto_https disable_redirects
	preferred_chains {
		root_common_name "ISRG Root X1"
	}
}

asu-02.infra.openwrt.org asu-02.infra.openwrt.org:80 sysupgrade.openwrt.org sysupgrade.openwrt.org:80 {
	tls mail@aparcar.org
	root * /home/openwrt/store/
	file_server /store/* 
        import cors {header.origin}
	header Access-Control-Allow-Methods "POST, GET, OPTIONS"
	header Access-Control-Allow-Headers "*"
	header Access-Control-Allow-Origin "*"
	reverse_proxy * localhost:8000
	redir /api/overview /json/v1/overview.json
}
```

AliveDevil, 2024-10-28:
Can create a PR tomorrow (CET) with an improved staticfiles handler.

the bandaid solution would be Caddy `handle-path` on the /store-prefix to overwrite the content-type: https://caddyserver.com/docs/caddyfile/directives/handle_path

AliveDevil, 2024-12-04:
> Could you please try again and see if the issue is resolved? If so the Caddy docs are broken, caddy is broken or I misunderstand some stuff, however it worked on my test run with Firefox[^1]
[^1]: https://github.com/openwrt/asu/pull/1064#issuecomment-2515075977

Checked some files with `curl -I`, all return `Content-Type: application/octet-stream`.

erik434, 2025-08-13:
I just spent a few hours digging into this problem (before I found this issue had already been logged), as I was seeing Firefox try to open .itb files as text instead of just download them (on my self-hosted ASU instance). As @AliveDevil noted here, the root cause is that FastAPI (via Starlette, via mimetypes) defaults to `content-type: text/plain` when it's missing a MIME type mapping for a given extension, and many firmware file extensions (including .itb) aren't in the default map for mimetypes.

@aparcar would you consider taking #1064 or a similar fix? I see you commented on that PR that you fixed the issue by having Caddy serve the `/store/` path, but that only helps people using your ASU instance, or their own ASU instance with a custom reverse proxy configuration.

It seems like having the backend return the right content-type in the first place would be the best solution here, as that would let people run ASU without a reverse proxy if they desire. Or even if they're already using a reverse proxy, it would be great to not need hacks (different for each reverse proxy!) to avoid this issue.

erik434, 2025-08-13:
A simple piecemeal solution for anyone else who encounters this problem with their own ASU instance: you can add a mime.types file to your `server` container with something like this under `volumes` in your compose file: `- ./mime.types:/etc/mime.types:ro`. Then put something like this in the ./mime.types file:

```
application/octet-stream itb
```

The values in that file seem to get appended to the default map in the mimetypes library, so you only need to add extensions that are missing or don't have the right default mapping. Additional extensions (like trx/firm/fit) that you'd like to be served with the same content-type can be added to the same line, separated by spaces. And if you want to add other content-types, you can put them on separate lines (with their applicable extensions separated by spaces).

The downside to this approach is that you need to list every extension, as I don't see any way to specify a default/fallback option. Not as comprehensive a fix as AliveDevil's proposed solution, but it's a quick and easy band-aid which doesn't require messing with the ASU source code, just a simple addition to your compose file.