openwrt/asu #1070: [WRT3200ACM] Error building the snapshot firmware-selector & attendedsysupgrade

Blasterdick, 2024-11-06:
```
{
    "url": "https://sysupgrade.openwrt.org",
    "branch": "SNAPSHOT",
    "revision": "r27300-e7ea93e1e3",
    "efi": null,
    "advanced_mode": "0",
    "request_hash": "",
    "sha256_unsigned": "",
    "client": "luci/24.310.66693~1c61a66",
    "packages": {
        "kmod-usb-storage": "6.6.48-r1",
        "luci-app-adblock": "24.224.27442~06037ad",
        "luci-app-mwan3": "24.148.86114~d174bb8",
        "opkg": "2022.02.24~d038e5b6-r2",
        "wpad-basic-mbedtls": "2024.03.09~695277a5-r3",
        "kmod-usb-storage-uas": "6.6.48-r1",
        "libpthread": "1.2.5-r4",
        "busybox": "1.36.1-r1",
        "kmod-btmrvl": "6.6.48-r1",
        "kmod-nft-fib": "6.6.48-r1",
        "kmod-nfnetlink": "6.6.48-r1",
        "kmod-nf-reject6": "6.6.48-r1",
        "bash": "5.2.32-r1",
        "kmod-nf-flow": "6.6.48-r1",
        "kmod-lib-crc-ccitt": "6.6.48-r1",
        "getrandom": "2024.04.26~85f10530-r1",
        "mc": "4.8.32-r1",
        "kmod-pppoe": "6.6.48-r1",
        "kmod-pppox": "6.6.48-r1",
        "kmod-nf-reject": "6.6.48-r1",
        "procd-ujail": "2024.07.07~f230c117-r1",
        "base-files": "1614~e7ea93e1e3",
        "kmod-nf-nat": "6.6.48-r1",
        "kmod-crypto-crc32c": "6.6.48-r1",
        "libustream-mbedtls": "2024.07.28~99bd3d2b-r1",
        "firewall4": "2024.05.21~4c01d1eb-r1",
        "adblock": "4.2.2-r5",
        "uboot-envtools": "2024.07-r1",
        "dnsmasq": "2.90-r3",
        "procd": "2024.07.07~f230c117-r1",
        "usbutils": "017-r1",
        "kmod-crypto-acompress": "6.6.48-r1",
        "kmod-lib-crc32c": "6.6.48-r1",
        "lsblk": "2.40.2-r1",
        "block-mount": "2024.07.14~408c2cc4-r1",
        "kmod-nft-nat": "6.6.48-r1",
        "kmod-usb3": "6.6.48-r1",
        "libmbedtls": "3.6.0-r1",
        "libmnl": "1.0.5-r1",
        "jansson": "2.14-r3",
        "odhcp6c": "2023.05.12~bcd28363-r20",
        "uclient-fetch": "2024.04.19~e8780fa7-r1",
        "uci": "2023.08.10~5781664d-r1",
        "kmod-fs-ext4": "6.6.48-r1",
        "dropbear": "2024.85-r1",
        "mwlwifi-firmware-88w8964": "2024.04.19~a737d348-r1",
        "libnftnl": "1.2.6-r1",
        "mtd": "26",
        "odhcpd-ipv6only": "2024.05.08~a2988231-r1",
        "procd-seccomp": "2024.07.07~f230c117-r1",
        "libjson-script": "2024.03.29~eb9bcb64-r1",
        "kmod-mwifiex-sdio": "6.6.48.6.9.9-r1",
        "kmod-nf-log": "6.6.48-r1",
        "sudo": "1.9.15_p5-r1",
        "urandom-seed": "3",
        "ppp": "2.4.9_git20210104-r5",
        "kmod-gpio-button-hotplug": "6.6.48-r3",
        "logd": "2024.04.26~85f10530-r1",
        "kmod-nf-log6": "6.6.48-r1",
        "luci-app-attendedsysupgrade": "24.310.66693~1c61a66",
        "kmod-mwlwifi": "6.6.48.2024.04.19~a737d348-r1",
        "e2fsprogs": "1.47.0-r2",
        "kmod-ppp": "6.6.48-r1",
        "kmod-nft-offload": "6.6.48-r1",
        "mdadm": "4.3-r1",
        "luci-app-minidlna": "24.148.86114~d174bb8",
        "kmod-nf-conntrack": "6.6.48-r1",
        "kmod-md-raid0": "6.6.48-r1",
        "ca-bundle": "20240203-r1",
        "libuclient": "2024.04.19~e8780fa7-r1",
        "bind-dig": "9.20.0-r1",
        "luci": "23.051.66410~a505bb1",
        "owut": "2024.10.27~8c7e42f0-r1",
        "fdisk": "2.40.2-r1",
        "kmod-nf-conntrack6": "6.6.48-r1",
        "kmod-lib-lzo": "6.6.48-r1",
        "blkid": "2.40.2-r1",
        "luci-app-transmission": "24.148.86114~d174bb8",
        "kmod-nft-core": "6.6.48-r1",
        "transmission-web": "4.0.6-r1",
        "gdisk": "1.0.10-r1",
        "urngd": "2023.11.01~44365eb1-r1",
        "kmod-slhc": "6.6.48-r1",
        "ppp-mod-pppoe": "2.4.9_git20210104-r5"
    },
    "profile": "linksys,wrt3200acm",
    "target": "mvebu/cortexa9",
    "version": "SNAPSHOT",
    "diff_packages": true,
    "filesystem": "squashfs"
}
```
Error:
```
make: *** No rule to make target 'info'.  Stop.
```

Same on the site (with default package set):
[image]

While with `owut upgrade`:
```
......
  luci-app-opkg                   24.191.51591~bff2a43                       missing to-version
......
1 packages missing in target version, cannot upgrade
151 packages are out-of-date

ERROR: Update checks reveal errors, can't proceed
```

Any help is highly appreciated :) Great thanks in advance.

nereocystis, 2024-11-06:
My error is slightly different.
Server response: Unsupported profile: linksys,wrt3200acm
{
    "url": "https://sysupgrade.openwrt.org",
    "revision": "r16495-bf0c965af0",
    "advanced_mode": "1",
    "branch": "21.02",
    "efi": null,
    "profile": "linksys,wrt3200acm",
    "target": "mvebu/cortexa9",
    "version": "22.03.7",
    "packages": [
        "base-files",
        "busybox",
        "ca-bundle",
        "cgi-io",
        "diffutils",
        "dnsmasq",
        "dropbear",
        "firewall",
        "fstools",
        "fwtool",
        "getrandom",
        "hostapd-common",
        "ip6tables",
        "iptables",
        "iwinfo",
        "jshn",
        "jsonfilter",
        "kernel",
        "kmod-btmrvl",
        "kmod-gpio-button-hotplug",
        "kmod-ip6tables",
        "kmod-ipt-conntrack",
        "kmod-ipt-core",
        "kmod-ipt-nat",
        "kmod-ipt-offload",
        "kmod-lib-crc-ccitt",
        "kmod-mwlwifi",
        "kmod-nf-conntrack",
        "kmod-nf-conntrack6",
        "kmod-nf-flow",
        "kmod-nf-ipt",
        "kmod-nf-ipt6",
        "kmod-nf-nat",
        "kmod-nf-reject",
        "kmod-nf-reject6",
        "kmod-ppp",
        "kmod-pppoe",
        "kmod-pppox",
        "kmod-slhc",
        "libblobmsg-json",
        "libip4tc",
        "libip6tc",
        "libiwinfo",
        "libiwinfo-data",
        "libiwinfo-lua",
        "libjson-c",
        "libjson-script",
        "liblua",
        "liblucihttp",
        "liblucihttp-lua",
        "libnl-tiny",
        "libubox",
        "libubus",
        "libubus-lua",
        "libuci",
        "libuclient",
        "libustream-wolfssl",
        "libxtables",
        "logd",
        "lua",
        "luci",
        "luci-app-advanced-reboot",
        "luci-app-attendedsysupgrade",
        "luci-app-firewall",
        "luci-app-opkg",
        "luci-base",
        "luci-lib-base",
        "luci-lib-ip",
        "luci-lib-jsonc",
        "luci-lib-nixio",
        "luci-mod-admin-full",
        "luci-mod-network",
        "luci-mod-status",
        "luci-mod-system",
        "luci-proto-ipv6",
        "luci-proto-ppp",
        "luci-ssl",
        "luci-theme-bootstrap",
        "luci-theme-material",
        "luci-theme-openwrt",
        "luci-theme-openwrt-2020",
        "mtd",
        "mwlwifi-firmware-88w8964",
        "netifd",
        "odhcp6c",
        "odhcpd-ipv6only",
        "openssh-sftp-server",
        "openwrt-keyring",
        "opkg",
        "ppp",
        "ppp-mod-pppoe",
        "procd",
        "px5g-wolfssl",
        "rpcd",
        "rpcd-mod-file",
        "rpcd-mod-iwinfo",
        "rpcd-mod-luci",
        "rpcd-mod-rrdns",
        "tcpdump",
        "ubi-utils",
        "uboot-envtools",
        "ubox",
        "ubus",
        "ubusd",
        "uci",
        "uclient-fetch",
        "uhttpd",
        "uhttpd-mod-ubus",
        "urandom-seed",
        "urngd",
        "usign",
        "wireless-regdb",
        "wpad-basic-wolfssl"
    ],
    "diff_packages": true,
    "filesystem": "squashfs",
    "client": "luci/git-22.210.54423-3b0b325"
}

nereocystis, 2024-11-07:
And the auc output is:
> # auc -n -b 22.03
> auc/0.3.1-1
> Server:    https://sysupgrade.openwrt.org
> Running:   21.02.2 r16495-bf0c965af0 on mvebu/cortexa9 (linksys,wrt3200acm)
> Invalid argument (22)
> root@havenbmedia-3200:~# auc -n -b 22.03 -B 22.03.7
> auc/0.3.1-1
> Server:    https://sysupgrade.openwrt.org
> Running:   21.02.2 r16495-bf0c965af0 on mvebu/cortexa9 (linksys,wrt3200acm)
> Invalid argument (22)

nereocystis, 2024-11-09:
In my case, I was trying to get to 22.03.7, but the most recent release on that branch is 22.03.2. And that doesn't seem to be available on the server.
If I try 23.05.5 instead, I receive a different set of errors, which is probably fixable by adjusting the packages.
{
    "url": "https://sysupgrade.openwrt.org",
    "revision": "r16495-bf0c965af0",
    "advanced_mode": "1",
    "branch": "21.02",
    "efi": null,
    "profile": "linksys,wrt3200acm",
    "target": "mvebu/cortexa9",
    "version": "23.05.5",
    "packages": [
        "auc",
        "base-files",
        "busybox",
        "ca-bundle",
        "cgi-io",
        "diffutils",
        "dnsmasq",
        "dropbear",
        "firewall",
        "fstools",
        "fwtool",
        "getrandom",
        "hostapd-common",
        "ip6tables",
        "iptables",
        "iwinfo",
        "jshn",
        "jsonfilter",
        "kernel",
        "kmod-btmrvl",
        "kmod-gpio-button-hotplug",
        "kmod-ip6tables",
        "kmod-ipt-conntrack",
        "kmod-ipt-core",
        "kmod-ipt-nat",
        "kmod-ipt-offload",
        "kmod-lib-crc-ccitt",
        "kmod-mwlwifi",
        "kmod-nf-conntrack",
        "kmod-nf-conntrack6",
        "kmod-nf-flow",
        "kmod-nf-ipt",
        "kmod-nf-ipt6",
        "kmod-nf-nat",
        "kmod-nf-reject",
        "kmod-nf-reject6",
        "kmod-ppp",
        "kmod-pppoe",
        "kmod-pppox",
        "kmod-slhc",
        "libblobmsg-json",
        "libip4tc",
        "libip6tc",
        "libiwinfo",
        "libiwinfo-data",
        "libiwinfo-lua",
        "libjson-c",
        "libjson-script",
        "liblua",
        "liblucihttp",
        "liblucihttp-lua",
        "libnl-tiny",
        "libubox",
        "libubus",
        "libubus-lua",
        "libuci",
        "libuclient",
        "libustream-wolfssl",
[... 78 lines cut ...]
Downloading https://downloads.openwrt.org/releases/23.05.5/packages/arm_cortex-a9_vfpv3-d16/luci/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/23.05.5/packages/arm_cortex-a9_vfpv3-d16/packages/Packages.gz
Updated list of available packages in /builder/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/root-mvebu/../../../../builder/dl/openwrt_packages
Downloading https://downloads.openwrt.org/releases/23.05.5/packages/arm_cortex-a9_vfpv3-d16/packages/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/23.05.5/packages/arm_cortex-a9_vfpv3-d16/routing/Packages.gz
Updated list of available packages in /builder/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/root-mvebu/../../../../builder/dl/openwrt_routing
Downloading https://downloads.openwrt.org/releases/23.05.5/packages/arm_cortex-a9_vfpv3-d16/routing/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/23.05.5/packages/arm_cortex-a9_vfpv3-d16/telephony/Packages.gz
Updated list of available packages in /builder/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/root-mvebu/../../../../builder/dl/openwrt_telephony
Downloading https://downloads.openwrt.org/releases/23.05.5/packages/arm_cortex-a9_vfpv3-d16/telephony/Packages.sig
Signature check passed.
Downloading file:packages/Packages
Updated list of available packages in /builder/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/root-mvebu/../../../../builder/dl/imagebuilder
Downloading file:packages/Packages.sig
Signature check passed.
Collected errors:
 * check_data_file_clashes: Package libustream-mbedtls20201210 wants to install file /builder/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/root-mvebu/lib/libustream-ssl.so
	But that file is already provided by package  * libustream-wolfssl20201210
 * opkg_install_cmd: Cannot install package luci-ssl.
make[2]: *** [Makefile:189: package_install] Error 255
make[1]: *** [Makefile:154: _call_manifest] Error 2
make: *** [Makefile:274: manifest] Error 2

nereocystis, 2024-11-09:
And the fireware selector does not support customizing installed packages for 22.03.2

> Unsupported version: 22.03.2