openwrt/asu #1075: owut: ERROR: Build failed with status 500; ASU server stderr = make: *** No rule to make target 'info'.  Stop.

pko76, 2024-11-07:
root@OpenWrt:~# owut upgrade
owut - OpenWrt Upgrade Tool 2024.10.27~8c7e42f0-r1 (/usr/bin/owut)
ASU-Server     https://sysupgrade.openwrt.org
Upstream       https://downloads.openwrt.org
Target         ipq40xx/generic
Profile        zte_mf286d
Package-arch   arm_cortex-a7_neon-vfpv4
Root-FS-type   squashfs
Sys-type       sysupgrade
Version-from   SNAPSHOT r28019-685e979d47 (kernel 6.6.59)
Version-to     SNAPSHOT r28021-e29bc67f9d (kernel 6.6.59)
Build-FS-type  squashfs
Build-at       2024-11-06T10:41:20Z (~35 hours ago)
Image-prefix   openwrt-ipq40xx-generic-zte_mf286d
Image-URL      https://downloads.openwrt.org/snapshots/targets/ipq40xx/generic
Image-file     openwrt-ipq40xx-generic-zte_mf286d-squashfs-sysupgrade.bin
Installed      288 packages
Top-level      106 packages
Default         41 packages
User-installed  71 packages (top-level only)

Package version changes:
  base-files                          1630~685e979d47                            1630~e29bc67f9d
1 packages are out-of-date

Default package analysis:
  Default                             Provided-by
  nftables                            nftables-json

There are currently package build failures for SNAPSHOT arm_cortex-a7_neon-vfpv4:
  Feed: packages
    fwupd                             Wed Nov  6 04:45:33 2024 - not installed
    micropython-lib                   Wed Nov  6 02:36:07 2024 - not installed
    umurmur                           Wed Nov  6 03:23:33 2024 - not installed
  Feed: telephony
    freetdm                           Wed Nov  6 05:02:35 2024 - not installed
Failures don't affect this device, details at
  https://downloads.openwrt.org/snapshots/faillogs/arm_cortex-a7_neon-vfpv4/

Request:
  Version r28021-e29bc67f9d (kernel 6.6.59)
Request hash:
  aebdc3e7785cb3ae5ede25104f4a9684
--
Status:   init
Progress:   0s total =   0s in queue +   0s in build

Build failed in   0s total =   0s in queue +   0s to build:
ASU server error =
Traceback (most recent call last):
  File "/usr/local/lib/python3.12/site-packages/rq/worker.py", line 1430, in perform_job
    rv = job.perform()
         ^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/rq/job.py", line 1280, in perform
    self._result = self._execute()
                   ^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/rq/job.py", line 1317, in _execute
    result = self.func(*self.args, **self.kwargs)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/asu/build.py", line 176, in build
    version_code = re.search('Current Revision: "(r.+)"', job.meta["stdout"]).group(1)
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
AttributeError: 'NoneType' object has no attribute 'group'

ASU server stderr =
make: *** No rule to make target 'info'.  Stop.

ERROR: Build failed with status 500
The above errors are often due to the upgrade server lagging behind the
build server, first suggestion is to wait a while and try again.

tlaurion, 2024-11-07:
@aparcar all recent issues are reporting about the same issue. This title.

Any hint of where it's coming from and fixing things so it's retrocompatible and firneares are upgradeable again?

Soromeister, 2024-11-08:
Same for me on AX9000. It just fails at init with the same error.

Request headers:
```
Content-Length: 1394
Content-Type: application/json
Host: sysupgrade.openwrt.org
User-Agent: HTTPie
```

Response:
```
HTTP/1.1 500 Internal Server Error
Access-Control-Allow-Credentials: true
Access-Control-Allow-Headers: *
Access-Control-Allow-Methods: POST, GET, OPTIONS
Access-Control-Allow-Origin: *
Access-Control-Expose-Headers: Authorization
Connection: close
Content-Length: 2234
Content-Type: application/json
Date: Fri, 08 Nov 2024 08:06:36 GMT
Server: Caddy
Vary: Origin
X-Imagebuilder-Status: failed
X-Queue-Position: 0

{"detail":"init","imagebuilder_status":"validate_revision","request":{"distro":"openwrt","version":"SNAPSHOT","version_code":"r28022-6691ff86b7","target":"qualcommax/ipq807x","profile":"xiaomi_ax9000","packages":["ath10k-firmware-qca9887","ath11k-firmware-ipq8074","ath11k-firmware-qcn9074","base-files","busybox","ca-bundle","coreutils","coreutils-ls","coreutils-uptime","curl","dnsmasq","dropbear","e2fsprogs","firewall4","gre","htop","ipq-wifi-xiaomi_ax9000","iw","iwinfo","jq","kernel","kmod-ath10k-ct","kmod-ath11k-ahb","kmod-ath11k-pci","kmod-fs-ext4","kmod-gpio-button-hotplug","kmod-ieee802154-6lowpan","kmod-leds-gpio","kmod-mac802154","kmod-mppe","kmod-phy-aquantia","kmod-qca-nss-dp","kmod-sit","kmod-udptunnel6","kmod-usb-dwc3","kmod-usb-dwc3-qcom","kmod-usb3","libc","libustream-wolfssl","libwolfsslcpu-crypto5.7.2.e624513f","logd","losetup","luci","luci-app-attendedsysupgrade","luci-app-irqbalance","mtd","ncat-full","odhcp6c","odhcpd-ipv6only","opkg","ppp","ppp-mod-pppoe","procd","procd-seccomp","procd-ujail","uboot-envtools","uci","urandom-seed","urngd","wget-ssl","wpad-wolfssl"],"packages_versions":{},"diff_packages":true,"defaults":null,"rootfs_size_mb":null,"repositories":{},"repository_keys":[],"client":null},"stdout":"","stderr":"make: *** No rule to make target 'info'.  Stop.\n","status":500,"error":"Traceback (most recent call last):\n  File \"/usr/local/lib/python3.12/site-packages/rq/worker.py\", line 1430, in perform_job\n    rv = job.perform()\n         ^^^^^^^^^^^^^\n  File \"/usr/local/lib/python3.12/site-packages/rq/job.py\", line 1280, in perform\n    self._result = self._execute()\n                   ^^^^^^^^^^^^^^^\n  File \"/usr/local/lib/python3.12/site-packages/rq/job.py\", line 1317, in _execute\n    result = self.func(*self.args, **self.kwargs)\n             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^\n  File \"/app/asu/build.py\", line 176, in build\n    version_code = re.search('Current Revision: \"(r.+)\"', job.meta[\"stdout\"]).group(1)\n                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^\nAttributeError: 'NoneType' object has no attribute 'group'\n","enqueued_at":"2024-11-08T08:04:03.268198","request_hash":"21463dc7b6650de363c36ce42ffdd2bc"}
```

efahl, 2024-11-12:
All part of the `apk` transition, should be fixed in a week or two.  You can use imagebuilders in the interim, they appear to be working for all the platforms I've tested thus far.

pko76, 2024-11-12:
@efahl 
Everything is totally messed  today. 
Trying to gest any custom build for MF286D based on SNAPSHOT (r28062-be3e7a84bd) fails:

init
STDERR
make: *** No rule to make target 'info'.  Stop.

Trying owut, fails:
owut - OpenWrt Upgrade Tool 2024.10.27~8c7e42f0-r1 (/usr/bin/owut)
ERROR: Response status 500 while downloading
  https://sysupgrade.openwrt.org/json/v1/snapshots/packages/arm_cortex-a7_neon-vfpv4-index.json
ASU-Server     https://sysupgrade.openwrt.org
Upstream       https://downloads.openwrt.org
Target         ipq40xx/generic
Profile        zte_mf286d
Package-arch   arm_cortex-a7_neon-vfpv4
Root-FS-type   squashfs
Sys-type       sysupgrade
Version-from   SNAPSHOT r28021-e29bc67f9d (kernel 6.6.59)
Version-to     SNAPSHOT r28062-be3e7a84bd (kernel unknown)
Build-FS-type  squashfs
Build-at       2024-11-12T02:31:06Z (~13 hours ago)
Image-prefix   openwrt-ipq40xx-generic-zte_mf286d
Image-URL      https://downloads.openwrt.org/snapshots/targets/ipq40xx/generic
Image-file     openwrt-ipq40xx-generic-zte_mf286d-squashfs-sysupgrade.bin
Installed      289 packages
Top-level      111 packages
Default         37 packages
User-installed  82 packages (top-level only)

Package version changes:
  adb                                 5.0.2~6fe92d1a-r3                          missing to-version
  ath10k-board-qca4019                20241017-r1                                missing to-version

...
287 packages missing in target version, cannot upgrade
287 packages are out-of-date

ERROR: Update checks reveal errors, can't proceed

Nothing currently works. It is not possible to get new version via attendedsysupgrade:

Building package index...
ERROR: unable to select packages:
  libffmpeg (virtual):
    note: please select one of the 'provided by'
          packages explicitly
    provided by: libffmpeg-audio-dec-6.1.2-r1
                 libffmpeg-full
                 libffmpeg-mini-6.1.2-r1
    required by: minidlna-1.3.3-r1[libffmpeg]
  libjpeg (virtual):
    note: please select one of the 'provided by'
          packages explicitly
    provided by: libjpeg-turbo
    required by: minidlna-1.3.3-r1[libjpeg]
make[2]: *** [Makefile:224: package_install] Error 5
make[1]: *** [Makefile:165: _call_manifest] Error 2
make: *** [Makefile:339: manifest] Error 2

closed 2024-11-22 by pko76