openwrt/asu #1077: Error: Impossible package selection

Norbert-Rotorwerk, 2024-11-10:
after owut upgrade here are the output:

owut - OpenWrt Upgrade Tool 2024.10.27~8c7e42f0-r1 (/usr/bin/owut)
ASU-Server     https://sysupgrade.openwrt.org
Upstream       https://downloads.openwrt.org
Target         mediatek/filogic
Profile        openwrt_one
Package-arch   aarch64_cortex-a53
Root-FS-type   squashfs
Sys-type       sysupgrade
Version-from   SNAPSHOT r27935-bcd95cb9c4 (kernel 6.6.58)
Version-to     SNAPSHOT r28041-a7bc6bf7db (kernel 6.6.59)
Build-FS-type  squashfs
Build-at       2024-11-09T16:31:59Z (~18 hours ago)
Image-prefix   openwrt-mediatek-filogic-openwrt_one
Image-URL      https://downloads.openwrt.org/snapshots/targets/mediatek/filogic
Image-file     openwrt-mediatek-filogic-openwrt_one-squashfs-sysupgrade.itb
Installed      245 packages
Top-level      145 packages
Default         42 packages
User-installed 109 packages (top-level only)

Package version changes:
  base-files                          1630~bcd95cb9c4                            1630~a7bc6bf7db
  ddns-scripts                        2.8.2-r50                                  2.8.2-r51
  ddns-scripts-services               2.8.2-r50                                  2.8.2-r51
  kmod-crypto-acompress               6.6.58-r1                                  6.6.59-r1
  kmod-crypto-aead                    6.6.58-r1                                  6.6.59-r1
  kmod-crypto-authenc                 6.6.58-r1                                  6.6.59-r1
  kmod-crypto-ccm                     6.6.58-r1                                  6.6.59-r1
  kmod-crypto-crc32c                  6.6.58-r1                                  6.6.59-r1
  kmod-crypto-ctr                     6.6.58-r1                                  6.6.59-r1
  kmod-crypto-des                     6.6.58-r1                                  6.6.59-r1
  kmod-crypto-gf128                   6.6.58-r1                                  6.6.59-r1
  kmod-crypto-ghash                   6.6.58-r1                                  6.6.59-r1
  kmod-crypto-hmac                    6.6.58-r1                                  6.6.59-r1
  kmod-crypto-hw-safexcel             6.6.58-r1                                  6.6.59-r1
  kmod-crypto-manager                 6.6.58-r1                                  6.6.59-r1
  kmod-crypto-md5                     6.6.58-r1                                  6.6.59-r1
  kmod-crypto-seqiv                   6.6.58-r1                                  6.6.59-r1
  kmod-crypto-sha1                    6.6.58-r1                                  6.6.59-r1
  kmod-crypto-sha256                  6.6.58-r1                                  6.6.59-r1
  kmod-crypto-sha512                  6.6.58-r1                                  6.6.59-r1
  kmod-gpio-button-hotplug            6.6.58-r5                                  6.6.59-r5
  kmod-hwmon-core                     6.6.58-r1                                  6.6.59-r1
  kmod-i2c-core                       6.6.58-r1                                  6.6.59-r1
  kmod-leds-gpio                      6.6.58-r1                                  6.6.59-r1
  kmod-lib-crc-ccitt                  6.6.58-r1                                  6.6.59-r1
  kmod-lib-crc32c                     6.6.58-r1                                  6.6.59-r1
  kmod-lib-lzo                        6.6.58-r1                                  6.6.59-r1
  kmod-libphy                         6.6.58-r1                                  6.6.59-r1
  kmod-mac80211                       6.6.58.6.11.2-r1                           6.6.59.6.11.2-r1
  kmod-mt76-connac                    6.6.58.2024.10.11.1~ecca0e77-r1            6.6.59.2024.10.11.1~ecca0e77-r1
  kmod-mt76-core                      6.6.58.2024.10.11.1~ecca0e77-r1            6.6.59.2024.10.11.1~ecca0e77-r1
  kmod-mt7915e                        6.6.58.2024.10.11.1~ecca0e77-r1            6.6.59.2024.10.11.1~ecca0e77-r1
  kmod-mt7981-firmware                6.6.58.2024.10.11.1~ecca0e77-r1            6.6.59.2024.10.11.1~ecca0e77-r1
  kmod-nf-flow                        6.6.58-r1                                  6.6.59-r1
  kmod-nf-nat                         6.6.58-r1                                  6.6.59-r1
  kmod-nf-reject                      6.6.58-r1                                  6.6.59-r1
  kmod-nf-reject6                     6.6.58-r1                                  6.6.59-r1
  kmod-nfnetlink                      6.6.58-r1                                  6.6.59-r1
  kmod-nft-fib                        6.6.58-r1                                  6.6.59-r1
  kmod-nft-nat                        6.6.58-r1                                  6.6.59-r1
  kmod-nvme                           6.6.58-r1                                  6.6.59-r1
  kmod-phy-airoha-en8811h             6.6.58-r1                                  6.6.59-r1
  kmod-phy-aquantia                   6.6.58-r1                                  6.6.59-r1
  kmod-pppoe                          6.6.58-r1                                  6.6.59-r1
  kmod-pppox                          6.6.58-r1                                  6.6.59-r1
  kmod-rtc-pcf8563                    6.6.58-r1                                  6.6.59-r1
  kmod-thermal                        6.6.58-r1                                  6.6.59-r1
[... 105 lines cut ...]
Downloading https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/packages/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/routing/Packages.gz
Updated list of available packages in /builder/build_dir/target-aarch64_cortex-a53_musl/root-mediatek/../../../../builder/dl/openwrt_routing
Downloading https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/routing/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/telephony/Packages.gz
Updated list of available packages in /builder/build_dir/target-aarch64_cortex-a53_musl/root-mediatek/../../../../builder/dl/openwrt_telephony
Downloading https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/telephony/Packages.sig
Signature check passed.
Downloading file:packages/Packages
Updated list of available packages in /builder/build_dir/target-aarch64_cortex-a53_musl/root-mediatek/../../../../builder/dl/imagebuilder
Downloading file:packages/Packages.sig
Signature check passed.
/builder/staging_dir/host/bin/opkg: the ``install'' command requires at least one argument
make[2]: *** [Makefile:224: package_install] Error 1
make[1]: *** [Makefile:165: _call_manifest] Error 2
make: *** [Makefile:339: manifest] Error 2

ERROR: Build failed with status 500
The above errors are often due to the upgrade server lagging behind the
build server, first suggestion is to wait a while and try again.

Regards
Norbert

efahl, 2024-11-11:
That one is fixed.  Once the apk builds go through, we need to fix the missing package index on the upstream site and fix the ubus call on clients.  Could be a week or two before owut works again...

lorand-horvath, 2024-11-11:
> Could be a week or two before owut works again...

@efahl Thanks for responding. I'm not using owut at the moment, but the firmware-selector site. Is that in the same situation as owut? Or can I expect it to get fixed quicker than 1-2 weeks?

efahl, 2024-11-11:
FS doesn't rely on either of the two bugs I mentioned, so as soon as the apk builds are done, it should work.  That might take 3-4 days though, as it requires a phase1 rebuild of all the targets/subtargets/profiles and then all of the packages in phase2.