openwrt/asu #1084: APK not being included by default

robimarko, 2024-11-16:
Ever since the move to making APK the default package manager in OpenWrt ASU has been generating images without APK nor OPKG by default.

For example, I have requested a zbtlink zbt-z800ax image with the default package set (So no changes) and it does not include APK itself after unpacking:
```
[robimarko@fedora squashfs-root]$ find . -name "apk"
./etc/apk
./lib/apk
./run/apk
```

Only after adding the apk-mbedtls (The default APK version) to the package list it gets included:
```
[robimarko@fedora squashfs-root]$ find . -name "apk"
./etc/apk
./lib/apk
./run/apk
./usr/bin/apk
```

robimarko, 2024-11-16:
I did some digging and the root cause is that APK is not being included in `default_packages` in `profiles.json`

robimarko, 2024-11-16:
And that seems to boil down to the fact that ever since: https://github.com/openwrt/openwrt/commit/44598c233dd9a676bc34666968f33f8b9ff4dd0c

OPKG nor APK are ever added to the `DEFAULT_PACKAGES` list

robimarko, 2024-11-16:
This is even worse, `procd` as well as `procd-seccomp` has been dropped from `DEFAULT_PACKAGES` so procd is only getting included by luck as its a dependency of `procd-ujail` which was left in target.mk and `DEFAULT_PACKAGES`

brada4, 2024-11-17:
Thats the spirit of QA

brada4, 2024-11-17:
ujail for dnsmasq

robimarko, 2024-11-17:
@KA2107 And that works for imagebuilder, but not for ASU which assembles the package list based of default_packages in profiles.json

closed 2024-11-17 by openwrt-bot

Catfriend1, 2024-11-17:
How long will it approximately take, until ASU will produce correct images - so when could I retry ?