openwrt/asu #1088: Error building the firmware image

Megachip, 2024-11-18:
Server response: Error: Impossible package selection

[Please report the error message and request](https://github.com/openwrt/asu/issues)

### Request Data:
```
{
    "url": "https://sysupgrade.openwrt.org",
    "branch": "SNAPSHOT",
    "revision": "r27998-591272d197",
    "efi": null,
    "advanced_mode": "0",
    "rebuilder": [
        "http://asu.hauke-m.de:8000",
        "https://asu.aparcar.org"
    ],
    "request_hash": "",
    "sha256_unsigned": "",
    "client": "luci/24.290.85088~46fb560",
    "packages": {
        "kmod-rtlwifi-btcoexist": "6.6.58.6.11.2-r1",
        "luci-proto-wireguard": "24.299.70893~f1b302d",
        "luci-app-statistics": "24.298.47125~4d75286",
        "libc": "1.2.5-r4",
        "kmod-usb-net-asix-ax88179": "6.6.58-r1",
        "opkg": "2024.10.16~38eccbb1-r1",
        "luci-app-vnstat2": "23.340.05065~c0c3790",
        "busybox": "1.36.1-r2",
        "psmisc": "23.7-r1",
        "ip-full": "6.11.0-r1",
        "kmod-rtw88-8821cu": "6.6.58.6.11.2-r1",
        "kmod-rt2800-usb": "6.6.58.6.11.2-r1",
        "kmod-lib-crc-ccitt": "6.6.58-r1",
        "getrandom": "2024.04.26~85f10530-r1",
        "kmod-pppoa": "6.6.58-r1",
        "kmod-pppoe": "6.6.58-r1",
        "kmod-rtw88-8822cu": "6.6.58.6.11.2-r1",
        "iw-full": "6.9-r1",
        "kmod-veth": "6.6.58-r1",
        "procd-ujail": "2024.10.20~ef3ab8bc-r1",
        "base-files": "1630~591272d197",
        "luci-mod-dsl": "24.114.50270~6c0b9a9",
        "ltq-vdsl-vr9-app": "4.17.18.6-r7",
        "libustream-mbedtls": "2024.07.28~99bd3d2b-r1",
        "firewall4": "2024.05.21~4c01d1eb-r1",
        "dnsmasq": "2.90-r3",
        "procd": "2024.10.20~ef3ab8bc-r1",
        "usbutils": "017-r1",
        "kmod-crypto-acompress": "6.6.58-r1",
        "kmod-rtl8xxxu": "6.6.58.6.11.2-r1",
        "luci-app-ddns": "24.255.83885~f8ef86b",
        "kmod-usb3": "6.6.58-r1",
        "odhcp6c": "2023.05.12~bcd28363-r20",
        "uci": "2023.08.10~5781664d-r1",
        "dsl-vrx200-firmware-xdsl-a": "05.08.01.08.01.06.05.08.00.11.01.01-r1",
        "dropbear": "2024.85-r1",
        "curl": "8.10.1-r1",
        "mtd": "26",
        "odhcpd-ipv6only": "2024.05.08~a2988231-r1",
        "rtl8188eu-firmware": "20241017-r1",
        "procd-seccomp": "2024.10.20~ef3ab8bc-r1",
        "wpa-supplicant": "2024.09.15~5ace39b0-r1",
        "libjson-script": "2024.03.29~eb9bcb64-r1",
        "eject": "2.40.2-r1",
        "usb-modeswitch": "2022.02.24~3c8595a4-r1",
        "kmod-ltq-ptm-vr9": "6.6.58-r4",
        "urandom-seed": "3",
        "ppp": "2.5.1-r1",
        "kmod-leds-gpio": "6.6.58-r1",
        "kmod-gpio-button-hotplug": "6.6.58-r5",
        "logd": "2024.04.26~85f10530-r1",
        "kmod-rtw89-8852ae": "6.6.58.6.11.2-r1",
        "kmod-rtl8192cu": "6.6.58.6.11.2-r1",
        "kmod-rtl8192du": "6.6.58.6.11.2-r1",
        "luci-app-attendedsysupgrade": "24.290.85088~46fb560",
        "kmod-rtw89-8852ce": "6.6.58.6.11.2-r1",
        "drill": "1.8.3-r1",
        "kmod-ltq-atm-vr9": "6.6.58-r3",
        "kmod-ppp": "6.6.58-r1",
        "luci-app-wol": "24.148.86114~d174bb8",
        "luci-app-upnp": "24.148.86114~d174bb8",
        "hostapd": "2024.09.15~5ace39b0-r1",
        "git": "2.46.2-r1",
        "rtl8192eu-firmware": "20241017-r1",
        "ca-bundle": "20240203-r1",
        "libuclient": "2024.10.22~88ae8f20-r1",
        "luci": "24.297.79519~bcd13b9",
        "owut": "2024.10.25~8c8907c6-r1",
        "hwinfo": "21.71-r4",
        "fritz-tffs": "2",
        "kmod-lib-lzo": "6.6.58-r1",
        "kernel": "6.6.58~83e37076b9d5ba94ff15ab398428b2c2-r1",
        "git-http": "2.46.2-r1",
        "kmod-ltq-deu-vr9": "6.6.58-r45",
        "ltq-vdsl-vr9-vectoring-fw-installer": "6.8.6-r4",
        "iperf3": "3.17.1-r3",
        "dsl-vrx200-firmware-xdsl-b-patch": "05.08.01.08.01.06.05.08.00.11.01.01-r1",
        "urngd": "2023.11.01~44365eb1-r1",
        "ppp-mod-pppoa": "2.5.1-r1",
        "ppp-mod-pppoe": "2.5.1-r1"
    },
    "profile": "avm,fritz3490",
    "target": "lantiq/xrx200",
    "version": "SNAPSHOT",
    "diff_packages": true,
    "filesystem": "squashfs"
}
```
### STDERR:
```
Generate local signing keys...
WARNING: can't open config file: /builder/shared-workdir/build/staging_dir/host/etc/ssl/openssl.cnf
WARNING: can't open config file: /builder/shared-workdir/build/staging_dir/host/etc/ssl/openssl.cnf
read EC key
writing EC key
fetch https://downloads.openwrt.org/snapshots/targets/lantiq/xrx200/packages/packages.adb
fetch https://downloads.openwrt.org/snapshots/packages/mips_24kc/base/packages.adb
fetch https://downloads.openwrt.org/snapshots/targets/lantiq/xrx200/kmods/6.6.61-1-16eebff70d52859e7a62e0ddb8650de1/packages.adb
fetch https://downloads.openwrt.org/snapshots/packages/mips_24kc/luci/packages.adb
fetch https://downloads.openwrt.org/snapshots/packages/mips_24kc/packages/packages.adb
fetch https://downloads.openwrt.org/snapshots/packages/mips_24kc/routing/packages.adb
fetch https://downloads.openwrt.org/snapshots/packages/mips_24kc/telephony/packages.adb
WARNING: opening packages/packages.adb: No such file or directory
Package list missing or not up-to-date, generating it.

Building package index...
ERROR: unable to select packages:
  opkg (no such package):
    required by: world[opkg]
  libjpeg (virtual):
    note: please select one of the 'provided by'
          packages explicitly
    provided by: libjpeg-turbo
    required by: libgd-2.3.3-r1[libjpeg]
  libudev (virtual):
    note: please select one of the 'provided by'
          packages explicitly
    provided by: libudev-zero
    required by: usbutils-017-r1[libudev]
make[2]: *** [Makefile:225: package_install] Error 7
make[1]: *** [Makefile:166: _call_manifest] Error 2
make: *** [Makefile:340: manifest] Error 2

```

Megachip, 2024-11-18:
Maybe related to #1078 and [#16986](https://github.com/openwrt/openwrt/pull/16986) ... but gui requested a a new ;)

robimarko, 2024-11-18:
This is: https://github.com/openwrt/openwrt/issues/16976

https://github.com/openwrt/openwrt/pull/17008 should fix it

Megachip, 2024-11-25:
> > [openwrt/openwrt#17008](https://github.com/openwrt/openwrt/pull/17008) should fix it
> 
> so just give it some time. perfect.

```
Building package index...
ERROR: unable to select packages:
  opkg (no such package):
    required by: world[opkg]
make[2]: *** [Makefile:225: package_install] Error 1
make[1]: *** [Makefile:166: _call_manifest] Error 2
make: *** [Makefile:340: manifest] Error 2
```
so it now just lacking of world?

robimarko, 2024-11-25:
opkg is not an user installable package, its either you use OPKG or APK and we are now using APK by default so you dont need opkg at all

Megachip, 2024-11-25:
> opkg is not an user installable package, its either you use OPKG or APK and we are now using APK by default so you dont need opkg at all

just hit the sys upgrade button ;) seen that opkg was in manual requirements. Removing it fixed the build process.

robimarko, 2024-11-28:
LuCI package manager works for both apk and OPKG, all of the main tooling has been converted

Megachip, 2024-11-28:
> LuCI package manager works for both apk and OPKG, all of the main tooling has been converted

looks like both aren't installed anymore ... any chance to fix this remotely, eg via ssh?

robimarko, 2024-11-28:
It is for sure in the default package list unless you remove it, you can simply sysupgrade to an image containing it