openwrt/asu #1090: Error: Impossible package selection

gizmoscott, 2024-11-19:
Error building the firmware image
Server response: Error: Impossible package selection

[Please report the error message and request](https://github.com/openwrt/asu/issues)
Request Data:

{
    "url": "https://sysupgrade.openwrt.org",
    "revision": "r20265-f85a79bcb4",
    "advanced_mode": "1",
    "branch": "22.03",
    "efi": null,
    "profile": "zyxel,nbg6817",
    "target": "ipq806x/generic",
    "version": "23.05.5",
    "packages": [
        "ath10k-firmware-qca9984-ct",
        "base-files",
        "busybox",
        "ca-bundle",
        "ca-certificates",
        "cgi-io",
        "dnsmasq",
        "dropbear",
        "e2fsprogs",
        "firewall4",
        "fstools",
        "fwtool",
        "getrandom",
        "hostapd-common",
        "ip-tiny",
        "ip6tables-nft",
        "ipset",
        "iptables-mod-conntrack-extra",
        "iptables-mod-ipopt",
        "iptables-nft",
        "iw",
        "iwinfo",
        "jansson",
        "jshn",
        "jsonfilter",
        "kernel",
        "kmod-ata-ahci",
        "kmod-ata-ahci-platform",
        "kmod-ata-core",
        "kmod-ath",
        "kmod-ath10k-ct",
        "kmod-cfg80211",
        "kmod-crypto-aead",
        "kmod-crypto-ccm",
        "kmod-crypto-cmac",
        "kmod-crypto-crc32c",
        "kmod-crypto-ctr",
        "kmod-crypto-gcm",
        "kmod-crypto-gf128",
        "kmod-crypto-ghash",
        "kmod-crypto-hash",
        "kmod-crypto-hmac",
        "kmod-crypto-manager",
        "kmod-crypto-null",
        "kmod-crypto-rng",
        "kmod-crypto-seqiv",
        "kmod-crypto-sha256",
        "kmod-fs-ext4",
        "kmod-gpio-button-hotplug",
        "kmod-hwmon-core",
        "kmod-ipt-nat",
        "kmod-ipt-offload",
        "kmod-leds-gpio",
        "kmod-lib-crc-ccitt",
        "kmod-lib-crc32c",
        "kmod-mac80211",
        "kmod-nf-conntrack",
        "kmod-nf-conntrack6",
        "kmod-nf-flow",
        "kmod-nf-log",
        "kmod-nf-log6",
        "kmod-nf-nat",
        "kmod-nf-reject",
        "kmod-nf-reject6",
        "kmod-nfnetlink",
        "kmod-nft-core",
        "kmod-nft-fib",
        "kmod-nft-nat",
        "kmod-nft-offload",
        "kmod-nls-base",
        "kmod-phy-qcom-ipq806x-usb",
        "kmod-ppp",
        "kmod-pppoe",
        "kmod-pppox",
        "kmod-scsi-core",
        "kmod-slhc",
        "kmod-usb-core",
        "kmod-usb-dwc3",
        "kmod-usb-dwc3-qcom",
        "kmod-usb-ehci",
        "kmod-usb-ledtrig-usbport",
        "kmod-usb-ohci",
        "kmod-usb-xhci-hcd",
        "kmod-usb2",
        "kmod-usb3",
        "libblobmsg-json",
        "libip4tc",
        "libip6tc",
        "libiwinfo",
        "libiwinfo-data",
        "libiwinfo-lua",
        "libjson-c",
        "libjson-script",
        "liblua",
        "liblucihttp",
        "liblucihttp-lua",
        "libmnl",
        "libnftnl",
        "libnl-tiny",
        "libubox",
        "libubus",
        "libubus-lua",
        "libuci",
        "libuclient",
        "libucode",
        "libustream-wolfssl",
        "libwolfssl",
        "logd",
        "losetup",
        "lua",
        "luci",
        "luci-app-attendedsysupgrade",
        "luci-app-firewall",
        "luci-app-mwan3",
        "luci-app-opkg",
        "luci-base",
        "luci-compat",
        "luci-i18n-attendedsysupgrade-en",
        "luci-i18n-mwan3-en",
        "luci-lib-base",
        "luci-lib-ip",
        "luci-lib-jsonc",
        "luci-lib-nixio",
        "luci-mod-admin-full",
        "luci-mod-network",
        "luci-mod-status",
        "luci-mod-system",
        "luci-proto-ipv6",
        "luci-proto-ppp",
        "luci-ssl",
        "luci-theme-bootstrap",
        "mtd",
        "netifd",
        "nftables-json",
        "odhcp6c",
        "odhcpd-ipv6only",
        "openssh-sftp-server",
        "openssl-util",
        "openwrt-keyring",
        "opkg",
        "ppp",
        "ppp-mod-pppoe",
        "procd",
        "procd-seccomp",
        "procd-ujail",
        "px5g-wolfssl",
        "rpcd",
        "rpcd-mod-file",
        "rpcd-mod-iwinfo",
        "rpcd-mod-luci",
        "rpcd-mod-rrdns",
        "swconfig",
        "ubi-utils",
        "uboot-envtools",
        "ubox",
        "ubus",
        "ubusd",
        "uci",
        "uclient-fetch",
        "ucode",
        "ucode-mod-fs",
        "ucode-mod-ubus",
        "ucode-mod-uci",
        "uhttpd",
        "uhttpd-mod-ubus",
        "urandom-seed",
        "urngd",
        "usign",
        "wireless-regdb",
        "wpad-basic-wolfssl",
        "zlib"
    ],
    "diff_packages": true,
    "filesystem": "squashfs",
    "client": "luci/git-23.093.42303-58b861d"
}
STDERR:
Generate local signing keys...
Generate local certificate...
Package list missing or not up-to-date, generating it.

Building package index...
Downloading https://downloads.openwrt.org/releases/23.05.5/targets/ipq806x/generic/packages/Packages.gz
Updated list of available packages in /builder/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/root-ipq806x/../../../../builder/dl/openwrt_core
Downloading https://downloads.openwrt.org/releases/23.05.5/targets/ipq806x/generic/packages/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/23.05.5/packages/arm_cortex-a15_neon-vfpv4/base/Packages.gz
Updated list of available packages in /builder/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/root-ipq806x/../../../../builder/dl/openwrt_base
Downloading https://downloads.openwrt.org/releases/23.05.5/packages/arm_cortex-a15_neon-vfpv4/base/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/23.05.5/packages/arm_cortex-a15_neon-vfpv4/luci/Packages.gz
Updated list of available packages in /builder/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/root-ipq806x/../../../../builder/dl/openwrt_luci
Downloading https://downloads.openwrt.org/releases/23.05.5/packages/arm_cortex-a15_neon-vfpv4/luci/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/23.05.5/packages/arm_cortex-a15_neon-vfpv4/packages/Packages.gz
Updated list of available packages in /builder/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/root-ipq806x/../../../../builder/dl/openwrt_packages
Downloading https://downloads.openwrt.org/releases/23.05.5/packages/arm_cortex-a15_neon-vfpv4/packages/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/23.05.5/packages/arm_cortex-a15_neon-vfpv4/routing/Packages.gz
Updated list of available packages in /builder/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/root-ipq806x/../../../../builder/dl/openwrt_routing
Downloading https://downloads.openwrt.org/releases/23.05.5/packages/arm_cortex-a15_neon-vfpv4/routing/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/23.05.5/packages/arm_cortex-a15_neon-vfpv4/telephony/Packages.gz
Updated list of available packages in /builder/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/root-ipq806x/../../../../builder/dl/openwrt_telephony
Downloading https://downloads.openwrt.org/releases/23.05.5/packages/arm_cortex-a15_neon-vfpv4/telephony/Packages.sig
Signature check passed.
Downloading file:packages/Packages
Updated list of available packages in /builder/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/root-ipq806x/../../../../builder/dl/imagebuilder
Downloading file:packages/Packages.sig
Signature check passed.
Collected errors:
 * opkg_install_cmd: Cannot install package luci-i18n-attendedsysupgrade-en.
 * opkg_install_cmd: Cannot install package luci-i18n-mwan3-en.
 * check_data_file_clashes: Package libustream-mbedtls20201210 wants to install file /builder/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/root-ipq806x/lib/libustream-ssl.so
	But that file is already provided by package  * libustream-wolfssl20201210
 * opkg_install_cmd: Cannot install package luci-ssl.
make[2]: *** [Makefile:189: package_install] Error 255
make[1]: *** [Makefile:154: _call_manifest] Error 2
make: *** [Makefile:274: manifest] Error 2
Close

robimarko, 2024-11-19:
You cannot have multiple libustream libs installed, simply add `-libustream-mbedtls` to the requested package list, this is not an ASU bug

gizmoscott, 2024-11-19:
> Error building the firmware image Server response: Error: Impossible package selection
>
> [Please report the error message and request](https://github.com/openwrt/asu/issues) Request Data:
>
> { "url": [...]

Removed wolf package and sysupgrade.en package requests and the build completed