openwrt/asu #1092: luci-attendedsysupgrade interface throwing TypeError promises[0].packages is undefined.

nishikant, 2024-11-23:
```
TypeError: promises[0].packages is undefined
    load http://192.168.3.1/luci-static/resources/view/attendedsysupgrade/overview.js?v=24.328.00185~03f37cf:25
    __init__ http://192.168.3.1/luci-static/resources/luci.js?v=24.328.00185~03f37cf:147
    super http://192.168.3.1/luci-static/resources/luci.js?v=24.328.00185~03f37cf:22
    ClassConstructor http://192.168.3.1/luci-static/resources/luci.js?v=24.328.00185~03f37cf:12
    compileClass http://192.168.3.1/luci-static/resources/luci.js?v=24.328.00185~03f37cf:182
    promise callback*compileClass http://192.168.3.1/luci-static/resources/luci.js?v=24.328.00185~03f37cf:178
    promise callback*require http://192.168.3.1/luci-static/resources/luci.js?v=24.328.00185~03f37cf:184
    instantiateView http://192.168.3.1/luci-static/resources/ui.js?v=24.328.00185~03f37cf:361
    <anonymous> http://192.168.3.1/cgi-bin/luci/admin/system/attendedsysupgrade:51
    promise callback* http://192.168.3.1/cgi-bin/luci/admin/system/attendedsysupgrade:50
luci.js:161:9

```
[image]

mskpluk, 2024-11-24:
same issue on banana pi bpi-r3

gjws, 2024-11-24:
Same issue on my GL-MT2500 with SNAPSHOT r28146-52b6c92479

EtoTen, 2024-11-24:
Same issue on LuCI Master (24.319.72217~9fe1e34) /[ OpenWrt SNAPSHOT (r28085-6720c4ccba)](https://openwrt.org/) Bananapi BPI-R4

dannil, 2024-11-26:
This is neither an issue with asu or luci-app-attendedsysupgrade, see https://github.com/openwrt/openwrt/issues/16907, the LuCI app invokes `ubus call rpc-sys packagelist` under the hood when getting the packages.

nishikant, 2024-12-03:
This is seems to be fixed in latest updates.