openwrt/asu #1096: S12 Pro

ktauchathuranga, 2024-11-28:
Request Data:

{
    "url": "https://sysupgrade.openwrt.org",
    "revision": "r23124-49aaab08ea",
    "advanced_mode": "0",
    "sha256_unsigned": "",
    "branch": "23.05",
    "efi": null,
    "profile": "tozed,zlt-s12-pro",
    "target": "ramips/mt7621",
    "version": "23.05-SNAPSHOT",
    "packages": [
        "adb",
        "attendedsysupgrade-common",
        "banip",
        "base-files",
        "bcp38",
        "busybox",
        "ca-bundle",
        "cgi-io",
        "chat",
        "comgt",
        "comgt-ncm",
        "dnsmasq",
        "dropbear",
        "firewall4",
        "fstools",
        "fwtool",
        "getrandom",
        "hostapd-common",
        "ip-tiny",
        "ip6tables-nft",
        "ipset",
        "iptables-mod-conntrack-extra",
        "iptables-mod-ipopt",
        "iptables-nft",
        "iw",
        "iwinfo",
        "jansson",
        "jshn",
        "jsonfilter",
        "kernel",
        "kmod-cfg80211",
        "kmod-crypto-acompress",
        "kmod-crypto-aead",
        "kmod-crypto-ccm",
        "kmod-crypto-cmac",
        "kmod-crypto-crc32c",
        "kmod-crypto-ctr",
        "kmod-crypto-gcm",
        "kmod-crypto-gf128",
        "kmod-crypto-ghash",
        "kmod-crypto-hash",
        "kmod-crypto-hmac",
        "kmod-crypto-lib-chacha20",
        "kmod-crypto-lib-chacha20poly1305",
        "kmod-crypto-lib-curve25519",
        "kmod-crypto-lib-poly1305",
        "kmod-crypto-manager",
        "kmod-crypto-null",
        "kmod-crypto-rng",
        "kmod-crypto-seqiv",
        "kmod-crypto-sha512",
        "kmod-gpio-button-hotplug",
        "kmod-ifb",
        "kmod-ip6tables",
        "kmod-ipt-conntrack",
        "kmod-ipt-conntrack-extra",
        "kmod-ipt-core",
[... 207 lines cut ...]
Downloading https://downloads.openwrt.org/releases/23.05-SNAPSHOT/packages/mipsel_24kc/luci/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/23.05-SNAPSHOT/packages/mipsel_24kc/packages/Packages.gz
Updated list of available packages in /builder/build_dir/target-mipsel_24kc_musl/root-ramips/../../../../builder/dl/openwrt_packages
Downloading https://downloads.openwrt.org/releases/23.05-SNAPSHOT/packages/mipsel_24kc/packages/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/23.05-SNAPSHOT/packages/mipsel_24kc/routing/Packages.gz
Updated list of available packages in /builder/build_dir/target-mipsel_24kc_musl/root-ramips/../../../../builder/dl/openwrt_routing
Downloading https://downloads.openwrt.org/releases/23.05-SNAPSHOT/packages/mipsel_24kc/routing/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/23.05-SNAPSHOT/packages/mipsel_24kc/telephony/Packages.gz
Updated list of available packages in /builder/build_dir/target-mipsel_24kc_musl/root-ramips/../../../../builder/dl/openwrt_telephony
Downloading https://downloads.openwrt.org/releases/23.05-SNAPSHOT/packages/mipsel_24kc/telephony/Packages.sig
Signature check passed.
Downloading file:packages/Packages
Updated list of available packages in /builder/build_dir/target-mipsel_24kc_musl/root-ramips/../../../../builder/dl/imagebuilder
Downloading file:packages/Packages.sig
Signature check passed.
Collected errors:
 * opkg_install_cmd: Cannot install package luci-app-3ginfo-lite.
 * opkg_install_cmd: Cannot install package modemband.
 * opkg_install_cmd: Cannot install package s12-pro-signald.
make[2]: *** [Makefile:189: package_install] Error 255
make[1]: *** [Makefile:154: _call_manifest] Error 2
make: *** [Makefile:274: manifest] Error 2

efahl, 2024-11-28:
Those three packages don't appear to be in any of the official OpenWrt repos.  Are you running a vendor-specific variant?

ktauchathuranga, 2024-11-29:
thank you for the reply,
i tried so many things, and at one point it fixed,
now i dont remember exactly what caused that.
but however, the problem is fixed for me.
and i am sorry i cant exactly tell to the community how it was fixed. 👻

closed 2024-11-29 by ktauchathuranga