openwrt/asu #1097: Unable to upgrade to lastest SNAPSHOT version

UltraHKR, 2024-11-28:
Trying to upgrade Netgear WNDR3800CH using Luci Attended-sysupgrade
From: OpenWrt SNAPSHOT r27279-8b6c63cfaf / LuCI Master 24.212.79282~65b8002
To:  SNAPSHOT - r28199-1bdb6d8404

If I add OPKG as a dependency it throws this error
NOTE: opkg is NOT availabl anywhere in https://downloads.openwrt.org/snapshots/packages/mips_24kc/

Request Data:
{
    "url": "https://sysupgrade.openwrt.org",
    "branch": "SNAPSHOT",
    "revision": "r27279-8b6c63cfaf",
    "efi": null,
    "advanced_mode": "1",
    "request_hash": "",
    "sha256_unsigned": "",
    "client": "luci/24.205.37920~e55f9d9",
    "packages": {
        "libc": "1.2.5-r4",
        "libpthread": "1.2.5-r4",
        "busybox": "1.36.1-r1",
        "zram-swap": "32",
        "swconfig": "12",
        "kmod-lib-crc-ccitt": "6.6.48-r1",
        "getrandom": "2024.04.26~85f10530-r1",
        "kmod-pppoe": "6.6.48-r1",
        "procd-ujail": "2024.07.07~f230c117-r1",
        "base-files": "1614~8b6c63cfaf",
        "libustream-mbedtls": "2024.07.28~99bd3d2b-r1",
        "firewall4": "2024.05.21~4c01d1eb-r1",
        "uboot-envtools": "2024.07-r1",
        "kmod-usb-ohci": "6.6.48-r1",
        "dnsmasq": "2.90-r3",
        "procd": "2024.07.07~f230c117-r1",
        "kmod-crypto-acompress": "6.6.48-r1",
        "kmod-leds-reset": "6.6.48-r1",
        "kmod-usb2": "6.6.48-r1",
        "libmbedtls": "3.6.0-r1",
        "odhcp6c": "2023.05.12~bcd28363-r20",
        "kmod-ath9k": "6.6.48.6.9.9-r1",
        "uci": "2023.08.10~5781664d-r1",
        "dropbear": "2024.85-r1",
        "luci-app-dawn": "24.188.45693~65f8c14",
        "mtd": "26",
        "odhcpd-ipv6only": "2024.05.08~a2988231-r1",
        "wpad": "2024.03.09~695277a5-r3",
        "kmod-owl-loader": "6.6.48.6.9.9-r1",
        "procd-seccomp": "2024.07.07~f230c117-r1",
        "libjson-script": "2024.03.29~eb9bcb64-r1",
        "urandom-seed": "3",
        "ppp": "2.4.9_git20210104-r5",
        "kmod-gpio-button-hotplug": "6.6.48-r3",
        "logd": "2024.04.26~85f10530-r1",
        "luci-app-attendedsysupgrade": "24.205.37920~e55f9d9",
        "kmod-ppp": "6.6.48-r1",
        "ca-bundle": "20240203-r1",
        "libuclient": "2024.04.19~e8780fa7-r1",
        "luci": "23.051.66410~a505bb1",
        "owut": "2024.08.09~8a6becd8-r1",
        "kmod-switch-rtl8366s": "6.6.48-r1",
        "kmod-usb-ledtrig-usbport": "6.6.48-r1",
        "kmod-lib-lzo": "6.6.48-r1",
        "kmod-ath": "6.6.48.6.9.9-r1",
        "kernel": "6.6.48~cbf8446a39d3bcd8258a4b4723e5f5b5-r1",
        "urngd": "2023.11.01~44365eb1-r1",
        "ppp-mod-pppoe": "2.4.9_git20210104-r5"
    },
    "profile": "netgear,wndr3800ch",
    "target": "ath79/generic",
    "version": "SNAPSHOT",
    "diff_packages": true,
    "filesystem": "squashfs"
}
STDERR:
Generate local signing keys...
WARNING: can't open config file: /builder/shared-workdir/build/staging_dir/host/etc/ssl/openssl.cnf
WARNING: can't open config file: /builder/shared-workdir/build/staging_dir/host/etc/ssl/openssl.cnf
read EC key
writing EC key
fetch https://downloads.openwrt.org/snapshots/targets/ath79/generic/packages/packages.adb
fetch https://downloads.openwrt.org/snapshots/packages/mips_24kc/base/packages.adb
fetch https://downloads.openwrt.org/snapshots/targets/ath79/generic/kmods/6.6.63-1-7d0e1a34e0ba3672a93534d0608105ac/packages.adb
fetch https://downloads.openwrt.org/snapshots/packages/mips_24kc/luci/packages.adb
fetch https://downloads.openwrt.org/snapshots/packages/mips_24kc/packages/packages.adb
fetch https://downloads.openwrt.org/snapshots/packages/mips_24kc/routing/packages.adb
fetch https://downloads.openwrt.org/snapshots/packages/mips_24kc/telephony/packages.adb
WARNING: opening packages/packages.adb: No such file or directory
Package list missing or not up-to-date, generating it.

Building package index...
ERROR: unable to select packages:
  opkg (no such package):
    required by: world[opkg]
make[2]: *** [Makefile:225: package_install] Error 1
make[1]: *** [Makefile:166: _call_manifest] Error 2
make: *** [Makefile:340: manifest] Error 2

efahl, 2024-11-28:
> If I add OPKG as a dependency it throws this error

Sounds like a different issue than the device-based missing ubus call.  The `opkg` package is no longer available in SNAPSHOT, you need to replace it with `apk-mbedtls` (or build your own images setting `CONFIG_USE_APK=n`).

Lots of discussion here: https://forum.openwrt.org/t/the-future-is-now-opkg-vs-apk/201164

efahl, 2024-11-29:
Yup, I was just waiting for the decision that came through a couple days ago.  I submitted a PR to that effect yesterday: https://github.com/openwrt/asu/pull/1098

geak, 2024-12-06:
I also can not upgrade from one snapshot (r28091-77cfe8fd15) to a latest one (r28246-ce66385):

```json
{
    "url": "https://sysupgrade.openwrt.org",
    "branch": "SNAPSHOT",
    "revision": "r28091-77cfe8fd15",
    "efi": null,
    "advanced_mode": "1",
    "request_hash": "",
    "sha256_unsigned": "",
    "client": "luci/24.338.58761~e76155d",
    "packages": {
        "apk-mbedtls": "3.0.0_pre20241130-r1",
        "attendedsysupgrade-common": "9",
        "avahi-autoipd": "0.8-r9",
        "avahi-daemon-service-ssh": "0.8-r9",
        "base-files": "1633~77cfe8fd15",
        "bind-dig": "9.20.0-r1",
        "block-mount": "2024.12.02~49d36ba2-r1",
        "busybox": "1.36.1-r2",
        "ca-bundle": "20240203-r1",
        "comgt": "0.32-r36",
        "coreutils-sleep": "9.3-r1",
        "coreutils-sort": "9.3-r1",
        "coreutils-timeout": "9.3-r1",
        "curl": "8.10.1-r1",
        "dnsmasq": "2.90-r3",
        "dos2unix": "7.5.2-r1",
        "dosfstools": "4.2-r4",
        "dropbear": "2024.86-r1",
        "e2fsprogs": "1.47.0-r2",
        "eip197-mini-firmware": "20241110-r1",
        "ethtool-full": "6.10-r1",
        "exfat-fsck": "1.2.5-r1",
        "exfat-mkfs": "1.2.5-r1",
        "f2fs-tools": "1.16.0-r3",
        "f2fsck": "1.16.0-r3",
        "firewall4": "2024.05.21~4c01d1eb-r1",
        "fitblk": "2",
        "fstools": "2024.12.02~49d36ba2-r1",
        "fstrim": "2.40.2-r1",
        "fwtool": "2019.11.12~8f7fe925-r1",
        "gdisk": "1.0.10-r1",
        "getrandom": "2024.04.26~85f10530-r1",
        "gpioctl-sysfs": "0.0.7-r1",
        "gzip": "1.13-r1",
        "hostapd-common": "2024.09.15~5ace39b0-r2",
        "htop": "3.3.0-r2",
        "iperf3": "3.17.1-r4",
        "iptables-nft": "1.8.10-r1",
        "iwinfo": "2024.10.20~b94f066e-r1",
        "jansson4": "2.14-r3",
        "jshn": "2024.03.29~eb9bcb64-r1",
        "jsonfilter": "2024.01.23~594cfa86-r1",
        "kernel": "6.6.60~0561031e21f60f05c8f540bffb723036-r1",
        "kmod-crypto-acompress": "6.6.60-r1",
        "kmod-crypto-aead": "6.6.60-r1",
        "kmod-crypto-authenc": "6.6.60-r1",
        "kmod-crypto-crc32c": "6.6.60-r1",
        "kmod-crypto-des": "6.6.60-r1",
        "kmod-crypto-hash": "6.6.60-r1",
        "kmod-crypto-hmac": "6.6.60-r1",
        "kmod-crypto-hw-safexcel": "6.6.60-r1",
        "kmod-crypto-manager": "6.6.60-r1",
        "kmod-crypto-md5": "6.6.60-r1",
        "kmod-crypto-null": "6.6.60-r1",
        "kmod-crypto-sha1": "6.6.60-r1",
        "kmod-crypto-sha256": "6.6.60-r1",
        "kmod-crypto-sha512": "6.6.60-r1",
[... 154 lines cut ...]
        "ucode-mod-uloop": "2024.12.02~b0b5d938-r1",
        "umbim": "2022.08.13~146bc77c-r25",
        "unzip": "6.0-r8",
        "uqmi": "2024.08.25~28b48a10-r2",
        "urandom-seed": "3",
        "urngd": "2023.11.01~44365eb1-r1",
        "usbids": "0.387-r1",
        "usbutils": "017-r1",
        "usign": "2020.05.23~f1f65026-r1",
        "wireguard-tools": "1.0.20210914-r4",
        "wpad-basic-mbedtls": "2024.09.15~5ace39b0-r2",
        "zapret": "69.20241123",
        "zapret-ip2net": "69.20241123",
        "zapret-mdig": "69.20241123",
        "zlib": "1.3.1-r1",
        "zoneinfo-europe": "2024b-r1",
        "zoneinfo-simple": "2024b-r1"
    },
    "profile": "bananapi,bpi-r3-mini",
    "target": "mediatek/filogic",
    "version": "SNAPSHOT",
    "diff_packages": true,
    "filesystem": "squashfs"
}
```

efahl, 2024-12-06:
> I also can not upgrade from one snapshot (r28091-77cfe8fd15) to a latest one (r28246-ce66385):

What error are you seeing?

geak, 2024-12-06:
> > I also can not upgrade from one snapshot (r28091-77cfe8fd15) to a latest one (r28246-ce66385):
> 
> What error are you seeing?

As far as I remember there was nothing specific, just what I posted above. Unfortunately I can not check it again, because "Attended Sysupgrade" is stuck with this now 😔️️️️️️:
```
Searching for an available sysupgrade of SNAPSHOT - r28091-77cfe8fd15
```

geak, 2024-12-08:
```
# owut download -v
owut - OpenWrt Upgrade Tool 2024.11.29~286d1014-r1 (/usr/bin/owut)
Downloaded https://sysupgrade.openwrt.org/json/v1/overview.json to /tmp/owut-overview.json (15017B at 0.229 Mbps)
Downloaded https://downloads.openwrt.org/snapshots/targets/mediatek/filogic/profiles.json to /tmp/owut-platform.json (101267B at 0.773 Mbps)
Downloaded https://sysupgrade.openwrt.org/json/v1/snapshots/packages/aarch64_cortex-a53-index.json to /tmp/owut-packages-arch.json (346071B at 0.880 Mbps)
Downloaded https://sysupgrade.openwrt.org/json/v1/snapshots/targets/mediatek/filogic/index.json to /tmp/owut-packages-plat.json (36820B at 0.281 Mbps)
ASU-Server     https://sysupgrade.openwrt.org
Upstream       https://downloads.openwrt.org
Target         mediatek/filogic
Profile        bananapi_bpi-r3-mini
Package-arch   aarch64_cortex-a53
Root-FS-type   squashfs
Sys-type       sysupgrade
Version-from   SNAPSHOT r28091-77cfe8fd15 (kernel 6.6.60)
Version-to     SNAPSHOT r28251-b4cc5743d9 (kernel 6.6.63)
Build-FS-type  squashfs
Build-at       2024-12-05T01:34:01Z (~80 hours ago)
Image-prefix   openwrt-mediatek-filogic-bananapi_bpi-r3-mini
Image-URL      https://downloads.openwrt.org/snapshots/targets/mediatek/filogic
Image-file     openwrt-mediatek-filogic-bananapi_bpi-r3-mini-squashfs-sysupgrade.itb
Installed      371 packages
Top-level      222 packages
Default         41 packages
User-installed 186 packages (top-level only)

Package version changes:
  base-files                          1633~77cfe8fd15                            missing to-version
  kmod-cfg80211                       6.6.60.6.11.2-r1                           6.6.63.6.11.2-r1
  kmod-crypto-acompress               6.6.60-r1                                  6.6.63-r1
  kmod-crypto-aead                    6.6.60-r1                                  6.6.63-r1
  kmod-crypto-authenc                 6.6.60-r1                                  6.6.63-r1
  kmod-crypto-ccm                     6.6.60-r1                                  6.6.63-r1
  kmod-crypto-cmac                    6.6.60-r1                                  6.6.63-r1
  kmod-crypto-crc32                   6.6.60-r1                                  6.6.63-r1
  kmod-crypto-crc32c                  6.6.60-r1                                  6.6.63-r1
  kmod-crypto-ctr                     6.6.60-r1                                  6.6.63-r1
  kmod-crypto-des                     6.6.60-r1                                  6.6.63-r1
  kmod-crypto-gcm                     6.6.60-r1                                  6.6.63-r1
  kmod-crypto-geniv                   6.6.60-r1                                  6.6.63-r1
  kmod-crypto-gf128                   6.6.60-r1                                  6.6.63-r1
  kmod-crypto-ghash                   6.6.60-r1                                  6.6.63-r1
  kmod-crypto-hash                    6.6.60-r1                                  6.6.63-r1
  kmod-crypto-hmac                    6.6.60-r1                                  6.6.63-r1
  kmod-crypto-hw-safexcel             6.6.60-r1                                  6.6.63-r1
  kmod-crypto-kpp                     6.6.60-r1                                  6.6.63-r1
  kmod-crypto-lib-chacha20            6.6.60-r1                                  6.6.63-r1
  kmod-crypto-lib-chacha20poly1305    6.6.60-r1                                  6.6.63-r1
  kmod-crypto-lib-curve25519          6.6.60-r1                                  6.6.63-r1
  kmod-crypto-lib-poly1305            6.6.60-r1                                  6.6.63-r1
  kmod-crypto-manager                 6.6.60-r1                                  6.6.63-r1
  kmod-crypto-md5                     6.6.60-r1                                  6.6.63-r1
  kmod-crypto-null                    6.6.60-r1                                  6.6.63-r1
  kmod-crypto-rng                     6.6.60-r1                                  6.6.63-r1
  kmod-crypto-seqiv                   6.6.60-r1                                  6.6.63-r1
  kmod-crypto-sha1                    6.6.60-r1                                  6.6.63-r1
  kmod-crypto-sha256                  6.6.60-r1                                  6.6.63-r1
  kmod-crypto-sha3                    6.6.60-r1                                  6.6.63-r1
  kmod-crypto-sha512                  6.6.60-r1                                  6.6.63-r1
  kmod-fs-exfat                       6.6.60-r1                                  6.6.63-r1
  kmod-fs-ext4                        6.6.60-r1                                  6.6.63-r1
  kmod-fs-f2fs                        6.6.60-r1                                  6.6.63-r1
  kmod-fs-vfat                        6.6.60-r1                                  6.6.63-r1
  kmod-gpio-button-hotplug            6.6.60-r5                                  6.6.63-r5
  kmod-hwmon-core                     6.6.60-r1                                  6.6.63-r1
  kmod-hwmon-pwmfan                   6.6.60-r1                                  6.6.63-r1
  kmod-i2c-core                       6.6.60-r1                                  6.6.63-r1
  kmod-i2c-mux                        6.6.60-r1                                  6.6.63-r1
  kmod-i2c-mux-gpio                   6.6.60-r1                                  6.6.63-r1
  kmod-ipt-core                       6.6.60-r1                                  6.6.63-r1
[... 64 lines cut ...]
[... cut ...]
  kmod-wireguard                      6.6.60-r1                                  6.6.63-r1
  kmod-wwan                           6.6.60-r1                                  6.6.63-r1
  kmod-zram                           6.6.60-r1                                  6.6.63-r1
  openwrt-keyring                     2024.11.01~fbae29d7-r2                     2024.11.01~fbae29d7-r1
1 packages missing in target version, cannot upgrade
1 packages were downgraded
126 packages are out-of-date
```

It says missing to-version for base-files.

efahl, 2024-12-10:
@geak
> It says "missing to-version" for base-files.

Thanks for confirming, I've seen that on some combination of oldish-snapshot going to new snapshot, too. I will investigate...

geak, 2024-12-11:
Seems it works now without any issues and allows me to install latest snapshot firmware. And.. It successfully did it!

```
Model            Bananapi BPi-R3 Mini
Architecture     ARMv8 Processor rev 4
Target Platform  mediatek/filogic
Firmware Version OpenWrt SNAPSHOT r28277-59dd9cddf9 / LuCI Master 24.342.58028~679d3b8
```

efahl, 2024-12-12:
> Are there replacements for those packages marked as `missing to-version`, or should those packages be remapped to other package names?

The missing `base-files` was due to a change in the builds, and was fixed in `owut` yesterday (awaiting the PR in packages).

The rest of those are "missing" because the apk versions of the package contain ABI-version info on the name and there's no way to know that, say, `jansson4` and `jansson` are the same package.  This is an ongoing thing, and won't be fixed soon, but once you get past the `opkg` -> `apk` transition, they won't be visible as they'll (mostly) be consistently wrong.  

But, since all of those are either defaults or included by dependencies, it won't break anything to ignore them.  Give `--force` a try and see if `owut` will allow you to build.  If not, you'll have to generate the package list (`owut list` can do that) and run it through the Firmware Selector, which shouldn't care as it doesn't do any package consistency checks (it relies on the imagebuilder to do that).

satmandu, 2024-12-12:
I'm supposed to do `owut upgrade -v -a comma_separated_package_list --force`, right?

```bash
packages=$(owut list -f fs-all)
echo ${packages// /,}
bind-server-filter-aaaa,luci-proto-wireguard,libuci-lua,iwinfo,openssh-sftp-server,liblucihttp,cgi-io,libbz2,mkf2fs,kmod-usb-net-asix-ax88179,libwebsockets-full,coreutils-chmod,iw,diffutils,busybox,jq,kmod-sound-arm-bcm2835,libubus-lua,ustp,rpcd-mod-wireguard,libiwinfo-lua,bash,kmod-usb-dwc2,openssh-client-utils,haproxy,liblzma,kmod-lib-crc-ccitt,getrandom,luci-theme-openwrt,libusb-1.0,kmod-pppoe,kmod-ipt-conntrack,procd-ujail,base-files,pciids,luci-app-ttyd,partx-utils,dosfstools,libustream-mbedtls,coreutils,libf2fs,libsmartcols,6in4,luci-app-uhttpd,kmod-r8169,firewall4,pciutils,kmod-mii,procd,libkmod,kmod-of-mdio,qrencode,kmod-crypto-acompress,kmod-usb-hid,kmod-usb-ehci,kmod-usb-net-rndis,kmod-fs-vfat,kmod-crypto-lib-chacha20poly1305,luci-app-ddns,kmod-usb2,kmod-phy-realtek,kmod-ipt-conntrack-extra,bcm27xx-utils,kmod-ip6tables,odhcp6c,iptables-nft,kmod-udptunnel4,kmod-udptunnel6,kmod-usb-net-lan78xx,ethtool,cypress-firmware-43455-sdio,uci,lua,luci-ssl,nmap,dropbear,kmod-crypto-kpp,curl,rrdtool1,luci-theme-openwrt-2020,luci-compat,kmod-ipvlan,mtd,odhcpd-ipv6only,procd-seccomp,libjson-script,usb-modeswitch,kmod-brcmfmac,urandom-seed,ipset,ppp,bind-rndc,kmod-ikconfig,logd,kmod-usb-net,kmod-wireguard,ddns-scripts-cloudflare,luci-app-attendedsysupgrade,tmux,irqbalance,coreutils-od,e2fsprogs,attendedsysupgrade-common,ca-certificates,kmod-ppp,brcmfmac-firmware-usb,libpcap,openssh-sftp-client,kmod-hid-generic,brcmfmac-nvram-43455-sdio,htop,kmod-tcp-bbr,liblucihttp-lua,bind-dig,luci-app-snmpd,luci,owut,libstdcpp,bind-server,libqrencode,kmod-ipt-nat6,librrd1,kmod-lib-lzo,kmod-crypto-lib-curve25519,kmod-crypto-lib-chacha20,blkid,openssh-client,libnetsnmp,kmod-fixed-phy,libgdbm,liblucihttp-ucode,kmod-hid,bcm27xx-gpu-fw,kmod-usb-net-cdc-ether,acme,kmod-mdio-devres,luci-app-acme,ppp-mod-pppoe,kmod-phy-microchip,kmod-ipt-nat,dnsmasq-full,libevent2-core,apk-mbedtls
owut upgrade -v  -a ${packages// /,}
owut upgrade -v  -a ${packages// /,} --force
owut - OpenWrt Upgrade Tool 2024.10.27~8c7e42f0-r1 (/usr/bin/owut)
Downloaded https://sysupgrade.openwrt.org/json/v1/overview.json to /tmp/owut-overview.json (14797B at 0.113 Mbps)
Downloaded https://downloads.openwrt.org/snapshots/targets/bcm27xx/bcm2711/profiles.json to /tmp/owut-platform.json (2226B at 0.034 Mbps)
Mapping platform raspberrypi_4-compute-module to rpi-4
Downloaded https://sysupgrade.openwrt.org/json/v1/snapshots/packages/aarch64_cortex-a72-index.json to /tmp/owut-packages-arch.json (346092B at 2.640 Mbps)
Downloaded https://sysupgrade.openwrt.org/json/v1/snapshots/targets/bcm27xx/bcm2711/index.json to /tmp/owut-packages-plat.json (39774B at 0.303 Mbps)
ASU-Server     https://sysupgrade.openwrt.org
Upstream       https://downloads.openwrt.org
Target         bcm27xx/bcm2711
Profile        rpi-4
Package-arch   aarch64_cortex-a72
Root-FS-type   ext4
Sys-type       sysupgrade
Version-from   SNAPSHOT r28050-be952e98bc (kernel 6.6.60)
Version-to     SNAPSHOT r28282-0d21cc8a92 (kernel 6.6.63)
Build-FS-type  ext4
Build-at       2024-12-10T20:18:06Z (~32 hours ago)
Image-prefix   openwrt-bcm27xx-bcm2711-rpi-4
Image-URL      https://downloads.openwrt.org/snapshots/targets/bcm27xx/bcm2711
Image-file     openwrt-bcm27xx-bcm2711-rpi-4-ext4-sysupgrade.img.gz
Installed      309 packages
Top-level      143 packages
Default         42 packages
User-installed 115 packages (top-level only)

Package version changes:
  apk-mbedtls                         not-installed                              3.0.0_pre20241130-r1
  base-files                          1631~be952e98bc                            missing to-version
  brcmfmac-firmware-usb               20241017-r1                                20241110-r1
  brcmfmac-nvram-43455-sdio           20241017-r1                                20241110-r1
  busybox                             1.36.1-r2                                  1.37.0-r3
  dropbear                            2024.85-r1                                 2024.86-r1
  fstools                             2024.07.14~408c2cc4-r1                     2024.12.02~49d36ba2-r1
  haproxy                             3.0.5-r1                                   3.0.6-r1
  jansson                             2.14-r3                                    missing to-version
  kmod-brcmfmac                       6.6.60.6.11.2-r1                           6.6.63.6.11.2-r1
  kmod-brcmutil                       6.6.60.6.11.2-r1                           6.6.63.6.11.2-r1
  kmod-cfg80211                       6.6.60.6.11.2-r1                           6.6.63.6.11.2-r1
  kmod-crypto-acompress               6.6.60-r1                                  6.6.63-r1
  kmod-crypto-crc32c                  6.6.60-r1                                  6.6.63-r1
  kmod-crypto-hash                    6.6.60-r1                                  6.6.63-r1
  kmod-crypto-kpp                     6.6.60-r1                                  6.6.63-r1
  kmod-crypto-lib-chacha20            6.6.60-r1                                  6.6.63-r1
  kmod-crypto-lib-chacha20poly1305    6.6.60-r1                                  6.6.63-r1
  kmod-crypto-lib-curve25519          6.6.60-r1                                  6.6.63-r1
  kmod-crypto-lib-poly1305            6.6.60-r1                                  6.6.63-r1
[... cut ...]
  ucode-mod-uloop                     2024.07.22~b610860d-r3                     2024.12.06~209f041f-r1
50 packages missing in target version, cannot upgrade
1 packages were downgraded
194 packages are out-of-date

Automatic package replacements/removals:
  Package                             Replaced-by
  opkg                                apk-mbedtls
Details at https://sysupgrade.openwrt.org/json/v1/overview.json

ERROR: Update checks reveal errors, can't proceed
Downloaded 4 files, 402889 bytes at 0.878 Mbps
```

satmandu, 2024-12-12:
(That's using the owut ` VERSION = "source-e38844ae";`)

satmandu, 2024-12-12:
And with the most recent owut version:
```bash
owut download -v --force
owut - OpenWrt Upgrade Tool source-e38844ae (/usr/bin/owut)
Downloaded https://sysupgrade.openwrt.org/json/v1/overview.json to /tmp/owut-overview.json (14797B at 0.113 Mbps)
Downloaded https://downloads.openwrt.org/snapshots/targets/bcm27xx/bcm2711/profiles.json to /tmp/owut-platform.json (2226B at 0.034 Mbps)
Mapping platform raspberrypi_4-compute-module to rpi-4
Downloaded https://sysupgrade.openwrt.org/json/v1/snapshots/packages/aarch64_cortex-a72-index.json to /tmp/owut-packages-arch.json (346092B at 2.640 Mbps)
Downloaded https://sysupgrade.openwrt.org/json/v1/snapshots/targets/bcm27xx/bcm2711/index.json to /tmp/owut-packages-plat.json (39774B at 0.303 Mbps)
ASU-Server     https://sysupgrade.openwrt.org
Upstream       https://downloads.openwrt.org
Target         bcm27xx/bcm2711
Profile        rpi-4
Package-arch   aarch64_cortex-a72
Root-FS-type   ext4
Sys-type       sysupgrade
Version-from   SNAPSHOT r28050-be952e98bc (kernel 6.6.60)
Version-to     SNAPSHOT r28282-0d21cc8a92 (kernel 6.6.63)
Build-FS-type  ext4
Build-at       2024-12-10T20:18:06Z (~32 hours ago)
Image-prefix   openwrt-bcm27xx-bcm2711-rpi-4
Image-URL      https://downloads.openwrt.org/snapshots/targets/bcm27xx/bcm2711
Image-file     openwrt-bcm27xx-bcm2711-rpi-4-ext4-sysupgrade.img.gz
Installed      309 packages
Top-level      145 packages
Default         42 packages
User-installed 115 packages (top-level only)

Package version changes:
  apk-mbedtls                         not-installed                              3.0.0_pre20241130-r1
  brcmfmac-firmware-usb               20241017-r1                                20241110-r1
  brcmfmac-nvram-43455-sdio           20241017-r1                                20241110-r1
  busybox                             1.36.1-r2                                  1.37.0-r3
  dropbear                            2024.85-r1                                 2024.86-r1
  fstools                             2024.07.14~408c2cc4-r1                     2024.12.02~49d36ba2-r1
  haproxy                             3.0.5-r1                                   3.0.6-r1
  jansson                             2.14-r3                                    missing to-version
  kmod-brcmfmac                       6.6.60.6.11.2-r1                           6.6.63.6.11.2-r1
  kmod-brcmutil                       6.6.60.6.11.2-r1                           6.6.63.6.11.2-r1
  kmod-cfg80211                       6.6.60.6.11.2-r1                           6.6.63.6.11.2-r1
  kmod-crypto-acompress               6.6.60-r1                                  6.6.63-r1
  kmod-crypto-crc32c                  6.6.60-r1                                  6.6.63-r1
  kmod-crypto-hash                    6.6.60-r1                                  6.6.63-r1
  kmod-crypto-kpp                     6.6.60-r1                                  6.6.63-r1
  kmod-crypto-lib-chacha20            6.6.60-r1                                  6.6.63-r1
  kmod-crypto-lib-chacha20poly1305    6.6.60-r1                                  6.6.63-r1
  kmod-crypto-lib-curve25519          6.6.60-r1                                  6.6.63-r1
  kmod-crypto-lib-poly1305            6.6.60-r1                                  6.6.63-r1
  kmod-fixed-phy                      6.6.60-r1                                  6.6.63-r1
  kmod-fs-vfat                        6.6.60-r1                                  6.6.63-r1
  kmod-hid                            6.6.60-r1                                  6.6.63-r1
  kmod-hid-generic                    6.6.60-r1                                  6.6.63-r1
  kmod-ikconfig                       6.6.60-r1                                  6.6.63-r1
  kmod-input-core                     6.6.60-r1                                  6.6.63-r1
  kmod-input-evdev                    6.6.60-r1                                  6.6.63-r1
  kmod-ip6tables                      6.6.60-r1                                  6.6.63-r1
  kmod-ipt-conntrack                  6.6.60-r1                                  6.6.63-r1
  kmod-ipt-conntrack-extra            6.6.60-r1                                  6.6.63-r1
  kmod-ipt-core                       6.6.60-r1                                  6.6.63-r1
  kmod-ipt-ipset                      6.6.60-r1                                  6.6.63-r1
  kmod-ipt-nat                        6.6.60-r1                                  6.6.63-r1
  kmod-ipt-nat6                       6.6.60-r1                                  6.6.63-r1
  kmod-iptunnel                       6.6.60-r1                                  6.6.63-r1
  kmod-iptunnel4                      6.6.60-r1                                  6.6.63-r1
  kmod-ipvlan                         6.6.60-r1                                  6.6.63-r1
  kmod-lib-crc-ccitt                  6.6.60-r1                                  6.6.63-r1
  kmod-lib-crc32c                     6.6.60-r1                                  6.6.63-r1
  kmod-lib-lzo                        6.6.60-r1                                  6.6.63-r1
  kmod-libphy                         6.6.60-r1                                  6.6.63-r1
  kmod-mdio-devres                    6.6.60-r1                                  6.6.63-r1
[... 138 lines cut ...]
  rpcd-mod-file                       2024.09.17~9f4b86e7-r1                     2024.12.02~cc9a471c-r1
[... cut ...]
  ucode-mod-uloop                     2024.07.22~b610860d-r3                     2024.12.06~209f041f-r1
48 packages missing in target version, cannot upgrade
1 packages were downgraded
192 packages are out-of-date

Automatic package replacements/removals:
  Package                             Replaced-by
  opkg                                apk-mbedtls
Details at https://sysupgrade.openwrt.org/json/v1/overview.json

ERROR: Update checks reveal errors, can't proceed
Downloaded 4 files, 402889 bytes at 0.878 Mbps
```

satmandu, 2024-12-12:
After updating owut manually, doing `owut list -f fs-all` and then adding that list to the Firmware Selector dialog box for packages allowed me to create a sysupgrade firmware that worked for me. Thanks for the help!