openwrt/asu #1109: unknown package '(null)' requested. 23.05.5 -> 24.10.0-rc2

piojan, 2024-12-05:
How can I further debug this problem?

```
root@OpenWrt:~# auc -b 24.10 -B 24.10.0-rc2 -n
auc/0.3.2-1
Server:    https://sysupgrade.openwrt.org
Running:   23.05.5 r24106-10cc5fcd00 on x86/64 (generic)
Available: 24.10.0-rc2 r28161-ea17e958b9
Requesting package lists...
 terminfo: 6.4-2 -> 6.4-r2
 luci-proto-wireguard: git-24.332.79522-a493155 -> 24.338.58761~e76155d
 libopenssl: 3.0.15-1 -> 3.0.15-r1
 liblucihttp: 2023-03-15-9b5b683f-1 -> 2023.03.15~9b5b683f-r1
 cgi-io: 2022-08-10-901b0f04-21 -> 2022.08.10~901b0f04-r21
 luci-app-adblock: git-24.332.79522-a493155 -> 24.338.58761~e76155d
 luci-lib-base: git-24.332.79522-a493155 -> 24.338.58761~e76155d
 mkf2fs: 1.16.0-1 -> 1.16.0-r3
 banip: 1.0.1-2 -> 1.0.1-r2
 opkg: 2022-02-24-d038e5b6-2 -> 2024.10.16~38eccbb1-r1
Package luci-app-opkg should be replaced by luci-app-package-manager.
Apply change [N/y] y
 luci-app-opkg: git-24.332.79522-a493155 -> luci-app-package-manager: 24.338.58761~e76155d
 ubus: 2023-06-05-f787c97b-1 -> 2024.10.20~252a9b0c-r1
 libuci: 2023-08-10-5781664d-1 -> 2023.08.10~5781664d-r1
 kmod-ixgbe: 5.15.167-1 -> 6.6.63-r1
 rpcd: 2023-07-01-c07ab2f9-1 -> 2024.09.17~9f4b86e7-r1
 busybox: 1.36.1-1 -> 1.36.1-r2
 libiwinfo: 2023-07-01-ca79f641-1 -> 2024.10.20~b94f066e-r1
 luci-lib-ip: git-24.332.79522-a493155 -> 24.338.58761~e76155d
 kmod-nft-fib: 5.15.167-1 -> 6.6.63-r1
 kmod-nfnetlink: 5.15.167-1 -> 6.6.63-r1
 libubus-lua: 2023-06-05-f787c97b-1 -> 2024.10.20~252a9b0c-r1
 kmod-crypto-hash: 5.15.167-1 -> 6.6.63-r1
 kmod-nf-reject6: 5.15.167-1 -> 6.6.63-r1
 glib2: 2.74.7-1 -> 2.82.0-r1
 libiwinfo-lua: 2023-07-01-ca79f641-1 -> 2024.10.20~b94f066e-r1
 bnx2-firmware: 20230804-1 -> 20241017-r1
 luci-mod-system: git-24.332.79522-a493155 -> 24.338.58761~e76155d
 kmod-nf-flow: 5.15.167-1 -> 6.6.63-r1
 libnl-tiny: 2023-07-27-bc92a280-1 -> 2023.12.05~965c4bf4-r1
 kmod-lib-crc-ccitt: 5.15.167-1 -> 6.6.63-r1
 px5g-mbedtls: 10 -> 11
 getrandom: 2022-08-13-4c7b720b-2 -> 2024.04.26~85f10530-r1
 ucode-mod-ubus: 2024-07-11-1a8a0bcf-3 -> 2024.07.22~b610860d-r3
 mc: 4.8.27-3 -> 4.8.32-r1
 r8169-firmware: 20230804-1 -> 20241017-r1
 luci-theme-bootstrap: git-24.332.79522-a493155 -> 24.338.58761~e76155d
 kmod-pppoe: 5.15.167-1 -> 6.6.63-r1
 libuuid: 2.39-2 -> 2.40.2-r1
 luci-lua-runtime: git-24.332.79522-a493155 -> 24.338.58761~e76155d
 kmod-pppox: 5.15.167-1 -> 6.6.63-r1
 kmod-nf-reject: 5.15.167-1 -> 6.6.63-r1
 procd-ujail: 2023-06-25-2db83655-2 -> 2024.11.13~7330fa55-r1
 base-files: 1562-r24106-10cc5fcd00 -> 1638~ea17e958b9
 kmod-nf-nat: 5.15.167-1 -> 6.6.63-r1
 unbound-daemon: 1.20.0-1 -> 1.21.0-r1
 kmod-input-core: 5.15.167-1 -> 6.6.63-r1
 kmod-crypto-crc32c: 5.15.167-1 -> 6.6.63-r1
 ucode-mod-uci: 2024-07-11-1a8a0bcf-3 -> 2024.07.22~b610860d-r3
 partx-utils: 2.39-2 -> 2.40.2-r1
 libustream-mbedtls: 2023-02-25-498f6e26-1 -> 2024.07.28~99bd3d2b-r1
 netifd: 2024-01-04-c18cc79d-2 -> 2024.11.08~34eb11eb-r1
 coreutils: 9.3-1 -> 9.3-r1
 libf2fs: 1.16.0-1 -> 1.16.0-r3
 libsmartcols: 2.39-2 -> 2.40.2-r1
 kmod-r8169: 5.15.167-1 -> 6.6.63-r1
 firewall4: 2023-09-01-598d9fbb-1 -> 2024.05.21~4c01d1eb-r1
 adblock: 4.2.3-2 -> 4.2.3-r2
 dnsmasq: 2.90-2 -> 2.90-r3
 kmod-mii: 5.15.167-1 -> 6.6.63-r1
[... 117 lines cut ...]
 luci-mod-network: git-24.332.79522-a493155 -> 24.338.58761~e76155d
 kmod-nft-core: 5.15.167-1 -> 6.6.63-r1
 libucode: 2024-07-11-1a8a0bcf-3 -> 2024.07.22~b610860d-r3
 kmod-button-hotplug: 5.15.167-3 -> 6.6.63-r3
 kmod-nls-base: 5.15.167-1 -> 6.6.63-r1
 uhttpd-mod-ubus: 2023-06-25-34a8a74d-2 -> 2023.06.25~34a8a74d-r4
 mtr-json: 0.95-3 -> 0.95-r3
 fwtool: 2019-11-12-8f7fe925-1 -> 2019.11.12~8f7fe925-r1
 jsonfilter: 2024-01-23-594cfa86-1 -> 2024.01.23~594cfa86-r1
 kmod-mdio: 5.15.167-1 -> 6.6.63-r1
 liblucihttp-ucode: 2023-03-15-9b5b683f-1 -> 2023.03.15~9b5b683f-r1
 libubox: 2023-05-23-75a3b870-1 -> 2024.03.29~eb9bcb64-r1
 kmod-igb: 5.15.167-1 -> 6.6.63-r1
 kmod-igc: 5.15.167-1 -> 6.6.63-r1
 kmod-bnx2: 5.15.167-1 -> 6.6.63-r1
 kmod-i2c-algo-bit: 5.15.167-1 -> 6.6.63-r1
 libattr: 2.5.1-1 -> 2.5.2-r3
 kmod-mdio-devres: 5.15.167-1 -> 6.6.63-r1
 urngd: 2023-11-01-44365eb1-1 -> 2023.11.01~44365eb1-r1
 kmod-slhc: 5.15.167-1 -> 6.6.63-r1
 ppp-mod-pppoe: 2.4.9.git-2021-01-04-4 -> 2.5.1-r1
Are you sure you want to continue the upgrade process? [N/y] y
unknown package '(null)' requested.
Bad message (74)
```

aparcar, 2024-12-05:
@dangowrt ?

piojan, 2024-12-08:
Looks like the same build is now carried out successfully. Noting has been change on my side.

```
root@OpenWrt-Eurx:~# auc -b 24.10 -B 24.10.0-rc2
auc/0.3.2-1
Server:    https://sysupgrade.openwrt.org
Running:   23.05.5 r24106-10cc5fcd00 on x86/64 (generic)
Available: 24.10.0-rc2 r28161-ea17e958b9
Requesting package lists...
...
Are you sure you want to continue the upgrade process? [N/y] y
Requesting build.......................................................................
Downloading image from https://sysupgrade.openwrt.org/store/
...
```

marcust, 2024-12-10:
I also saw that on a freshly flashed 23.05.5, after doing an opkg package upgrade it was gone, I guess because there are no packages to upgrade anymore.
I don't see it on ` auc -n -b 24.10 -B 24.10.0-rc2`, but I don't know if it was there before the opkg upgrade or the dry run influences it.

marek22k, 2024-12-11:
Same issue here:
```
root@homeguard:~# auc -b 23.05.5 -y
auc/0.3.2-1
Server:    https://sysupgrade.openwrt.org
Running:   23.05.5 r24106-10cc5fcd00 on mediatek/filogic (glinet,gl-mt6000)
Available: 23.05.5 r24106-10cc5fcd00
Requesting package lists...
 luci-proto-wireguard: git-24.299.74085-56294d0 -> git-24.342.06664-a42c53a
 luci-app-statistics: git-24.313.38121-76e4eca -> git-24.342.06664-a42c53a
 luci-lib-base: git-22.308.54612-9118452 -> git-24.342.06664-a42c53a
 banip: 1.0.0-9 -> 1.0.1-2
 luci-app-opkg: git-24.302.38724-3ded930 -> git-24.342.06664-a42c53a
 libiperf3: 3.17.1-2 -> 3.17.1-4
 luci-lib-ip: git-23.311.79290-c2a887e -> git-24.342.06664-a42c53a
 luci-mod-system: git-24.067.01860-7a82b2f -> git-24.342.06664-a42c53a
 luci-app-lldpd: git-24.284.72256-4988bcd -> git-24.342.06664-a42c53a
 luci-theme-bootstrap: git-24.086.46634-1ffe078 -> git-24.342.06664-a42c53a
 luci-lua-runtime: git-23.233.52805-dae2684 -> git-24.342.06664-a42c53a
 luci-app-banip: git-24.313.38121-76e4eca -> git-24.342.06664-a42c53a
 luci-app-advanced-reboot: 1.0.1-r10 -> 1.0.1
 luci-mod-status: git-24.212.61237-b6da3f2 -> git-24.342.06664-a42c53a
 luci-app-ddns: git-23.346.52990-28c4a65 -> git-24.342.06664-a42c53a
 luci-app-firewall: git-24.067.01746-69867db -> git-24.342.06664-a42c53a
 luci-app-unbound: git-24.075.49994-17051a4 -> git-24.342.06664-a42c53a
 luci-app-diag-core: git-24.313.38121-76e4eca -> git-24.342.06664-a42c53a
 luci-ssl: git-23.035.26083-7550ad6 -> git-24.342.06664-a42c53a
 luci-theme-openwrt-2020: git-24.067.02009-8516918 -> git-24.342.06664-a42c53a
 luci-compat: git-24.086.45108-51aee90 -> git-24.342.06664-a42c53a
 luci-app-sqm: git-24.313.38121-76e4eca -> git-24.342.06664-a42c53a
 luci-proto-ppp: git-24.135.44542-f1ec9c2 -> git-24.342.06664-a42c53a
 luci-mod-dashboard: git-24.313.38121-76e4eca -> git-24.342.06664-a42c53a
 luci-mod-admin-full: git-19.253.48496-3f93650 -> git-24.342.06664-a42c53a
 luci-base: git-24.264.56413-c7a3562 -> git-24.342.06664-a42c53a
 luci-app-commands: git-22.299.76611-9d8feac -> git-24.342.06664-a42c53a
 luci-proto-ipv6: git-24.086.45108-51aee90 -> git-24.342.06664-a42c53a
 luci-app-attendedsysupgrade: git-24.313.38121-76e4eca -> git-24.342.06664-a42c53a
 luci-app-wol: git-21.154.28269-c7b7b42 -> git-24.342.06664-a42c53a
 luci-app-upnp: git-24.313.38121-76e4eca -> git-24.342.06664-a42c53a
 luci-lib-nixio: git-24.034.54875-21210dc -> git-24.342.06664-a42c53a
 luci-lib-jsonc: git-23.298.74571-62eb535 -> git-24.342.06664-a42c53a
 luci: git-23.051.66410-a505bb1 -> git-24.342.06664-a42c53a
 luci-light: git-23.024.33244-34dee82 -> git-24.342.06664-a42c53a
 luci-lib-uqr: git-24.299.73750-f2bbf4e -> git-24.342.06664-a42c53a
 luci-mod-network: git-24.281.58052-a6c2279 -> git-24.342.06664-a42c53a
 iperf3: 3.17.1-2 -> 3.17.1-4
unknown package '(null)' requested.
Bad message (74)
```
Is there a workaround?

howels, 2024-12-12:
Same here when upgrading 23.x via `auc`

```
Are you sure you want to continue the upgrade process? [N/y] y
unknown package '(null)' requested.
Bad message (77)
```

aparcar, 2024-12-12:
I'll see what's happening there and let you know

metlinux, 2024-12-28:
> I'll see what's happening there and let you know

Were you able to check ? Because I am not able either to use auc with my 23.05.5 version running on netgear,wax202 and tplink,eap615-wall-v1 and get the exact same error as Howels (package null, bad message 77)

aparcar, 2025-01-31:
I'll have a look tomorrow

metlinux, 2025-02-03:
> I'll have a look tomorrow

Where you able to check this time ??? To be honest you would already do so back in december ....
Can you please check and solve this so we can upgrade and make sure we are not running infected versions

aptinio, 2025-02-03:
@aparcar, feel free to go back on your word. According to the [openwrt license](https://openwrt.org/license) you are not obliged to provide support.

Thanks for all the work you provide for free. <3

aparcar, 2025-02-03:
Thanks @aptinio! It's in my interest to have this running well but yeah, it's always fun to see how people expect things

piojan, 2025-02-03:
There is one more option for upgrades: https://firmware-selector.openwrt.org/
You can customise installed packages. Download the image, backup config and do a manual sysupgrade with image and config.

aparcar, 2025-02-03:
I can't reproduce this, tested with rc2 and it builds fine:

```
Image available at https://sysupgrade.openwrt.org/store/9d9c4250b2065eeef1e64b17bd788d92b8459bd20c40516c4fb2e5939f97d886/
```

However, the error message comes from `auc` itself and is triggered by a HTTP 422 response

```c
  case 422:  
                fprintf(stderr, "unknown package '%s' requested.\n",
                        blobmsg_get_string(tb[H_UNKNOWN_PACKAGE]));
                rc=-ENOPKG;
                request_done(cl);
                break;  
```

The ASU server used to respond a 422 whenever a package was invalid, now it responds that whenever the request is invalid. A valid request is defined in the [docs](https://sysupgrade.openwrt.org/docs#/default/api_v1_build_post_api_v1_build_post). Right now I can't tell how `auc` confuses the server, since I can't reproduce it. If one of the error reporters are capable of doing so, please provide a log of the request i.e. via Wireshark or `tcpdump`. If not, please provide the full list of installed packages.

Apart from that, I'm not sure how to resolve this.

dangowrt, 2025-02-04:
You can build `auc` with `CONFIG_DEBUG=y` which gives you additional options to dump both headers and request bodies. That should definitely give you a clue.

aparcar, 2025-02-04:
Thanks, a single char fixed the issue. The plus sign wasn't accepted anymore in version strings. I'll add test cases "tomorrow" to make sure those legacy cases are correctly handled. It was a bit hard to track down since the kmods @tomporter518 (including the + sign) wouldn't show up in @piojan AUC request, maybe @dangowrt has an idea what could be the issue?

While at it, I tried to install all the packages mentioned in @tomporter518 report but three failed to install, I'm not sure if they were dropped mid release, renamed or something. @dangowrt do you have further ideas on the three packages below?

```
Collected errors:
 * opkg_install_cmd: Cannot install package kmod-usb-xhci-mtk.
 * opkg_install_cmd: Cannot install package ubi-utils.
 * opkg_install_cmd: Cannot install package kmod-sdhci-mt7620.
 ```

closed 2025-02-04 by aparcar

aparcar, 2025-02-04:
@dangowrt actually scratch that comment on the missing packages, those were fore ramips while I ran this on a x86 VM - sorry