openwrt/asu #1112: Issue with auc - Connection error: Connection failed

simarmannsingh, 2024-12-07:
I went through many issues related to auc, I can see this had been a persistent issue in the past.
Unfortunately, it has stuck me now.

```sh
root@OpenWrt:~# auc -n
auc/0.3.2-1
Server:    https://sysupgrade.openwrt.org
Running:   23.05.0 r23497-6637af95aa on ath79/generic (tplink,archer-c7-v5)
Connection error: Connection failed
No data available (61)
```

Doesn't work via Gui either. Request simply fails.

What I already checked:-
- if I'm connected to any VPN or other mesh network : NO
- if internet is available: YES
- if device is working as router: YES

Only the auc call fails.

Any suggestions?

popy2k14, 2024-12-07:
I think it has Todo with CVE-2024-54143: https://www.cve.org/CVERecord?id=CVE-2024-54143

I got two Mails regarding this from the openwrt mailing list explaining the issue. 

They are starting that is near to 0 chance that malicious images where built. The sysupgrade server was reinstalled.
Maybe that's why it is offline. 

In one mail there is an advise to just do an unattended sysupgrade again to the same version to be 100% sure.

Also for me it's unreachable since yesterday.

simarmannsingh, 2024-12-07:
@popy2k14 : Hey thanks for the information. It is very helpful. I was trying and it's still not reachable for me. I will keep trying, hoping the server is reachable again.

popy2k14, 2024-12-07:
You're welcome
Now it's reachable for me, but creating of images does not work. I get an init error.

```
Fehler beim Erstellen des Firmware-Images
Serverantwort: init
```

It's German for init error. 

Will try it tomorrow again.

h4de5, 2024-12-08:
seems to be working again. just xreated a new image - not yet tried to install it thou.

popy2k14, 2024-12-08:
Also created one.
Then after sysupgrade my nanopi r4s was dead, but it was my fault because I had expanded the root partition wrong before. Flashed the new image with again with Rufus, restored my backup and all is working.

P.S.: OT: is there a way to resize the root partition, without messing everything up on unattended sysupgrade?

Thx

simarmannsingh, 2024-12-09:
Thanks @popy2k14  and @h4de5 for responding. Yes, I can also confirm the server is back online. It must have been because of the CVE.

Hence, I will be closing the issue.

closed 2024-12-09 by simarmannsingh