openwrt/asu #1113: Luci ASU error when requesting image

HLiTE, 2024-12-07:
Here is the return :

Error building the firmware image
Server response: init

[Please report the error message and request](https://github.com/openwrt/asu/issues)
Request Data:

{
    "url": "https://sysupgrade.openwrt.org",
    "revision": "r24106-10cc5fcd00",
    "advanced_mode": "1",
    "sha256_unsigned": "",
    "branch": "23.05",
    "efi": null,
    "profile": "sophos,ap55",
    "target": "ath79/generic",
    "version": "23.05.5",
    "packages": [
        "ath10k-firmware-qca988x-ct",
        "auc",
        "base-files",
        "busybox",
        "ca-bundle",
        "collectd-mod-conntrack",
        "collectd-mod-df",
        "collectd-mod-ipstatistics",
        "collectd-mod-ping",
        "collectd-mod-sensors",
        "collectd-mod-wireless",
        "dnsmasq",
        "dropbear",
        "ethtool-full",
        "firewall4",
        "fstools",
        "gpioctl-sysfs",
        "gpiod-tools",
        "htop",
        "iftop",
        "iperf3",
        "kernel",
        "kmod-ath10k-ct",
        "kmod-ath9k",
        "kmod-gpio-button-hotplug",
        "kmod-nft-offload",
        "kmod-usb2",
        "lftp",
        "libc",
        "libgcc",
        "libustream-mbedtls",
        "logd",
        "lsof",
        "luci",
        "luci-app-attendedsysupgrade",
        "luci-app-firewall",
        "luci-app-opkg",
        "luci-app-statistics",
        "luci-app-uhttpd",
        "luci-base",
        "luci-compat",
        "luci-lib-base",
        "luci-lib-ip",
        "luci-lib-jsonc",
        "luci-lib-nixio",
        "luci-light",
        "luci-lua-runtime",
        "luci-mod-admin-full",
        "luci-mod-dashboard",
        "luci-mod-network",
        "luci-mod-status",
        "luci-mod-system",
        "luci-proto-ipv6",
        "luci-proto-ppp",
        "luci-ssl",
        "luci-theme-bootstrap",
        "msmtp",
        "mtd",
        "mutt",
        "netifd",
        "nftables-json",
        "odhcp6c",
        "odhcpd-ipv6only",
        "opkg",
        "pciutils",
        "ppp",
        "ppp-mod-pppoe",
        "procd",
        "procd-seccomp",
        "procd-ujail",
        "swconfig",
        "u2pnpd",
        "uboot-envtools",
        "uci",
        "uclient-fetch",
        "urandom-seed",
        "urngd",
        "usbutils",
        "wireless-regdb",
        "wpad-basic-mbedtls"
    ],
    "diff_packages": true,
    "filesystem": "squashfs",
    "client": "luci/git-24.338.58779-24d8916"
}****

HLiTE, 2024-12-07:
https://firmware-selector.openwrt.org return : "init" too

h4de5, 2024-12-07:
seems sysupgrade server are not yet up.
[image]

h4de5, 2024-12-07:
not sure if it helps - but the API (https://sysupgrade.openwrt.org/docs#/) also delivers an error:

```
Traceback (most recent call last):
File "/usr/local/lib/python3.12/site-packages/podman/api/uds.py", line 41, in connect\n super().connect(netloc)
ConnectionRefusedError: [Errno 111] Connection refused
The above exception was the direct cause of the following exception:
Traceback (most recent call last):
File "/usr/local/lib/python3.12/site-packages/urllib3/connectionpool.py", line 789, in urlopen\n response = self._make_request(\n ^^^^^^^^^^^^^^^^^^^
File "/usr/local/lib/python3.12/site-packages/urllib3/connectionpool.py", line 495, in _make_request\n conn.request(\n
File "/usr/local/lib/python3.12/site-packages/urllib3/connection.py", line 441, in request\n self.endheaders()\n
File "/usr/local/lib/python3.12/http/client.py", line 1331, in endheaders\n self._send_output(message_body, encode_chunked=encode_chunked)\n
File "/usr/local/lib/python3.12/http/client.py", line 1091, in _send_output\n self.send(msg)\n
File "/usr/local/lib/python3.12/http/client.py", line 1035, in send\n self.connect()\n
File "/usr/local/lib/python3.12/site-packages/podman/api/uds.py", line 86, in connect\n sock.connect()\n
File "/usr/local/lib/python3.12/site-packages/podman/api/uds.py", line 43, in connect\n raise APIError(f"Unable to make connection to UDS '{netloc}'") from e\npodman.errors.exceptions.APIError: Unable to make connection to UDS '/run/user/1000/podman/podman.sock'
During handling of the above exception, another exception occurred:
Traceback (most recent call last):
File "/usr/local/lib/python3.12/site-packages/requests/adapters.py", line 667, in send\n resp = conn.urlopen(\n ^^^^^^^^^^^^^
File "/usr/local/lib/python3.12/site-packages/urllib3/connectionpool.py", line 843, in urlopen\n retries = retries.increment(\n ^^^^^^^^^^^^^^^^^^^
File "/usr/local/lib/python3.12/site-packages/urllib3/util/retry.py", line 474, in increment\n raise reraise(type(error), error, _stacktrace)\n ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/usr/local/lib/python3.12/site-packages/urllib3/util/util.py", line 38, in reraise\n raise value.with_traceback(tb)\n
File "/usr/local/lib/python3.12/site-packages/urllib3/connectionpool.py", line 789, in urlopen\n response = self._make_request(\n ^^^^^^^^^^^^^^^^^^^
File "/usr/local/lib/python3.12/site-packages/urllib3/connectionpool.py", line 495, in _make_request\n conn.request(\n
File "/usr/local/lib/python3.12/site-packages/urllib3/connection.py", line 441, in request\n self.endheaders()\n
File "/usr/local/lib/python3.12/http/client.py", line 1331, in endheaders\n self._send_output(message_body, encode_chunked=encode_chunked)\n
File "/usr/local/lib/python3.12/http/client.py", line 1091, in _send_output\n self.send(msg)\n
File "/usr/local/lib/python3.12/http/client.py", line 1035, in send\n self.connect()\n
File "/usr/local/lib/python3.12/site-packages/podman/api/uds.py", line 86, in connect\n sock.connect()\n
File "/usr/local/lib/python3.12/site-packages/podman/api/uds.py", line 43, in connect\n raise APIError(f"Unable to make connection to UDS '{netloc}'") from e\nurllib3.exceptions.ProtocolError: ('Connection aborted.', APIError("Unable to make connection to UDS '/run/user/1000/podman/podman.sock'"))
During handling of the above exception, another exception occurred:
Traceback (most recent call last):
File "/usr/local/lib/python3.12/site-packages/podman/api/client.py", line 425, in _request\n self.request(\n
File "/usr/local/lib/python3.12/site-packages/requests/sessions.py", line 589, in request\n resp = self.send(prep, **send_kwargs)\n ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/usr/local/lib/python3.12/site-packages/requests/sessions.py", line 703, in send\n r = adapter.send(request, **kwargs)\n ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/usr/local/lib/python3.12/site-packages/requests/adapters.py", line 682, in send\n raise ConnectionError(err, request=request)\nrequests.exceptions.ConnectionError: ('Connection aborted.', APIError("Unable to make connection to UDS '/run/user/1000/podman/podman.sock'"))
The above exception was the direct cause of the following exception:
Traceback (most recent call last):
File "/usr/local/lib/python3.12/site-packages/rq/worker.py", line 1573, in perform_job\n rv = job.perform()\n ^^^^^^^^^^^^^
File "/usr/local/lib/python3.12/site-packages/rq/job.py", line 1322, in perform\n self._result = self._execute()\n ^^^^^^^^^^^^^^^
File "/usr/local/lib/python3.12/site-packages/rq/job.py", line 1356, in _execute\n result = self.func(*self.args, **self.kwargs)\n ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/app/asu/build.py", line 57, in build\n log.debug(f"Podman version: {podman.version()}\")\n ^^^^^^^^^^^^^^^^
File "/usr/local/lib/python3.12/site-packages/podman/client.py", line 204, in version\n return self.system.version(**kwargs)\n ^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/usr/local/lib/python3.12/site-packages/podman/domain/system.py", line 87, in version\n response = self.client.get("/version")\n ^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/usr/local/lib/python3.12/site-packages/podman/api/client.py", line 257, in get\n return self._request(\n ^^^^^^^^^^^^^
File "/usr/local/lib/python3.12/site-packages/podman/api/client.py", line 436, in _request\n raise APIError(uri.geturl(), explanation=f"{method.upper()} operation failed") from e\npodman.errors.exceptions.APIError: http://%2Frun%2Fuser%2F1000%2Fpodman%2Fpodman.sock/v5.2.0/libpod/version (GET operation failed)
```

HLiTE, 2024-12-07:
So we just have to wait they brings them up back ? No communications about that?

HLiTE, 2024-12-08:
Servers up again

closed 2024-12-08 by HLiTE