openwrt/asu #1122: sysupgrade.openwrt.org status

lorand-horvath, 2024-12-09:
@aparcar The ASU server is still unavailable. Is there an update? I'd like to get a fresh custom build via Firmware Selector or luci-app-attendedsysupgrade. Thanks!

aparcar, 2024-12-09:
The sysupgrade server works again on a fresh server installation. Please write more specific bug reports or else I can just respond with "works for me"

lorand-horvath, 2024-12-09:
@aparcar [It doesn't build ](https://firmware-selector.openwrt.org/) doesn't build for several days now. Same for the luci-app and owut. As far as I understand they are all frontends for sysupgrade.openwrt.org aka ASU, hence my report here.

benjaminlevine, 2024-12-09:
Facing the same issues. Like @lorand-horvath I'm typically stuck on validate manifest when building a snapshot for my Banana Pi R4.

However sometimes I instantly get a red error message 'init' with the following error (stderr is blank)

```
init
STDERR
STDOUT
Current Target: "mediatek/filogic"
Current Architecture: "aarch64"
Current Revision: "r28251-b4cc5743d9"
Default Packages: base-files ca-bundle dropbear fstools libc libgcc libustream-mbedtls logd mtd netifd uci uclient-fetch urandom-seed urngd kmod-leds-gpio kmod-gpio-button-hotplug fitblk kmod-phy-aquantia kmod-crypto-hw-safexcel wpad-basic-mbedtls uboot-envtools procd-ujail dnsmasq firewall4 nftables kmod-nft-offload odhcp6c odhcpd-ipv6only ppp ppp-mod-pppoe apk-mbedtls
Available Profiles:

abt_asr3000:
    ABT ASR3000
    Packages: kmod-mt7915e kmod-mt7981-firmware mt7981-wo-firmware
    hasImageMetadata: 1
    SupportedDevices: abt,asr3000
acelink_ew-7886cax:
    Acelink EW-7886CAX
    Packages: kmod-mt7915e kmod-mt7986-firmware mt7986-wo-firmware
    hasImageMetadata: 1
    SupportedDevices: acelink,ew-7886cax
acer_predator-w6:
    Acer Predator Connect W6
    Packages: kmod-usb3 kmod-mt7915e kmod-mt7916-firmware kmod-mt7986-firmware mt7986-wo-firmware e2fsprogs f2fsck mkf2fs
    hasImageMetadata: 1
    SupportedDevices: acer,predator-w6
acer_predator-w6d:
    Acer Predator Connect W6d
    Packages: kmod-usb3 kmod-mt7915e kmod-mt7916-firmware kmod-mt7986-firmware mt7986-wo-firmware e2fsprogs f2fsck mkf2fs
    hasImageMetadata: 1
    SupportedDevices: acer,predator-w6d
acer_vero-w6m:
    Acer Connect Vero W6m
    Packages: kmod-leds-ktd202x kmod-mt7915e kmod-mt7916-firmware kmod-mt7986-firmware mt7986-wo-firmware e2fsprogs f2fsck mkf2fs
    hasImageMetadata: 1
    SupportedDevices: acer,vero-w6m
smartrg_sdg-8612:
    Adtran SDG-8612
    Packages: e2fsprogs f2fsck mkf2fs kmod-hwmon-pwmfan kmod-mt7915e kmod-mt7986-firmware mt7986-wo-firmware
    hasImageMetadata: 1
    SupportedDevices: smartrg,sdg-8612
smartrg_sdg-8614:
    Adtran SDG-8614
    Packages: e2fsprogs f2fsck mkf2fs kmod-hwmon-pwmfan kmod-mt7915e kmod-mt7986-firmware mt7986-wo-firmware
    hasImageMetadata: 1
    SupportedDevices: smartrg,sdg-8614
smartrg_sdg-8622:
    Adtran SDG-8622
    Packages: e2fsprogs f2fsck mkf2fs kmod-hwmon-pwmfan kmod-mt7915e kmod-mt7915-firmware kmod-mt7986-firmware mt7986-wo-firmware
    hasImageMetadata: 1
    SupportedDevices: smartrg,sdg-8622
smartrg_sdg-8632:
    Adtran SDG-8632
    Packages: e2fsprogs f2fsck mkf2fs kmod-hwmon-pwmfan kmod-mt7915e kmod-mt7915-firmware kmod-mt7986-firmware mt7986-wo-firmware
    hasImageMetadata: 1
    SupportedDevices: smartrg,sdg-8632
smartrg_sdg-8733:
    Adtran SDG-8733
    Packages: e2fsprogs f2fsck mkf2fs kmod-hwmon-pwmfan kmod-mt7996-firmware kmod-phy-aquantia kmod-usb3 mt7988-wo-firmware
    hasImageMetadata: 1
    SupportedDevices: smartrg,sdg-8733
smartrg_sdg-8733a:
    Adtran SDG-8733A
    Packages: e2fsprogs f2fsck mkf2fs kmod-hwmon-pwmfan mt7988-2p5g-phy-firmware kmod-mt7996-233-firmware kmod-phy-aquantia mt7988-wo-firmware
    hasImageMetadata: 1
    SupportedDevices: smartrg,sdg-8733a
smartrg_sdg-8734:
[... 315 lines cut ...]
    Zbtlink ZBT-Z8103AX
    Packages: kmod-mt7915e kmod-mt7981-firmware mt7981-wo-firmware
    hasImageMetadata: 1
    SupportedDevices: zbtlink,zbt-z8103ax
zyxel_ex5601-t0-stock:
    Zyxel EX5601-T0 (stock layout)
    Packages: kmod-mt7915e kmod-mt7986-firmware mt7986-wo-firmware kmod-usb3
    hasImageMetadata: 1
    SupportedDevices: mediatek,mt7986a-rfb-snand
zyxel_ex5601-t0-ubootmod:
    Zyxel EX5601-T0 (OpenWrt U-Boot layout)
    Packages: kmod-mt7915e kmod-mt7986-firmware mt7986-wo-firmware kmod-usb3
    hasImageMetadata: 1
    SupportedDevices: zyxel,ex5601-t0-ubootmod
zyxel_ex5700-telenor:
    Zyxel EX5700 (Telenor)
    Packages: kmod-ubootenv-nvram kmod-usb3 kmod-mt7915e kmod-mt7916-firmware kmod-mt7986-firmware mt7986-wo-firmware
    hasImageMetadata: 1
    SupportedDevices: zyxel,ex5700-telenor
zyxel_nwa50ax-pro:
    Zyxel NWA50AX Pro
    Packages: kmod-mt7915e kmod-mt7981-firmware mt7981-wo-firmware zyxel-bootconfig
    hasImageMetadata: 1
    SupportedDevices: zyxel,nwa50ax-pro
````

aparcar, 2024-12-09:
Please give this another try. The fresh server installation deployed different `ulimit` values which caused `fakeroot` to stuck forever when wrapping APK. I modified the limits and in my test it worked, please report if this issue is still ongoing.

aparcar, 2024-12-09:
@anelam this happens when the container times out, please give it 24h so the build cleans up automatically or modify a package to trigger a fresh build
@benjaminlevine I just tested a SNAPSHOT build for the R4 and it worked

benjaminlevine, 2024-12-09:
Thanks for the feedback, did a little more playing around and it seems package dependent:

e.g. adding the package 'luci-app-adblock' builds fine
[image]

However, adding the package 'ethtool' causes the 'init' error:
[image]

closed 2024-12-10 by lorand-horvath

aparcar, 2024-12-10:
@benjaminlevine I think that's because a "broken" image is cached. Please give it 24 hours

lorand-horvath, 2024-12-10:
@benjaminlevine Just specify a random comment in the `Script to run on first boot` field, then hit `Request build` to bypass the cache, like this:

[image]