openwrt/asu #1123: google ac 1304 Attended system upgrade from 23.05.4 to 23.05.5 broken

Edwardwich, 2024-12-09:
{
    "url": "https://sysupgrade.openwrt.org",
    "revision": "r24012-d8dd03c46f",
    "advanced_mode": "0",
    "sha256_unsigned": "",
    "branch": "23.05",
    "efi": null,
    "profile": "google,wifi",
    "target": "ipq40xx/chromium",
    "version": "23.05.5",
    "packages": [
        "ath10k-board-qca4019",
        "ath10k-firmware-qca4019-ct",
        "auc",
        "base-files",
        "brook",
        "busybox",
        "ca-bundle",
        "cfdisk",
        "cgi-io",
        "dnsmasq-full",
        "dropbear",
        "e2fsprogs",
        "firewall4",
        "fstools",
        "fwtool",
        "getrandom",
        "hostapd-common",
        "htop",
        "hysteria",
        "iw",
        "iwinfo",
        "jansson",
        "jshn",
        "jsonfilter",
        "kernel",
        "kmod-ath",
        "kmod-ath10k-ct",
        "kmod-cfg80211",
        "kmod-crypto-acompress",
        "kmod-crypto-aead",
        "kmod-crypto-ccm",
        "kmod-crypto-cmac",
        "kmod-crypto-crc32c",
        "kmod-crypto-ctr",
        "kmod-crypto-gcm",
        "kmod-crypto-gf128",
        "kmod-crypto-ghash",
        "kmod-crypto-hash",
        "kmod-crypto-hmac",
        "kmod-crypto-manager",
        "kmod-crypto-null",
        "kmod-crypto-rng",
        "kmod-crypto-seqiv",
        "kmod-crypto-sha512",
        "kmod-fs-ext4",
        "kmod-fs-f2fs",
        "kmod-google-firmware",
        "kmod-gpio-button-hotplug",
        "kmod-hwmon-core",
        "kmod-leds-gpio",
        "kmod-lib-crc-ccitt",
        "kmod-lib-crc32c",
        "kmod-lib-lzo",
        "kmod-mac80211",
        "kmod-nf-conntrack",
        "kmod-nf-conntrack6",
        "kmod-nf-flow",
        "kmod-nf-log",
        "kmod-nf-log6",
        "kmod-nf-nat",
        "kmod-nf-reject",
        "kmod-nf-reject6",
        "kmod-nfnetlink",
        "kmod-nft-core",
        "kmod-nft-fib",
        "kmod-nft-nat",
        "kmod-nft-offload",
        "kmod-nft-socket",
        "kmod-nft-tproxy",
        "kmod-nls-base",
        "kmod-ppp",
        "kmod-pppoe",
        "kmod-pppox",
        "kmod-ramoops",
        "kmod-slhc",
        "kmod-usb-core",
        "kmod-usb-dwc3",
        "kmod-usb-dwc3-qcom",
        "kmod-usb-xhci-hcd",
        "kmod-usb3",
        "libblobmsg-json",
        "libiwinfo",
        "libiwinfo-data",
        "libjson-c",
        "libjson-script",
        "liblucihttp",
        "liblucihttp-ucode",
        "libmbedtls",
        "libmnl",
        "libnftnl",
        "libnl-tiny",
        "libubox",
        "libubus",
        "libuci",
        "libuclient",
        "libucode",
        "libustream-mbedtls",
        "logd",
        "luci",
        "luci-app-argon-config",
        "luci-app-attendedsysupgrade",
        "luci-app-firewall",
        "luci-app-opkg",
        "luci-app-passwall",
        "luci-app-passwall2",
        "luci-base",
        "luci-lib-ipkg",
        "luci-light",
        "luci-mod-admin-full",
        "luci-mod-network",
        "luci-mod-status",
        "luci-mod-system",
        "luci-proto-ipv6",
        "luci-proto-ppp",
        "luci-ssl",
        "luci-theme-argon",
        "luci-theme-bootstrap",
        "mkf2fs",
        "mtd",
        "netifd",
        "nftables-json",
        "odhcp6c",
        "odhcpd-ipv6only",
        "openwrt-keyring",
        "opkg",
        "partx-utils",
        "ppp",
        "ppp-mod-pppoe",
        "procd",
        "procd-seccomp",
        "procd-ujail",
        "px5g-mbedtls",
        "resize2fs",
        "rpcd",
        "rpcd-mod-file",
        "rpcd-mod-iwinfo",
        "rpcd-mod-luci",
        "rpcd-mod-rrdns",
        "rpcd-mod-ucode",
        "sing-box",
        "trojan-go",
        "tune2fs",
        "ubi-utils",
        "uboot-envtools",
        "ubox",
        "ubus",
        "ubusd",
        "uci",
        "uclient-fetch",
        "ucode",
        "ucode-mod-fs",
        "ucode-mod-html",
        "ucode-mod-math",
        "ucode-mod-nl80211",
        "ucode-mod-rtnl",
        "ucode-mod-ubus",
        "ucode-mod-uci",
        "ucode-mod-uloop",
        "uhttpd",
        "uhttpd-mod-ubus",
        "urandom-seed",
        "urngd",
        "usign",
        "v2ray-core",
        "v2ray-geosite-ir",
        "wireless-regdb",
        "wpad-basic-mbedtls",
        "xray-core"
    ],
    "diff_packages": true,
    "filesystem": "squashfs",
    "client": "luci/git-24.342.06664-a42c53a"
}

Edwardwich, 2024-12-09:
Generate local signing keys...
Generate local certificate...
Package list missing or not up-to-date, generating it.

Building package index...
Downloading https://downloads.openwrt.org/releases/23.05.5/targets/ipq40xx/chromium/packages/Packages.gz
Updated list of available packages in /builder/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/root-ipq40xx/../../../../builder/dl/openwrt_core
Downloading https://downloads.openwrt.org/releases/23.05.5/targets/ipq40xx/chromium/packages/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/23.05.5/packages/arm_cortex-a7_neon-vfpv4/base/Packages.gz
Updated list of available packages in /builder/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/root-ipq40xx/../../../../builder/dl/openwrt_base
Downloading https://downloads.openwrt.org/releases/23.05.5/packages/arm_cortex-a7_neon-vfpv4/base/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/23.05.5/packages/arm_cortex-a7_neon-vfpv4/luci/Packages.gz
Updated list of available packages in /builder/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/root-ipq40xx/../../../../builder/dl/openwrt_luci
Downloading https://downloads.openwrt.org/releases/23.05.5/packages/arm_cortex-a7_neon-vfpv4/luci/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/23.05.5/packages/arm_cortex-a7_neon-vfpv4/packages/Packages.gz
Updated list of available packages in /builder/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/root-ipq40xx/../../../../builder/dl/openwrt_packages
Downloading https://downloads.openwrt.org/releases/23.05.5/packages/arm_cortex-a7_neon-vfpv4/packages/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/23.05.5/packages/arm_cortex-a7_neon-vfpv4/routing/Packages.gz
Updated list of available packages in /builder/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/root-ipq40xx/../../../../builder/dl/openwrt_routing
Downloading https://downloads.openwrt.org/releases/23.05.5/packages/arm_cortex-a7_neon-vfpv4/routing/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/23.05.5/packages/arm_cortex-a7_neon-vfpv4/telephony/Packages.gz
Updated list of available packages in /builder/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/root-ipq40xx/../../../../builder/dl/openwrt_telephony
Downloading https://downloads.openwrt.org/releases/23.05.5/packages/arm_cortex-a7_neon-vfpv4/telephony/Packages.sig
Signature check passed.
Downloading file:packages/Packages
Updated list of available packages in /builder/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/root-ipq40xx/../../../../builder/dl/imagebuilder
Downloading file:packages/Packages.sig
Signature check passed.
Collected errors:
 * opkg_install_cmd: Cannot install package brook.
 * opkg_install_cmd: Cannot install package hysteria.
 * opkg_install_cmd: Cannot install package luci-app-argon-config.
 * opkg_install_cmd: Cannot install package luci-app-passwall.
 * opkg_install_cmd: Cannot install package luci-app-passwall2.
 * opkg_install_cmd: Cannot install package luci-theme-argon.
make[2]: *** [Makefile:189: package_install] Error 255
make[1]: *** [Makefile:154: _call_manifest] Error 2
make: *** [Makefile:274: manifest] Error 2

efahl, 2024-12-09:
I don't see any of those packages in the OpenWrt repositories.  Did you install them from somewhere else?

Edwardwich, 2024-12-09:
> I don't see any of those packages in the OpenWrt repositories.  Did you install them from somewhere else?

I buy this router with pre installed OpenWrt

Edwardwich, 2024-12-09:
I use this router for v2ray config

efahl, 2024-12-10:
That explains it.  We can't help you with a vendor-specific version, you'll need to contact whoever installed OpenWrt with those non-standard packages.

aparcar, 2024-12-10:
Out of curiosity, who sold you this? If this is done semi professionally they should provide their own mechanism of firmware upgrades or collaborate with upstream.

Edwardwich, 2024-12-10:
> Out of curiosity, who sold you this? If this is done semi professionally they should provide their own mechanism of firmware upgrades or collaborate with upstream. 

This device very hard to install OpenWrt in it. And I buy this with OpenWrt 

And for now I have problem with 2.4Ghz WiFi AP it's tern off after few minutes

Edwardwich, 2024-12-10:
[image]

Edwardwich, 2024-12-10:
[image]

Edwardwich, 2024-12-11:
@aparcar 
@aparcar 
Any way to help for fix this issue?

efahl, 2024-12-11:
As stated before, this is not an OpenWrt issue, it is one created by your vendor.  You must contact that vendor and see if they can help.  Since you won't even tell us who that is, there's no way we can help.

Edwardwich, 2024-12-11:
> As stated before, this is not an OpenWrt issue, it is one created by your vendor.  You must contact that vendor and see if they can help.  Since you won't even tell us who that is, there's no way we can help.

What information U need to have 
Idk 
I have access to the seller, but I don't know what to ask him?

Edwardwich, 2024-12-12:
https://brook.app/
https://v1.hysteria.network/docs/installation/
https://github.com/jerrykuku/luci-app-argon-config
https://github.com/jerrykuku/luci-theme-argon

@efahl I report this problem to seller and I wait for response 
and also search and find some of the package he install but its dosnt added to Opkg it can help ?

Edwardwich, 2024-12-15:
@efahl I rest fully my openwrt and install passwall agin :

Generate local signing keys...
Generate local certificate...
Package list missing or not up-to-date, generating it.

Building package index...
Downloading https://downloads.openwrt.org/releases/23.05.5/targets/ipq40xx/chromium/packages/Packages.gz
Updated list of available packages in /builder/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/root-ipq40xx/../../../../builder/dl/openwrt_core
Downloading https://downloads.openwrt.org/releases/23.05.5/targets/ipq40xx/chromium/packages/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/23.05.5/packages/arm_cortex-a7_neon-vfpv4/base/Packages.gz
Updated list of available packages in /builder/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/root-ipq40xx/../../../../builder/dl/openwrt_base
Downloading https://downloads.openwrt.org/releases/23.05.5/packages/arm_cortex-a7_neon-vfpv4/base/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/23.05.5/packages/arm_cortex-a7_neon-vfpv4/luci/Packages.gz
Updated list of available packages in /builder/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/root-ipq40xx/../../../../builder/dl/openwrt_luci
Downloading https://downloads.openwrt.org/releases/23.05.5/packages/arm_cortex-a7_neon-vfpv4/luci/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/23.05.5/packages/arm_cortex-a7_neon-vfpv4/packages/Packages.gz
Updated list of available packages in /builder/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/root-ipq40xx/../../../../builder/dl/openwrt_packages
Downloading https://downloads.openwrt.org/releases/23.05.5/packages/arm_cortex-a7_neon-vfpv4/packages/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/23.05.5/packages/arm_cortex-a7_neon-vfpv4/routing/Packages.gz
Updated list of available packages in /builder/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/root-ipq40xx/../../../../builder/dl/openwrt_routing
Downloading https://downloads.openwrt.org/releases/23.05.5/packages/arm_cortex-a7_neon-vfpv4/routing/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/23.05.5/packages/arm_cortex-a7_neon-vfpv4/telephony/Packages.gz
Updated list of available packages in /builder/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/root-ipq40xx/../../../../builder/dl/openwrt_telephony
Downloading https://downloads.openwrt.org/releases/23.05.5/packages/arm_cortex-a7_neon-vfpv4/telephony/Packages.sig
Signature check passed.
Downloading file:packages/Packages
Updated list of available packages in /builder/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/root-ipq40xx/../../../../builder/dl/imagebuilder
Downloading file:packages/Packages.sig
Signature check passed.
Collected errors:
 * opkg_install_cmd: Cannot install package chinadns-ng.
 * opkg_install_cmd: Cannot install package dns2socks.
 * opkg_install_cmd: Cannot install package dns2tcp.
 * opkg_install_cmd: Cannot install package geoview.
 * opkg_install_cmd: Cannot install package hysteria.
 * opkg_install_cmd: Cannot install package luci-app-passwall2.
 * opkg_install_cmd: Cannot install package luci-theme-argon.
 * opkg_install_cmd: Cannot install package naiveproxy.
 * opkg_install_cmd: Cannot install package shadowsocks-rust-sslocal.
 * opkg_install_cmd: Cannot install package shadowsocks-rust-ssserver.
 * opkg_install_cmd: Cannot install package shadowsocksr-libev-ssr-local.
 * opkg_install_cmd: Cannot install package shadowsocksr-libev-ssr-redir.
 * opkg_install_cmd: Cannot install package shadowsocksr-libev-ssr-server.
 * opkg_install_cmd: Cannot install package tcping.
 * opkg_install_cmd: Cannot install package tuic-client.
 * opkg_install_cmd: Cannot install package v2ray-plugin.
make[2]: *** [Makefile:189: package_install] Error 255
make[1]: *** [Makefile:154: _call_manifest] Error 2
make: *** [Makefile:274: manifest] Error 2

[image]