openwrt/asu #1135: ASU sysupgrade.openwrt.org down or malfunctioning

lorand-horvath, 2024-12-22:
@aparcar @efahl FYI, ASU is down or not working properly. Can't get a snapshot, it doesn't do anything on build request.

efahl, 2024-12-22:
Yeah, this is a weird one.  No code changes in over a week, I'm running all the latest code on my home server and it's fine.

I do some `curl` posts against the production server and it appears the front end validation is working, but the backend build is failing for some reason.

```
$ curl --json '{"version":"SNAPSHOT","target":"x86/64"}' https://sysupgrade.openwrt.org/api/v1/build
{"detail":[{"type":"missing","loc":["body","profile"],"msg":"Field required","input":{"version":"SNAPSHOT","target":"x86/64"}}]}

$ curl --json '{"version":"SNAPSHOT","target":"x86/64","profile":"generic"}' https://sysupgrade.openwrt.org/api/v1/build
Internal Server Error
```

lorand-horvath, 2024-12-29:
/wontfix

lorand-horvath, 2024-12-29:
**wontfix**

closed 2024-12-29 by lorand-horvath