openwrt/asu #1136: firmware build request failed (Unsupported profile: linksys_e8450

Fentanest, 2024-12-22:
[image]

packages included :
{auc base-files bind-host bind-libs busybox ca-bundle cgi-io curl ddns-scripts ddns-scripts-cloudflare ddns-scripts-services dnsmasq-full dropbear firewall4 fstools fwtool getrandom https-dns-proxy iw iwinfo jansson jshn jsonfilter kernel kmod-cfg80211 kmod-crypto-aead kmod-crypto-ccm kmod-crypto-cmac kmod-crypto-crc32c kmod-crypto-ctr kmod-crypto-gcm kmod-crypto-gf128 kmod-crypto-ghash kmod-crypto-hash kmod-crypto-hmac kmod-crypto-manager kmod-crypto-null kmod-crypto-rng kmod-crypto-seqiv kmod-crypto-sha256 kmod-gpio-button-hotplug kmod-hwmon-core kmod-leds-gpio kmod-lib-crc-ccitt kmod-lib-crc32c kmod-mac80211 kmod-mt76-connac kmod-mt76-core kmod-mt7615-common kmod-mt7615-firmware kmod-mt7615e kmod-mt7622-firmware kmod-mt7915-firmware kmod-mt7915e kmod-nf-conntrack kmod-nf-conntrack6 kmod-nf-flow kmod-nf-log kmod-nf-log6 kmod-nf-nat kmod-nf-reject kmod-nf-reject6 kmod-nfnetlink kmod-nft-core kmod-nft-fib kmod-nft-nat kmod-nft-offload kmod-ppp kmod-pppoe kmod-pppox kmod-slhc kmod-usb3 libblobmsg-json libc libcurl libgcc libiwinfo libiwinfo-data libiwinfo-lua libjson-c libjson-script liblua liblucihttp liblucihttp-lua libmbedtls libmnl libnftnl libnl-tiny libopenssl libubox libubus libubus-lua libuci libuclient libucode libustream-mbedtls libuv libwolfssl logd lua luci luci-app-attendedsysupgrade luci-app-ddns luci-app-firewall luci-app-https-dns-proxy luci-app-opkg luci-app-wol luci-base luci-compat luci-lib-base luci-lib-ip luci-lib-jsonc luci-lib-nixio luci-mod-admin-full luci-mod-dashboard luci-mod-network luci-mod-status luci-mod-system luci-proto-ipv6 luci-proto-ppp luci-ssl luci-theme-bootstrap luci-theme-material mtd nano netifd nftables-json odhcp6c odhcpd-ipv6only openwrt-keyring opkg ppp ppp-mod-pppoe procd procd-seccomp procd-ujail rpcd rpcd-mod-file rpcd-mod-iwinfo rpcd-mod-luci rpcd-mod-rrdns ubi-utils uboot-envtools ubox ubus ubusd uci uclient-fetch ucode ucode-mod-fs ucode-mod-math ucode-mod-nl80211 ucode-mod-rtnl ucode-mod-ubus ucode-mod-uci ucode-mod-uloop uhttpd uhttpd-mod-ubus urandom-seed urngd usign wireless-regdb wpad-basic-mbedtls base-files fitblk nftables}

`ubus` call system board

"kernel": "5.15.167",
        "hostname": "FentaNET",
        "system": "ARMv8 Processor rev 4",
        "model": "Linksys E8450 (UBI)",
        "board_name": "linksys,e8450-ubi",
        "rootfs_type": "squashfs",
        "release": {
                "distribution": "OpenWrt",
                "version": "23.05.5",
                "revision": "r24106-10cc5fcd00",
                "target": "mediatek/mt7622",
                "description": "OpenWrt 23.05.5 r24106-10cc5fcd00"

I used to build request feature to identify packages that couldn't be installed in the next version. I would then find and replace those packages one by one with alternatives, rebuild the firmware, and proceed with the update.

However, this time, I am only encountering "unsupported device" errors, making it impossible to proceed.

vincejv, 2024-12-24:
Same issue with xiaomi_mi-router-ax3000t

vincejv, 2024-12-25:
works again, can you try @Fentanest ?

Fentanest, 2024-12-26:
> works again, can you try @Fentanest ?

it works. really appreciate @vincejv

closed 2024-12-26 by Fentanest