openwrt/asu #1142: Failed to request firmware package

StephenJose-Dai, 2025-01-01:
**I tried to update my system, in other words, to package an image. In the luci interface。
But there seems to be some problems, please see my pictures**

[image]

[image]

[image]

[image]

[image]

**I don't know what happened, what should I do next to fix this?**

efahl, 2025-01-07:
Many of those packages simply don't exist in OpenWrt.  How did you get, for example, simple-obfs or tuic-client?  I've never heard of them and can't find any hint about them on the OpenWrt repositories.

StephenJose-Dai, 2025-01-12:
I'm getting them from passwall's branch > Many of those packages simply don't exist in OpenWrt.  How did you get, for example, simple-obfs or tuic-client?  I've never heard of them and can't find any hint about them on the OpenWrt repositories.

StephenJose-Dai, 2025-01-12:
I'm getting them from passwall's branch

efahl, 2025-01-13:
ASU server doesn't yet fully support external package repos, so you'll need to build an image with https://firmware-selector.openwrt.org, removing just those packages you got from passwall, then install those after you upgrade.

Could you list and copy your feeds files here, so I can try some things out?  You can just ssh into your router, do an `ls` and  `cat` the files, like this:
```
$ ls -la /etc/opkg/
drwxr-xr-x    1 root     root          1024 Sep 23 05:34 .
drwxr-xr-x    1 root     root          1024 Jan 10 07:01 ..
-rw-r--r--    1 root     root           103 Sep 23 05:34 customfeeds.conf
-rw-r--r--    1 root     root           555 Sep 23 05:34 distfeeds.conf
drwxr-xr-x    1 root     root          1024 Sep 30 13:24 keys

$ cat /etc/opkg/*conf
# add your custom package feeds here
#
# src/gz example_feed_name http://www.example.com/path/to/files
src/gz openwrt_core https://downloads.openwrt.org/releases/23.05.5/targets/x86/64/packages
src/gz openwrt_base https://downloads.openwrt.org/releases/23.05.5/packages/x86_64/base
src/gz openwrt_luci https://downloads.openwrt.org/releases/23.05.5/packages/x86_64/luci
src/gz openwrt_packages https://downloads.openwrt.org/releases/23.05.5/packages/x86_64/packages
src/gz openwrt_routing https://downloads.openwrt.org/releases/23.05.5/packages/x86_64/routing
src/gz openwrt_telephony https://downloads.openwrt.org/releases/23.05.5/packages/x86_64/telephony
```

StephenJose-Dai, 2025-01-15:
of course！

[image]