openwrt/asu #1147: Unable to upgrade to the latest snapshot

taikn, 2025-01-08:
**Server response:** 
Error: Impossible package selection

**Request Data:**
```
{
    "url": "https://sysupgrade.openwrt.org",
    "branch": "SNAPSHOT",
    "revision": "r28256-1c274227b0",
    "efi": null,
    "advanced_mode": "0",
    "request_hash": "4380e580500dfd1e10e05851ea20fe2eb2664d014da9752591903a652a0511a9",
    "sha256_unsigned": "",
    "client": "luci/25.002.60082~f001a35",
    "packages": {
        "apk-mbedtls": "3.0.0_pre20241130-r2",
        "base-files": "1643~1c274227b0",
        "busybox": "1.37.0-r4",
        "ca-bundle": "20240203-r1",
        "dnscrypt-proxy2": "2.1.5-r1",
        "dnsmasq": "2.90-r3",
        "dropbear": "2024.86-r1",
        "e2fsprogs": "1.47.0-r2",
        "ethtool": "6.10-r1",
        "firewall4": "2024.12.18~18fc0ead-r1",
        "fstools": "2024.12.02~49d36ba2-r1",
        "fwtool": "2019.11.12~8f7fe925-r1",
        "getrandom": "2024.04.26~85f10530-r1",
        "jansson4": "2.14-r3",
        "jshn": "2024.03.29~eb9bcb64-r1",
        "jsonfilter": "2024.01.23~594cfa86-r1",
        "kernel": "6.6.63~cfdd0807e1d80c785c5041ab3495769f-r1",
        "kmod-crypto-acompress": "6.6.63-r1",
        "kmod-crypto-crc32c": "6.6.63-r1",
        "kmod-crypto-hash": "6.6.63-r1",
        "kmod-gpio-button-hotplug": "6.6.63-r5",
        "kmod-i2c-gpio": "6.6.63-r1",
        "kmod-lib-crc-ccitt": "6.6.63-r1",
        "kmod-lib-crc32c": "6.6.63-r1",
        "kmod-lib-lzo": "6.6.63-r1",
        "kmod-nf-conntrack": "6.6.63-r1",
        "kmod-nf-conntrack6": "6.6.63-r1",
        "kmod-nf-flow": "6.6.63-r1",
        "kmod-nf-log": "6.6.63-r1",
        "kmod-nf-log6": "6.6.63-r1",
        "kmod-nf-nat": "6.6.63-r1",
        "kmod-nf-reject": "6.6.63-r1",
        "kmod-nf-reject6": "6.6.63-r1",
        "kmod-nfnetlink": "6.6.63-r1",
        "kmod-nft-core": "6.6.63-r1",
        "kmod-nft-fib": "6.6.63-r1",
        "kmod-nft-nat": "6.6.63-r1",
        "kmod-nft-offload": "6.6.63-r1",
        "kmod-ppp": "6.6.63-r1",
        "kmod-pppoe": "6.6.63-r1",
        "kmod-pppox": "6.6.63-r1",
        "kmod-slhc": "6.6.63-r1",
        "libblkid1": "2.40.2-r1",
        "libblobmsg-json20240329": "2024.03.29~eb9bcb64-r1",
        "libc": "1.2.5-r4",
        "libcomerr0": "1.47.0-r2",
        "libe2p2": "1.47.0-r2",
        "libext2fs2": "1.47.0-r2",
        "libf2fs6": "1.16.0-r3",
        "libgcc1": "13.3.0-r4",
        "libjson-c5": "0.18-r1",
        "libjson-script20240329": "2024.03.29~eb9bcb64-r1",
        "libmbedtls21": "3.6.2-r1",
        "libmnl0": "1.0.5-r1",
        "libnftnl11": "1.2.8-r1",
        "libnl-tiny1": "2023.12.05~965c4bf4-r1",
        "libpthread": "1.2.5-r4",
        "librt": "1.2.5-r4",
        "libsmartcols1": "2.40.2-r1",
        "libss2": "1.47.0-r2",
        "libubox20240329": "2024.03.29~eb9bcb64-r1",
        "libubus20241020": "2024.10.20~252a9b0c-r1",
        "libuci20130104": "2024.11.26~10f7996e-r1",
        "libuclient20201210": "2024.10.22~88ae8f20-r1",
        "libucode20230711": "2024.12.06~209f041f-r1",
        "libudebug": "2023.12.06~6d3f51f9",
        "libustream-mbedtls20201210": "2024.07.28~99bd3d2b-r1",
        "libuuid1": "2.40.2-r1",
        "logd": "2024.04.26~85f10530-r1",
        "luci": "25.002.60082~f001a35",
        "luci-app-attendedsysupgrade": "25.002.60082~f001a35",
        "luci-app-sqm": "25.002.60082~f001a35",
        "mkf2fs": "1.16.0-r3",
        "mtd": "26",
        "netifd": "2024.11.27~768027c5-r2",
        "nftables-json": "1.1.1-r1",
        "odhcp6c": "2024.09.25~b6ae9ffa-r1",
        "odhcpd-ipv6only": "2024.05.08~a2988231-r1",
        "openwrt-keyring": "2024.11.01~fbae29d7-r1",
        "owut": "2024.12.10~e38844ae-r1",
        "partx-utils": "2.40.2-r1",
        "ppp": "2.5.1-r1",
        "ppp-mod-pppoe": "2.5.1-r1",
        "procd": "2024.11.13~7330fa55-r1",
        "procd-seccomp": "2024.12.22~42d39376-r1",
        "procd-ujail": "2024.11.13~7330fa55-r1",
        "python3": "3.11.10-r1",
        "ubi-utils": "2.2.1-r1",
        "uboot-envtools": "2024.07-r1",
        "ubox": "2024.04.26~85f10530-r1",
        "ubus": "2024.10.20~252a9b0c-r1",
        "ubusd": "2025.01.02~afa57cce-r1",
        "uci": "2024.11.26~10f7996e-r1",
        "uclient-fetch": "2024.10.22~88ae8f20-r1",
        "ucode": "2024.12.06~209f041f-r1",
        "ucode-mod-fs": "2024.12.06~209f041f-r1",
        "ucode-mod-ubus": "2024.12.06~209f041f-r1",
        "ucode-mod-uci": "2024.12.06~209f041f-r1",
        "urandom-seed": "3",
        "urngd": "2023.11.01~44365eb1-r1",
        "usign": "2020.05.23~f1f65026-r1",
        "wget-ssl": "1.24.5-r2",
        "yafut": "2024.06.10~38439f8a-r1",
        "zlib": "1.3.1-r1"
    },
    "profile": "mikrotik,rb5009",
    "target": "mvebu/cortexa72",
    "version": "SNAPSHOT",
    "diff_packages": true,
    "filesystem": "squashfs"
}
```
**STDERR:**
```
Generate local signing keys...
WARNING: can't open config file: /builder/shared-workdir/build/staging_dir/host/etc/ssl/openssl.cnf
WARNING: can't open config file: /builder/shared-workdir/build/staging_dir/host/etc/ssl/openssl.cnf
read EC key
writing EC key
fetch https://downloads.openwrt.org/snapshots/targets/mvebu/cortexa72/packages/packages.adb
fetch https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a72/base/packages.adb
fetch https://downloads.openwrt.org/snapshots/targets/mvebu/cortexa72/kmods/6.6.69-1-2e0e0bbe3d2b95bdff65bf5ccf25ddbf/packages.adb
fetch https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a72/luci/packages.adb
fetch https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a72/packages/packages.adb
fetch https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a72/routing/packages.adb
fetch https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a72/telephony/packages.adb
fetch https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a72/video/packages.adb
WARNING: opening packages/packages.adb: No such file or directory
Package list missing or not up-to-date, generating it.

Building package index...
ERROR: unable to select packages:
  libubus20241020 (no such package):
    required by: world[libubus20241020]
make[2]: *** [Makefile:225: package_install] Error 1
make[1]: *** [Makefile:166: _call_manifest] Error 2
make: *** [Makefile:340: manifest] Error 2
```

aparcar, 2025-01-08:
@efahl do you think we need another package change thing?

efahl, 2025-01-08:
No, that's the apk missing-ABI-named version issue rearing its ugly head (`libubus` ABI version has jumped from 20241020 to 20250102).  @Ansuel has that new issue in on apk-tools (https://gitlab.alpinelinux.org/alpine/apk-tools/-/issues/11066) that should give us the solution to this one when it gets through the pipe, but it's not an ASU issue directly.  (I'll be rewriting the `ubus call rpc-sys packagelist` code again, as soon as we have ABI versions in the adb files.)

@taikn You probably can't use LuCI app to solve this, you need to either use Firmware Selector where you'll replace any occurrence of `libubus20241020` with just `libubus`, or use `owut download --remove libubus20241020` from the device on the command line.  (More details at https://forum.openwrt.org/t/the-openwrt-firmware-selector/81721/1194 )

closed 2025-01-09 by taikn

senposage, 2025-01-21:
I was able to use the advanced options in the firmware selector and replace libubus***** and libuci***** with just libubus and libuci
 I thought we where beyond package names causing issues in 2025