openwrt/asu #1161: Error building firmware image

Ritew, 2025-01-24:
### Server response: Unsupported profile: xiaomi, mi-router-4-a-gigabit

Request data:
```
{
    "url": "https://sysupgrade.openwrt.org",
    "branch": "24.10",
    "revision": "r28304-6dacba30a7",
    "efi": null,
    "advanced_mode": "0",
    "request_hash": "",
    "sha256_unsigned": "",
    "client": "luci/25.021.30718~36b6107",
    "packages": {
        "iwinfo": "2024.10.20~b94f066e-r1",
        "tcpdump-mini": "4.99.5-r1",
        "luci-app-adblock": "25.021.30718~36b6107",
        "opkg": "2024.10.16~38eccbb1-r1",
        "wpad-basic-mbedtls": "2024.09.15~5ace39b0-r2",
        "libpthread": "1.2.5-r4",
        "rpcd": "2024.09.17~9f4b86e7-r1",
        "libiwinfo": "2024.10.20~b94f066e-r1",
        "kmod-nft-fib": "6.6.69-r1",
        "kmod-nfnetlink": "6.6.69-r1",
        "kmod-nf-reject6": "6.6.69-r1",
        "kmod-crypto-aead": "6.6.69-r1",
        "luci-mod-system": "25.021.30718~36b6107",
        "kmod-nf-flow": "6.6.69-r1",
        "kmod-lib-crc-ccitt": "6.6.69-r1",
        "px5g-mbedtls": "11",
        "getrandom": "2024.04.26~85f10530-r1",
        "luci-theme-bootstrap": "25.021.30718~36b6107",
        "kmod-pppoe": "6.6.69-r1",
        "luci-i18n-sqm-pt-br": "25.021.30718~36b6107",
        "kmod-pppox": "6.6.69-r1",
        "kmod-nf-reject": "6.6.69-r1",
        "procd-ujail": "2024.12.22~42d39376-r1",
        "base-files": "1649~6dacba30a7",
        "kmod-nf-nat": "6.6.69-r1",
        "kmod-crypto-crc32c": "6.6.69-r1",
        "libustream-mbedtls": "2024.07.28~99bd3d2b-r1",
        "firewall4": "2024.12.18~18fc0ead-r1",
        "adblock": "4.2.3-r3",
        "dnsmasq": "2.90-r3",
        "rpcd-mod-ucode": "2024.09.17~9f4b86e7-r1",
        "kmod-crypto-acompress": "6.6.69-r1",
        "ucode-mod-math": "2024.07.22~b610860d-r3",
        "kmod-lib-crc32c": "6.6.69-r1",
        "luci-mod-status": "25.021.30718~36b6107",
        "kmod-nft-nat": "6.6.69-r1",
        "luci-app-firewall": "25.021.30718~36b6107",
        "libmbedtls": "3.6.2-r1",
        "luci-i18n-adblock-pt-br": "25.021.30718~36b6107",
        "luci-i18n-package-manager-pt-br": "25.021.30718~36b6107",
        "libmnl": "1.0.5-r1",
        "jansson": "2.14-r3",
        "odhcp6c": "2024.09.25~b6ae9ffa-r1",
        "uclient-fetch": "2024.10.22~88ae8f20-r1",
        "uci": "2025.01.17~fb3c2343-r1",
        "luci-ssl": "25.021.30718~36b6107",
        "dropbear": "2024.86-r1",
        "libnftnl": "1.2.8-r1",
        "rpcd-mod-file": "2024.09.17~9f4b86e7-r1",
        "mtd": "26",
        "odhcpd-ipv6only": "2024.05.08~a2988231-r1",
        "kmod-crypto-authenc": "6.6.69-r1",
        "luci-app-sqm": "25.021.30718~36b6107",
        "kmod-nf-log": "6.6.69-r1",
        "luci-app-package-manager": "25.021.30718~36b6107",
        "urandom-seed": "3",
        "luci-proto-ppp": "25.021.30718~36b6107",
        "ppp": "2.5.1-r1",
        "luci-mod-admin-full": "25.021.30718~36b6107",
        "luci-base": "25.021.30718~36b6107",
        "kmod-leds-gpio": "6.6.69-r1",
        "kmod-gpio-button-hotplug": "6.6.69-r5",
        "logd": "2024.04.26~85f10530-r1",
        "kmod-mt7603": "6.6.69.2025.01.04~e354436d-r1",
        "kmod-nf-log6": "6.6.69-r1",
        "luci-proto-ipv6": "25.021.30718~36b6107",
        "luci-i18n-base-pt-br": "25.021.30718~36b6107",
        "luci-app-attendedsysupgrade": "25.021.30718~36b6107",
        "kmod-crypto-des": "6.6.69-r1",
        "luci-i18n-attendedsysupgrade-pt-br": "25.021.30718~36b6107",
        "kmod-crypto-hw-eip93": "6.6.69-r1",
        "kmod-nft-offload": "6.6.69-r1",
        "kmod-ppp": "6.6.69-r1",
        "luci-app-upnp": "25.021.30718~36b6107",
        "uhttpd": "2023.06.25~34a8a74d-r4",
        "kmod-nf-conntrack": "6.6.69-r1",
        "luci-i18n-upnp-pt-br": "25.021.30718~36b6107",
        "htop": "3.3.0-r2",
        "ca-bundle": "20240203-r1",
        "libuclient": "2024.10.22~88ae8f20-r1",
        "luci": "25.021.30718~36b6107",
        "luci-i18n-firewall-pt-br": "25.021.30718~36b6107",
        "kmod-nf-conntrack6": "6.6.69-r1",
        "luci-light": "25.021.30718~36b6107",
        "kmod-lib-lzo": "6.6.69-r1",
        "rpcd-mod-iwinfo": "2024.09.17~9f4b86e7-r1",
        "kmod-mt76x2": "6.6.69.2025.01.04~e354436d-r1",
        "luci-mod-network": "25.021.30718~36b6107",
        "kmod-nft-core": "6.6.69-r1",
        "uhttpd-mod-ubus": "2023.06.25~34a8a74d-r4",
        "hostapd-common": "2024.09.15~5ace39b0-r2",
        "urngd": "2023.11.01~44365eb1-r1",
        "kmod-slhc": "6.6.69-r1",
        "luci-i18n-https-dns-proxy-pt-br": "25.021.30718~36b6107",
        "ppp-mod-pppoe": "2.5.1-r1"
    },
    "profile": "xiaomi,mi-router-4a-gigabit",
    "target": "ramips/mt7621",
    "version": "24.10.0-rc5",
    "diff_packages": true,
    "filesystem": "squashfs"
}
```

efahl, 2025-01-24:
Should be fixed as of about 8 hours ago.  Try again and close this if resolved.

Ritew, 2025-01-24:
The issue has been resolved, the system can be updated without any problems.
Thank you very much!

closed 2025-01-24 by Ritew