openwrt/asu #1167: Unsupported profile: raspberrypi_4-model-b

axpmonkey, 2025-01-27:
Error building the firmware image

Server response: Unsupported profile: raspberrypi_4-model-b

[Please report the error message and request](https://github.com/openwrt/asu/issues)
Request Data:

{
    "url": "https://sysupgrade.openwrt.org",
    "revision": "r24106-10cc5fcd00",
    "advanced_mode": "1",
    "sha256_unsigned": "",
    "branch": "23.05",
    "efi": null,
    "profile": "raspberrypi,4-model-b",
    "target": "bcm27xx/bcm2711",
    "version": "23.05.5",
    "packages": [
        "base-files",
        "bcm27xx-gpu-fw",
        "brcmfmac-nvram-43455-sdio",
        "busybox",
        "ca-bundle",
        "cypress-firmware-43455-sdio",
        "dnsmasq",
        "dropbear",
        "e2fsprogs",
        "firewall4",
        "fstools",
        "irqbalance",
        "iwinfo",
        "kernel",
        "kmod-brcmfmac",
        "kmod-fs-vfat",
        "kmod-nft-offload",
        "kmod-nls-cp437",
        "kmod-nls-iso8859-1",
        "kmod-r8169",
        "kmod-sound-arm-bcm2835",
        "kmod-sound-core",
        "kmod-usb-hid",
        "kmod-usb-net-lan78xx",
        "kmod-usb-net-rtl8152",
        "libc",
        "libgcc",
        "libustream-mbedtls",
        "logd",
        "luci",
        "luci-app-attendedsysupgrade",
        "luci-app-nextdns",
        "luci-ssl",
        "mkf2fs",
        "mtd",
        "netifd",
        "nftables-json",
        "odhcp6c",
        "odhcpd-ipv6only",
        "opkg",
        "partx-utils",
        "ppp",
        "ppp-mod-pppoe",
        "procd",
        "procd-seccomp",
        "procd-ujail",
        "uci",
        "uclient-fetch",
        "urandom-seed",
        "wpad-basic-mbedtls"
    ],
    "diff_packages": true,
    "filesystem": "ext4",
    "client": "luci/git-25.024.63972-9fa69d5"
}

----------------------------------
ubus call system board
{
	"kernel": "5.15.167",
	"hostname": "OpenWrt",
	"system": "ARMv8 Processor rev 3",
	"model": "Raspberry Pi 4 Model B Rev 1.1",
	"board_name": "raspberrypi,4-model-b",
	"rootfs_type": "ext4",
	"release": {
		"distribution": "OpenWrt",
		"version": "23.05.5",
		"revision": "r24106-10cc5fcd00",
		"target": "bcm27xx/bcm2711",
		"description": "OpenWrt 23.05.5 r24106-10cc5fcd00"
	}
}

aparcar, 2025-01-27:
Please try again

axpmonkey, 2025-01-28:
It's working.  Thank you.

closed 2025-01-28 by axpmonkey