openwrt/asu #1173: Attended Sysupgrade 24.10.0-rc6 to rc7 error

PeterMpeter, 2025-01-29:
Linksys e8450, using Attended Sysupgrade to go from v24.10.0-rc6 to rc7. Has anyone seen this error:

"Server response: Unsupported profile: linksys,e8450-ubi"

`{
    "url": "https://sysupgrade.openwrt.org",
    "branch": "24.10",
    "revision": "r28388-58d0057481",
    "efi": null,
    "advanced_mode": "1",
    "request_hash": "",
    "sha256_unsigned": "",
    "client": "luci/25.021.30718~36b6107",
    "packages": {
        "luci-proto-wireguard": "25.021.30718~36b6107",
        "libc": "1.2.5-r4",
        "opkg": "2024.10.16~38eccbb1-r1",
        "wpad-basic-mbedtls": "2024.09.15~5ace39b0-r2",
        "libpthread": "1.2.5-r4",
        "kmod-crypto-aead": "6.6.73-r1",
        "luci-app-https-dns-proxy": "2023.12.26-r1",
        "kmod-lib-crc-ccitt": "6.6.73-r1",
        "getrandom": "2024.04.26~85f10530-r1",
        "kmod-pppoe": "6.6.73-r1",
        "kmod-mt7622-firmware": "6.6.73.2025.01.14~8e4f72b6-r1",
        "procd-ujail": "2024.12.22~42d39376-r1",
        "base-files": "1651~58d0057481",
        "libustream-mbedtls": "2024.07.28~99bd3d2b-r1",
        "fitblk": "2",
        "firewall4": "2024.12.18~18fc0ead-r1",
        "uboot-envtools": "2024.07-r1",
        "dnsmasq": "2.90-r3",
        "kmod-mt7615-common": "6.6.73.2025.01.14~8e4f72b6-r1",
        "kmod-crypto-acompress": "6.6.73-r1",
        "kmod-mt76-connac": "6.6.73.2025.01.14~8e4f72b6-r1",
        "luci-app-ddns": "25.021.30718~36b6107",
        "kmod-usb3": "6.6.73-r1",
        "libmbedtls": "3.6.2-r1",
        "odhcp6c": "2024.09.25~b6ae9ffa-r1",
        "uci": "2025.01.20~16ff0bad-r1",
        "luci-ssl": "25.021.30718~36b6107",
        "dropbear": "2024.86-r1",
        "kmod-hwmon-core": "6.6.73-r1",
        "mtd": "26",
        "odhcpd-ipv6only": "2024.05.08~a2988231-r1",
        "kmod-crypto-ccm": "6.6.73-r1",
        "luci-app-sqm": "25.021.30718~36b6107",
        "urandom-seed": "3",
        "ppp": "2.5.1-r1",
        "kmod-leds-gpio": "6.6.73-r1",
        "kmod-gpio-button-hotplug": "6.6.73-r5",
        "logd": "2024.04.26~85f10530-r1",
        "kmod-mac80211": "6.6.73.6.12.6-r1",
        "luci-app-attendedsysupgrade": "25.021.30718~36b6107",
        "kmod-ppp": "6.6.73-r1",
        "kmod-mt7915-firmware": "6.6.73.2025.01.14~8e4f72b6-r1",
        "kmod-crypto-gcm": "6.6.73-r1",
        "ca-bundle": "20240203-r1",
        "libuclient": "2024.10.22~88ae8f20-r1",
        "kmod-crypto-cmac": "6.6.73-r1",
        "luci": "25.021.30718~36b6107",
        "kmod-lib-lzo": "6.6.73-r1",
        "kernel": "6.6.73~49f3b78722ed2df9df713e744e978eb6-r1",
        "iperf3": "3.17.1-r4",
        "urngd": "2023.11.01~44365eb1-r1",
        "kmod-cfg80211": "6.6.73.6.12.6-r1",
        "ppp-mod-pppoe": "2.5.1-r1"
    },
    "profile": "linksys,e8450-ubi",
    "target": "mediatek/mt7622",
    "version": "24.10.0-rc6",
    "diff_packages": true,
    "filesystem": "squashfs"
}`

kmitchel, 2025-01-29:
Same I received a few hours ago.  Was planning on trying again over the next few days.

aparcar, 2025-01-29:
Please try again

kmitchel, 2025-01-29:
Queue is very deep.  Hopefully that is a good sign.

PeterMpeter, 2025-01-29:
FYI
[https://www.reddit.com/r/openwrt/comments/1ia4m5x/openwrt_2410_and_snapshot_users_testers_needed/](url)

PeterMpeter, 2025-01-29:
> Queue is very deep.  Hopefully that is a good sign.
> […](#)

I've waited an hour to get to the top of the cure... now this is what I get "Building Firmware... Progress: 0% Received build request"

pmagid, 2025-01-30:
Looks like sysupgrade has fallen over due to the load...

Error connecting to upgrade server
Could not reach API at "https://sysupgrade.openwrt.org/api/v1/overview?1738196370035". Please try again later.

aparcar, 2025-01-30:
Please try again. The podman process somewhat crashed, I'll have to see why.

pmagid, 2025-01-30:
Works now

kmitchel, 2025-01-30:
I've finally been able to update with ASU.  Another related issue.  I had not `uci set system.@system[0].compat_version=2.0` so ASU failed silently.  When I tried manually with the generated firmware I was able to see the error.  TY.

aparcar, 2025-01-30:
What tool did you use to upgrade?

kmitchel, 2025-01-30:
luci-app-attendedsysupgrade

gentrydeng, 2025-02-14:
https://forum.openwrt.org/t/resolved-openwrt-image-builder-errors/224906

> 2025-Feb-14: The issues with the image builder system should now be resolved. If problems persist with Attended Sysupgrade (ASU), OpenWrt Upgrade Tool (OWUT) or the Firmware Selector custom image tool, please report them in [this thread](https://forum.openwrt.org/t/notice-problems-with-custom-firmware-image-creation/224829/4).