openwrt/asu #1176: Attended Sysupgrade fails from 24.10.0-rc5 to 24.10.0-rc7 on WZR-HP-AG300H

nhonda911, 2025-01-29:
{
    "url": "https://sysupgrade.openwrt.org",
    "branch": "24.10",
    "revision": "r28304-6dacba30a7",
    "efi": null,
    "advanced_mode": "1",
    "request_hash": "",
    "sha256_unsigned": "",
    "client": "luci/undefined",
    "packages": {
        "kmod-usb-storage": "6.6.69-r1",
        "libopenssl": "3.0.15-r1",
        "libpam": "1.5.2-r1",
        "libc": "1.2.5-r4",
        "avahi-dbus-daemon": "0.8-r9",
        "opkg": "2024.10.16~38eccbb1-r1",
        "wpad-basic-mbedtls": "2024.09.15~5ace39b0-r2",
        "luci-i18n-base-ja": "25.021.30718~36b6107",
        "libpthread": "1.2.5-r4",
        "libubus-lua": "2025.01.02~afa57cce-r1",
        "nano": "8.3-r1",
        "kmod-tun": "6.6.69-r1",
        "luci-i18n-attendedsysupgrade-ja": "25.021.30718~36b6107",
        "libtasn1": "4.19.0-r2",
        "swconfig": "12",
        "kmod-crypto-md5": "6.6.69-r1",
        "kmod-lib-crc-ccitt": "6.6.69-r1",
        "getrandom": "2024.04.26~85f10530-r1",
        "kmod-asn1-decoder": "6.6.69-r1",
        "kmod-pppoe": "6.6.69-r1",
        "libuuid": "2.40.2-r1",
        "libcap": "2.69-r1",
        "libavahi-client": "0.8-r9",
        "procd-ujail": "2024.12.22~42d39376-r1",
        "base-files": "1649~6dacba30a7",
        "libustream-mbedtls": "2024.07.28~99bd3d2b-r1",
        "ddns-scripts": "2.8.2-r52",
        "firewall4": "2024.12.18~18fc0ead-r1",
        "uboot-envtools": "2024.07-r1",
        "kmod-usb-ohci": "6.6.69-r1",
        "dnsmasq": "2.90-r3",
        "attr": "2.5.2-r3",
        "kmod-crypto-sha256": "6.6.69-r1",
        "kmod-crypto-acompress": "6.6.69-r1",
        "ddns-scripts-noip": "2.8.2-r52",
        "luci-app-ddns": "25.024.21113~4d6c9a4",
        "kmod-usb2": "6.6.69-r1",
        "libmbedtls": "3.6.2-r1",
        "libgnutls": "3.8.5-r1",
        "odhcp6c": "2024.09.25~b6ae9ffa-r1",
        "kmod-ath9k": "6.6.69.6.12.6-r1",
        "uci": "2025.01.20~16ff0bad-r1",
        "lua": "5.1.5-r11",
        "kmod-fs-ext4": "6.6.69-r1",
        "luci-ssl": "25.021.30718~36b6107",
        "dropbear": "2024.86-r1",
        "luci-i18n-package-manager-ja": "25.021.30718~36b6107",
        "kmod-nls-utf8": "6.6.69-r1",
        "libgmp": "6.3.0-r1",
        "mtd": "26",
        "odhcpd-ipv6only": "2024.05.08~a2988231-r1",
        "kmod-owl-loader": "6.6.69.6.12.6-r1",
        "samba4-server": "4.18.8-r1",
        "urandom-seed": "3",
        "ppp": "2.5.1-r1",
        "kmod-gpio-button-hotplug": "6.6.69-r5",
        "logd": "2024.04.26~85f10530-r1",
        "libreadline": "8.2-r2",
        "luci-i18n-ddns-ja": "25.024.21113~4d6c9a4",
        "kmod-dnsresolver": "6.6.69-r1",
        "kmod-crypto-ecb": "6.6.69-r1",
        "kmod-crypto-des": "6.6.69-r1",
        "attendedsysupgrade-common": "9",
        "kmod-ppp": "6.6.69-r1",
        "luci-i18n-firewall-ja": "25.021.30718~36b6107",
        "luci-app-samba4": "25.024.21113~4d6c9a4",
        "luci-proto-relay": "25.021.30718~36b6107",
        "liblua": "5.1.5-r11",
        "ca-bundle": "20240203-r1",
        "libuclient": "2024.10.22~88ae8f20-r1",
        "liblucihttp-lua": "2023.03.15~9b5b683f-r1",
        "luci": "25.021.30718~36b6107",
        "owut": "2025.01.06~e623a900-r1",
        "kmod-usb-ledtrig-usbport": "6.6.69-r1",
        "luci-i18n-openvpn-ja": "25.024.21113~4d6c9a4",
        "kmod-lib-lzo": "6.6.69-r1",
        "kmod-ath": "6.6.69.6.12.6-r1",
        "kernel": "6.6.69~8a59efb5b60b2afd0fd0c33f40800cbf-r1",
        "kmod-fs-cifs": "6.6.69-r1",
        "libatomic": "13.3.0-r4",
        "libtirpc": "1.3.4-r1",
        "libpopt": "1.19-r1",
        "openvpn-mbedtls": "2.6.12-r1",
        "kmod-scsi-core": "6.6.69-r1",
        "urngd": "2023.11.01~44365eb1-r1",
        "ppp-mod-pppoe": "2.5.1-r1",
        "luci-app-openvpn": "25.024.21113~4d6c9a4"
    },
    "profile": "buffalo,wzr-hp-ag300h",
    "target": "ath79/generic",
    "version": "24.10.0-rc5",
    "diff_packages": true,
    "filesystem": "squashfs"
}

aparcar, 2025-01-29:
Tested and it works fine

```
Status: 200
Request Hash: b8ce3b4bc4830387c4899ac297aaf16621e4757964b75ead15cadf497d21393e
Images:
openwrt-24.10.0-rc5-05603f1a1a6c-ath79-generic-buffalo_wzr-hp-ag300h-squashfs-sysupgrade.bin
openwrt-24.10.0-rc5-05603f1a1a6c-ath79-generic-buffalo_wzr-hp-ag300h-squashfs-factory.bin
openwrt-24.10.0-rc5-05603f1a1a6c-ath79-generic-buffalo_wzr-hp-ag300h-squashfs-tftp.bin
```

gentrydeng, 2025-02-14:
https://forum.openwrt.org/t/resolved-openwrt-image-builder-errors/224906

> 2025-Feb-14: The issues with the image builder system should now be resolved. If problems persist with Attended Sysupgrade (ASU), OpenWrt Upgrade Tool (OWUT) or the Firmware Selector custom image tool, please report them in [this thread](https://forum.openwrt.org/t/notice-problems-with-custom-firmware-image-creation/224829/4).