openwrt/asu #1177: Proxmox vm 24.10.0-rc6 to 24.10.0-rc7

donhwy, 2025-01-29:
Worked to upgrade to RC6. This is a copy of the error. Looks like "qemu-standard-pc-q35-ich9-2009" is broken?

Thanks. FYI

[OpenWrt](https://10.0.22.1/)

    [Status](https://10.0.22.1/cgi-bin/luci/admin/system/attendedsysupgrade#)
        [Overview](https://10.0.22.1/cgi-bin/luci/admin/status/overview)
        [Routing](https://10.0.22.1/cgi-bin/luci/admin/status/routes)
        [Firewall](https://10.0.22.1/cgi-bin/luci/admin/status/nftables)
        [System Log](https://10.0.22.1/cgi-bin/luci/admin/status/logs)
        [Processes](https://10.0.22.1/cgi-bin/luci/admin/status/processes)
        [Realtime Graphs](https://10.0.22.1/cgi-bin/luci/admin/status/realtime)
    [System](https://10.0.22.1/cgi-bin/luci/admin/system/attendedsysupgrade#)
        [System](https://10.0.22.1/cgi-bin/luci/admin/system/system)
        [Administration](https://10.0.22.1/cgi-bin/luci/admin/system/admin)
        [Software](https://10.0.22.1/cgi-bin/luci/admin/system/package-manager)
        [Startup](https://10.0.22.1/cgi-bin/luci/admin/system/startup)
        [Scheduled Tasks](https://10.0.22.1/cgi-bin/luci/admin/system/crontab)
        [Attended Sysupgrade](https://10.0.22.1/cgi-bin/luci/admin/system/attendedsysupgrade)
        [Backup / Flash Firmware](https://10.0.22.1/cgi-bin/luci/admin/system/flash)
        [Reboot](https://10.0.22.1/cgi-bin/luci/admin/system/reboot)
    [Services](https://10.0.22.1/cgi-bin/luci/admin/system/attendedsysupgrade#)
        [HTTPS DNS Proxy](https://10.0.22.1/cgi-bin/luci/admin/services/https-dns-proxy)
    [Network](https://10.0.22.1/cgi-bin/luci/admin/system/attendedsysupgrade#)
        [Interfaces](https://10.0.22.1/cgi-bin/luci/admin/network/network)
        [Routing](https://10.0.22.1/cgi-bin/luci/admin/network/routes)
        [DHCP and DNS](https://10.0.22.1/cgi-bin/luci/admin/network/dhcp)
        [Diagnostics](https://10.0.22.1/cgi-bin/luci/admin/network/diagnostics)
        [SQM QoS](https://10.0.22.1/cgi-bin/luci/admin/network/sqm)
        [Firewall](https://10.0.22.1/cgi-bin/luci/admin/network/firewall)
    [VPN](https://10.0.22.1/cgi-bin/luci/admin/system/attendedsysupgrade#)
        [OpenVPN](https://10.0.22.1/cgi-bin/luci/admin/vpn/openvpn)
    [Log out](https://10.0.22.1/cgi-bin/luci/admin/logout)

Paused

    [Overview](https://10.0.22.1/cgi-bin/luci/admin/system/attendedsysupgrade/overview)
    [Configuration](https://10.0.22.1/cgi-bin/luci/admin/system/attendedsysupgrade/configuration)

Attended Sysupgrade

The attended sysupgrade service allows to upgrade vanilla and custom firmware images easily.

This is done by building a new firmware on demand via an online service.

Currently running: 24.10.0-rc6 - r28388-58d0057481
Powered by[ LuCI openwrt-24.10 branch (25.021.30718~36b6107)](https://github.com/openwrt/luci) /[ OpenWrt 24.10.0-rc6 (r28388-58d0057481)](https://openwrt.org/)
Error building the firmware image

Server response: Unsupported profile: qemu-standard-pc-q35-ich9-2009
[Please report the error message and request](https://github.com/openwrt/asu/issues)

Request Data:

{
    "url": "https://sysupgrade.openwrt.org",
    "branch": "24.10",
    "revision": "r28388-58d0057481",
    "efi": null,
    "advanced_mode": "0",
    "request_hash": "",
    "sha256_unsigned": "",
    "client": "luci/25.021.30718~36b6107",
    "packages": {
        "kmod-ipt-proto": "6.6.73.3.26-r1",
        "kmod-natsemi": "6.6.73-r1",
        "libc": "1.2.5-r4",
        "librespeed-cli": "1.0.10-r3",
        "mkf2fs": "1.16.0-r3",
        "kmod-e100": "6.6.73-r1",
        "libevdev": "1.13.1-r1",
        "opkg": "2024.10.16~38eccbb1-r1",
        "kmod-dma-buf": "6.6.73-r1",
        "iperf": "2.1.9-r1",
        "librespeed-go": "1.1.5-r4",
        "kmod-ixgbe": "6.6.73-r1",
        "iperf3-ssl": "3.17.1-r4",
        "diffutils": "3.10-r1",
        "kmod-nft-arp": "6.6.73-r1",
        "libpci": "3.12.0-r2",
        "ip-full": "6.11.0-r1",
        "kmod-ipt-rtpengine": "6.6.73.11.5.1.18-r1",
        "e100-firmware": "20241110-r1",
        "nano": "8.3-r1",
        "glib2": "2.82.0-r1",
        "kmod-via-velocity": "6.6.73-r1",
        "luci-app-https-dns-proxy": "2023.12.26-r1",
        "kmod-br-netfilter": "6.6.73-r1",
        "kmod-ipt-filter": "6.6.73-r1",
        "kmod-ipt-nat-extra": "6.6.73-r1",
        "kmod-crypto-hw-ccp": "6.6.73-r1",
        "kmod-asn1-decoder": "6.6.73-r1",
        "liburcu": "0.14.1-r1",
        "procd-ujail": "2024.12.22~42d39376-r1",
        "base-files": "1651~58d0057481",
        "pciids": "0.387-r1",
        "partx-utils": "2.40.2-r1",
        "kmod-ipt-sysrq": "6.6.73.3.26-r1",
        "kmod-irqbypass": "6.6.73-r1",
        "libustream-mbedtls": "2024.07.28~99bd3d2b-r1",
        "kmod-ipt-raw": "6.6.73-r1",
        "kmod-r8169": "6.6.73-r1",
        "firewall4": "2024.12.18~18fc0ead-r1",
        "pciutils": "3.12.0-r2",
        "dnsmasq": "2.90-r3",
        "libudev-zero": "1.0.3-r1",
        "ip6tables-nft": "1.8.10-r1",
        "kmod-crypto-hmac": "6.6.73-r1",
        "libkmod": "32-r1",
        "kmod-crypto-acompress": "6.6.73-r1",
        "kmod-iptunnel": "6.6.73-r1",
        "kmod-ipt-condition": "6.6.73.3.26-r1",
        "kmod-ipt-nathelper-rtsp": "6.6.73.3.26-r1",
        "kmod-fs-vfat": "6.6.73-r1",
        "https-dns-proxy": "2023.12.26-r1",
        "kmod-ipt-compat-xtables": "6.6.73.3.26-r1",
        "grub2-efi": "2.12-r1",
        "kmod-ipt-conntrack-extra": "6.6.73-r1",
        "collectd-mod-irq": "5.12.0-r53",
        "kmod-ovpn-dco-v2": "6.6.73.0.2.20240320-r1",
        "grub2-bios-setup": "2.12-r1",
        "ethtool-full": "6.10-r1",
        "odhcp6c": "2024.09.25~b6ae9ffa-r1",
        "uci": "2025.01.20~16ff0bad-r1",
        "grub2": "2.12-r1",
        "luci-ssl": "25.021.30718~36b6107",
        "dropbear": "2024.86-r1",
        "kmod-input-evdev": "6.6.73-r1",
        "curl": "8.10.1-r1",
        "mtd": "26",
        "odhcpd-ipv6only": "2024.05.08~a2988231-r1",
        "kmod-crypto-authenc": "6.6.73-r1",
        "luci-app-sqm": "25.021.30718~36b6107",
        "kmod-ipt-socket": "6.6.73-r1",
        "kmod-ipt-conntrack-label": "6.6.73-r1",
        "urandom-seed": "3",
        "kmod-ipt-iface": "6.6.73.3.26-r1",
        "ppp": "2.5.1-r1",
        "kmod-nft-netdev": "6.6.73-r1",
        "logd": "2024.04.26~85f10530-r1",
        "libreadline": "8.2-r2",
        "kmod-tg3": "6.6.73-r1",
        "kmod": "32-r1",
        "qemu-ga": "9.1.0-r4",
        "kmod-dnsresolver": "6.6.73-r1",
        "kmod-ne2k-pci": "6.6.73-r1",
        "kmod-8139too": "6.6.73-r1",
        "kmod-pcnet32": "6.6.73-r1",
        "openvpn-openssl": "2.6.12-r1",
        "luci-app-attendedsysupgrade": "25.021.30718~36b6107",
        "libnl-core": "3.10.0-r1",
        "irqbalance": "1.9.4-r1",
        "drill": "1.8.3-r1",
        "e2fsprogs": "1.47.0-r2",
        "kmod-e1000": "6.6.73-r1",
        "rng-tools": "6.15-r2",
        "kmod-ipt-rpfilter": "6.6.73-r1",
        "kmod-iptunnel4": "6.6.73-r1",
        "collectd-mod-exec": "5.12.0-r53",
        "liblua5.3": "5.3.5-r6",
        "snmp-mibs": "5.9.4-r3",
        "htop": "3.3.0-r2",
        "ca-bundle": "20240203-r1",
        "kmod-forcedeth": "6.6.73-r1",
        "kmod-3c59x": "6.6.73-r1",
        "kmod-amd-xgbe": "6.6.73-r1",
        "kmod-crypto-rng": "6.6.73-r1",
        "kmod-dwmac-intel": "6.6.73-r1",
        "luci": "25.021.30718~36b6107",
        "owut": "2025.01.06~e623a900-r1",
        "kmod-lib-lzo": "6.6.73-r1",
        "ip6tables-extra": "1.8.10-r1",
        "kmod-via-rhine": "6.6.73-r1",
        "kmod-amazon-ena": "6.6.73-r1",
        "kernel": "6.6.73~a21259e4f338051d27a6443a3a7f7f1f-r1",
        "kmod-ipt-ipv4options": "6.6.73.3.26-r1",
        "kmod-e1000e": "6.6.73-r1",
        "kmod-button-hotplug": "6.6.73-r3",
        "mtr-json": "0.95-r3",
        "collectd-mod-openvpn": "5.12.0-r53",
        "kmod-igb": "6.6.73-r1",
        "kmod-igc": "6.6.73-r1",
        "kmod-ipt-offload": "6.6.73-r1",
        "kmod-bnx2": "6.6.73-r1",
        "libuv": "1.48.0-r1",
        "kmod-ipt-physdev": "6.6.73-r1",
        "urngd": "2023.11.01~44365eb1-r1",
        "iftop": "2018.10.03~77901c8c-r2",
        "kmod-kvm-amd": "6.6.73-r1",
        "luci-app-openvpn": "25.021.30718~36b6107",
        "ppp-mod-pppoe": "2.5.1-r1",
        "collectd-mod-sqm": "5.12.0-r53",
        "libnl-genl": "3.10.0-r1"
    },
    "profile": "qemu-standard-pc-q35-ich9-2009",
    "target": "x86/64",
    "version": "24.10.0-rc6",
    "diff_packages": true,
    "filesystem": "ext4"
}

Close

efahl, 2025-02-01:
Try again, -rc7 is now out and should build without issue.

gentrydeng, 2025-02-14:
https://forum.openwrt.org/t/resolved-openwrt-image-builder-errors/224906

> 2025-Feb-14: The issues with the image builder system should now be resolved. If problems persist with Attended Sysupgrade (ASU), OpenWrt Upgrade Tool (OWUT) or the Firmware Selector custom image tool, please report them in [this thread](https://forum.openwrt.org/t/notice-problems-with-custom-firmware-image-creation/224829/4).