openwrt/asu #1185: Bananapi BPI-R3 Update Error 23.05.05 -> 24.10.0-RC7 with Adguard-home-openwrt

Matador-67, 2025-01-31:
aarch64_cortex-a53_musl/root-mediatek/../../../../builder/dl/openwrt_luci
Downloading https://downloads.openwrt.org/releases/24.10.0-rc7/packages/aarch64_cortex-a53/luci/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/24.10.0-rc7/packages/aarch64_cortex-a53/packages/Packages.gz
Updated list of available packages in /builder/build_dir/target-aarch64_cortex-a53_musl/root-mediatek/../../../../builder/dl/openwrt_packages
Downloading https://downloads.openwrt.org/releases/24.10.0-rc7/packages/aarch64_cortex-a53/packages/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/24.10.0-rc7/packages/aarch64_cortex-a53/routing/Packages.gz
Updated list of available packages in /builder/build_dir/target-aarch64_cortex-a53_musl/root-mediatek/../../../../builder/dl/openwrt_routing
Downloading https://downloads.openwrt.org/releases/24.10.0-rc7/packages/aarch64_cortex-a53/routing/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/24.10.0-rc7/packages/aarch64_cortex-a53/telephony/Packages.gz
Updated list of available packages in /builder/build_dir/target-aarch64_cortex-a53_musl/root-mediatek/../../../../builder/dl/openwrt_telephony
Downloading https://downloads.openwrt.org/releases/24.10.0-rc7/packages/aarch64_cortex-a53/telephony/Packages.sig
Signature check passed.
Downloading file:packages/Packages
Updated list of available packages in /builder/build_dir/target-aarch64_cortex-a53_musl/root-mediatek/../../../../builder/dl/imagebuilder
Downloading file:packages/Packages.sig
Signature check passed.
Collected errors:
 * opkg_install_cmd: Cannot install package luci-app-diag-core.
 * opkg_install_cmd: Cannot install package luci-i18n-diag-core-de.
 * opkg_install_cmd: Cannot install package luci-i18n-opkg-de.
make[2]: *** [Makefile:234: package_install] Error 255
make[1]: *** [Makefile:171: _call_manifest] Error 2
make: *** [Makefile:349: manifest] Error 2

aparcar, 2025-01-31:
@efahl ideas on the first missing LuCI package?

efahl, 2025-01-31:
uci-app-diag-core was removed back in April of last year:

https://github.com/openwrt/luci/commit/744564d58cb34ac0cd78b637134eb17b09d89d26

aparcar, 2025-01-31:
I guess this stuff needs to go into package changes

efahl, 2025-01-31:
I think it's a never ending task, just in luci I found these with some digging:

```
bd36a83c5b 2024-12-19T17:50:49+00:00 Paul Donald     luci-proto-bonding: remove (in favour of netifd bonding in   2024-12-19T18:44:57+00:00
f19649b7be 2024-09-10T18:16:32+09:00 Yanase Yuki     luci-app-polipo: remove package                              2024-09-11T16:27:26+03:00
7c582f150e 2025-01-06T17:01:34+00:00 Paul Donald     luci-app-vnstat: drop lua app (prefer luci-app-vnstat2)      2025-01-06T17:01:34+00:00
e6b413c5b8 2024-07-10T09:11:52+00:00 Yousong Zhou    luci-app-shadowsocks-libev: drop the package                 2024-07-10T19:10:54+08:00
```

That's not even considering any of base or package feeds.

aparcar, 2025-01-31:
Could you kindly create a PR and add it to the package changes?

Matador-67, 2025-02-01:
New Test this morning, 
only These Errors left:
opkg_install_cmd: Cannot install package luci-app-diag-core.
opkg_install_cmd: Cannot install package luci-i18n-diag-core-de.

Unchecked them and Image was builded.

PR #1186 (2025-02-01): config: add removed package to package_changes
'luci-app-diag-core' was removed from the package feeds, include it in the 'package_changes' list.

Fixes: #1185
commit 93584bcc (2025-02-01): config: add removed package to package_changes

'luci-app-diag-core' was removed from the package feeds, include it in the
'package_changes' list.

Fixes: #1185
closed 2025-02-01 by aparcar