openwrt/asu #1189: Attended Sysupgrade fails from 24.10.0-rc7 to 24.10.0  on Totolink,x5000r

pioppertz, 2025-02-04:
``"url": "https://sysupgrade.openwrt.org",
    "branch": "24.10",
    "revision": "r28417-daef29c75d",
    "efi": null,
    "advanced_mode": "0",
    "request_hash": "",
    "sha256_unsigned": "",
    "client": "luci/25.027.83426~170375e",
    "packages": {
        "luci-proto-wireguard": "25.027.83426~170375e",
        "libc": "1.2.5-r4",
        "luci-app-travelmate": "25.027.83426~170375e",
        "opkg": "2024.10.16~38eccbb1-r1",
        "wpad-basic-mbedtls": "2024.09.15~5ace39b0-r2",
        "procd-ujail": "2024.12.22~42d39376-r1",
        "base-files": "1653~daef29c75d",
        "libustream-mbedtls": "2024.07.28~99bd3d2b-r1",
        "firewall4": "2024.12.18~18fc0ead-r1",
        "dnsmasq": "2.90-r4",
        "odhcp6c": "2024.09.25~b6ae9ffa-r1",
        "uci": "2025.01.20~16ff0bad-r1",
        "dropbear": "2024.86-r1",
        "mtd": "26",
        "odhcpd-ipv6only": "2024.05.08~a2988231-r1",
        "travelmate": "2.1.3-r3",
        "urandom-seed": "3",
        "luci-mod-dashboard": "25.027.83426~170375e",
        "ppp": "2.5.1-r1",
        "kmod-leds-gpio": "6.6.73-r1",
        "kmod-gpio-button-hotplug": "6.6.73-r5",
        "logd": "2024.04.26~85f10530-r1",
        "luci-app-attendedsysupgrade": "25.027.83426~170375e",
        "kmod-crypto-hw-eip93": "6.6.73-r1",
        "luci-app-wifischedule": "25.027.83426~170375e",
        "kmod-mt7915-firmware": "6.6.73.2025.01.14~8e4f72b6-r1",
        "ca-bundle": "20240203-r1",
        "luci": "25.027.83426~170375e",
        "kernel": "6.6.73~3abe85def815b59c6c75ac1f92135cb6-r1",
        "urngd": "2023.11.01~44365eb1-r1",
        "ppp-mod-pppoe": "2.5.1-r1"
    },
    "profile": "totolink,x5000r",
    "target": "ramips/mt7621",
    "version": "24.10.0-rc7",
    "diff_packages": true,
    "filesystem": "squashfs"
}

pioppertz, 2025-02-04:
"url": "https://sysupgrade.openwrt.org",
    "branch": "24.10",
    "revision": "r28417-daef29c75d",
    "efi": null,
    "advanced_mode": "0",
    "request_hash": "",
    "sha256_unsigned": "",
    "client": "luci/25.027.83426~170375e",
    "packages": {
        "luci-proto-wireguard": "25.027.83426~170375e",
        "libc": "1.2.5-r4",
        "luci-app-travelmate": "25.027.83426~170375e",
        "opkg": "2024.10.16~38eccbb1-r1",
        "wpad-basic-mbedtls": "2024.09.15~5ace39b0-r2",
        "procd-ujail": "2024.12.22~42d39376-r1",
        "base-files": "1653~daef29c75d",
        "libustream-mbedtls": "2024.07.28~99bd3d2b-r1",
        "firewall4": "2024.12.18~18fc0ead-r1",
        "dnsmasq": "2.90-r4",
        "odhcp6c": "2024.09.25~b6ae9ffa-r1",
        "uci": "2025.01.20~16ff0bad-r1",
        "dropbear": "2024.86-r1",
        "mtd": "26",
        "odhcpd-ipv6only": "2024.05.08~a2988231-r1",
        "travelmate": "2.1.3-r3",
        "urandom-seed": "3",
        "luci-mod-dashboard": "25.027.83426~170375e",
        "ppp": "2.5.1-r1",
        "kmod-leds-gpio": "6.6.73-r1",
        "kmod-gpio-button-hotplug": "6.6.73-r5",
        "logd": "2024.04.26~85f10530-r1",
        "luci-app-attendedsysupgrade": "25.027.83426~170375e",
        "kmod-crypto-hw-eip93": "6.6.73-r1",
        "luci-app-wifischedule": "25.027.83426~170375e",
        "kmod-mt7915-firmware": "6.6.73.2025.01.14~8e4f72b6-r1",
        "ca-bundle": "20240203-r1",
        "luci": "25.027.83426~170375e",
        "kernel": "6.6.73~3abe85def815b59c6c75ac1f92135cb6-r1",
        "urngd": "2023.11.01~44365eb1-r1",
        "ppp-mod-pppoe": "2.5.1-r1"
    },
    "profile": "totolink,x5000r",
    "target": "ramips/mt7621",
    "version": "24.10.0-rc7",
    "diff_packages": true,
    "filesystem": "squashfs"
}

pioppertz, 2025-02-04:
> "url": "https://sysupgrade.openwrt.org",
    "branch": "24.10",
    "revision": "r28417-daef29c75d",
    "efi": null,
    "advanced_mode": "0",
    "request_hash": "",
    "sha256_unsigned": "",
    "client": "luci/25.027.83426~170375e",
    "packages": {
        "luci-proto-wireguard": "25.027.83426~170375e",
        "libc": "1.2.5-r4",
        "luci-app-travelmate": "25.027.83426~170375e",
        "opkg": "2024.10.16~38eccbb1-r1",
        "wpad-basic-mbedtls": "2024.09.15~5ace39b0-r2",
        "procd-ujail": "2024.12.22~42d39376-r1",
        "base-files": "1653~daef29c75d",
        "libustream-mbedtls": "2024.07.28~99bd3d2b-r1",
        "firewall4": "2024.12.18~18fc0ead-r1",
        "dnsmasq": "2.90-r4",
        "odhcp6c": "2024.09.25~b6ae9ffa-r1",
        "uci": "2025.01.20~16ff0bad-r1",
        "dropbear": "2024.86-r1",
        "mtd": "26",
        "odhcpd-ipv6only": "2024.05.08~a2988231-r1",
        "travelmate": "2.1.3-r3",
        "urandom-seed": "3",
        "luci-mod-dashboard": "25.027.83426~170375e",
        "ppp": "2.5.1-r1",
        "kmod-leds-gpio": "6.6.73-r1",
        "kmod-gpio-button-hotplug": "6.6.73-r5",
        "logd": "2024.04.26~85f10530-r1",
        "luci-app-attendedsysupgrade": "25.027.83426~170375e",
        "kmod-crypto-hw-eip93": "6.6.73-r1",
        "luci-app-wifischedule": "25.027.83426~170375e",
        "kmod-mt7915-firmware": "6.6.73.2025.01.14~8e4f72b6-r1",
        "ca-bundle": "20240203-r1",
        "luci": "25.027.83426~170375e",
        "kernel": "6.6.73~3abe85def815b59c6c75ac1f92135cb6-r1",
        "urngd": "2023.11.01~44365eb1-r1",
        "ppp-mod-pppoe": "2.5.1-r1"
    },
    "profile": "totolink,x5000r",
    "target": "ramips/mt7621",
    "version": "24.10.0-rc7",
    "diff_packages": true,
    "filesystem": "squashfs"
}

aparcar, 2025-02-04:
What is the error message?

efahl, 2025-02-04:
24.10.0 has not been released, please wait for the release announcement before reporting any issues against it.

aparcar, 2025-02-04:
Should be working now.

@efahl maybe we can improve the error message and we should no longer encourage people to report these types of errors...

closed 2025-02-04 by aparcar

pioppertz, 2025-02-04:
> Should be working now.

Almost....But not yet...

Error message  :

Error building the firmware image

Server response: init
[Please report the error message and request](https://github.com/openwrt/asu/issues)

But I can wait, no prblem, haha... Keep up your good work....

aparcar, 2025-02-04:
I just tested and it worked:

Status: 200
Request Hash: a3b9380adb084fd117916b12f2e170cc284c79acbee3d05e4e65d8769f288bf4
Images:
openwrt-24.10.0-rc7-8ca650d71558-ramips-mt7621-totolink_x5000r-squashfs-sysupgrade.bin

efahl, 2025-02-04:
Yeah, good old ramips/mt7621 (and ath79/generic) are usually the last to be built, they take like 5-6 hours just for the build and then another 3 to get into the ASU pipeline...

@aparcar That error message is in the LuCI app, probably should just delete it.

https://github.com/openwrt/luci/blob/master/applications/luci-app-attendedsysupgrade/htdocs/luci-static/resources/view/attendedsysupgrade/overview.js#L226