openwrt/asu #1190: Attended sysupgrade from OpenWrt 23.05.5 r24106-10cc5fcd00 to 24.10.0 on Netgear WAX206 fails

FinnVonB, 2025-02-04:
All is updated to latest package.

{
    "url": "https://sysupgrade.openwrt.org",
    "revision": "r24106-10cc5fcd00",
    "advanced_mode": "1",
    "sha256_unsigned": "",
    "branch": "23.05",
    "efi": null,
    "request_hash": "d1e8cd4a7d2b7cb4da041d5aa871c9c6da9e51bd4778493f73b97be55d1e1dd8",
    "profile": "netgear,wax206",
    "target": "mediatek/mt7622",
    "version": "24.10.0",
    "packages": [
        "attendedsysupgrade-common",
        "auc",
        "base-files",
        "busybox",
        "ca-bundle",
        "dnsmasq",
        "dropbear",
        "firewall4",
        "fstools",
        "hostapd-common",
        "kernel",
        "kmod-gpio-button-hotplug",
        "kmod-leds-gpio",
        "kmod-mt7622-firmware",
        "kmod-mt7915-firmware",
        "kmod-nft-offload",
        "libc",
        "libmbedtls",
        "libucode",
        "libustream-mbedtls",
        "logd",
        "luci",
        "luci-app-attendedsysupgrade",
        "luci-app-firewall",
        "luci-app-opkg",
        "luci-base",
        "luci-light",
        "luci-mod-admin-full",
        "luci-mod-network",
        "luci-mod-status",
        "luci-mod-system",
        "luci-proto-ipv6",
        "luci-proto-ppp",
        "luci-theme-bootstrap",
        "mtd",
        "netifd",
        "nftables-json",
        "odhcp6c",
        "odhcpd-ipv6only",
        "opkg",
        "ppp",
        "ppp-mod-pppoe",
        "procd",
        "procd-seccomp",
        "procd-ujail",
        "rpcd-mod-luci",
        "uboot-envtools",
        "ubus",
        "ubusd",
        "uci",
        "uclient-fetch",
        "ucode",
        "ucode-mod-fs",
        "ucode-mod-math",
        "ucode-mod-nl80211",
        "ucode-mod-rtnl",
        "ucode-mod-ubus",
        "ucode-mod-uci",
        "ucode-mod-uloop",
        "urandom-seed",
        "urngd",
        "wpad-basic-mbedtls"
    ],
    "diff_packages": true,
    "filesystem": "squashfs",
    "client": "luci/git-25.024.63972-9fa69d5"
}

STDERR:

Generate local signing keys...
Generate local certificate...
Package list missing or not up-to-date, generating it.

Building package index...
Downloading https://downloads.openwrt.org/releases/24.10.0/targets/mediatek/mt7622/packages/Packages.gz
Updated list of available packages in /builder/build_dir/target-aarch64_cortex-a53_musl/root-mediatek/../../../../builder/dl/openwrt_core
Downloading https://downloads.openwrt.org/releases/24.10.0/targets/mediatek/mt7622/packages/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/24.10.0/packages/aarch64_cortex-a53/base/Packages.gz
Updated list of available packages in /builder/build_dir/target-aarch64_cortex-a53_musl/root-mediatek/../../../../builder/dl/openwrt_base
Downloading https://downloads.openwrt.org/releases/24.10.0/packages/aarch64_cortex-a53/base/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/24.10.0/targets/mediatek/mt7622/kmods/6.6.73-1-49f3b78722ed2df9df713e744e978eb6/Packages.gz
Updated list of available packages in /builder/build_dir/target-aarch64_cortex-a53_musl/root-mediatek/../../../../builder/dl/openwrt_kmods
Downloading https://downloads.openwrt.org/releases/24.10.0/targets/mediatek/mt7622/kmods/6.6.73-1-49f3b78722ed2df9df713e744e978eb6/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/24.10.0/packages/aarch64_cortex-a53/luci/Packages.gz
Updated list of available packages in /builder/build_dir/target-aarch64_cortex-a53_musl/root-mediatek/../../../../builder/dl/openwrt_luci
Downloading https://downloads.openwrt.org/releases/24.10.0/packages/aarch64_cortex-a53/luci/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/24.10.0/packages/aarch64_cortex-a53/packages/Packages.gz
Updated list of available packages in /builder/build_dir/target-aarch64_cortex-a53_musl/root-mediatek/../../../../builder/dl/openwrt_packages
Downloading https://downloads.openwrt.org/releases/24.10.0/packages/aarch64_cortex-a53/packages/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/24.10.0/packages/aarch64_cortex-a53/routing/Packages.gz
Updated list of available packages in /builder/build_dir/target-aarch64_cortex-a53_musl/root-mediatek/../../../../builder/dl/openwrt_routing
Downloading https://downloads.openwrt.org/releases/24.10.0/packages/aarch64_cortex-a53/routing/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/24.10.0/packages/aarch64_cortex-a53/telephony/Packages.gz
Updated list of available packages in /builder/build_dir/target-aarch64_cortex-a53_musl/root-mediatek/../../../../builder/dl/openwrt_telephony
Downloading https://downloads.openwrt.org/releases/24.10.0/packages/aarch64_cortex-a53/telephony/Packages.sig
Signature check passed.
Downloading file:packages/Packages
Updated list of available packages in /builder/build_dir/target-aarch64_cortex-a53_musl/root-mediatek/../../../../builder/dl/imagebuilder
Downloading file:packages/Packages.sig
Signature check passed.
Collected errors:
 * opkg_install_cmd: Cannot install package auc.
make[2]: *** [Makefile:234: package_install] Error 255
make[1]: *** [Makefile:171: _call_manifest] Error 2
make: *** [Makefile:349: manifest] Error 2

FinnVonB, 2025-02-04:
Sorry  -  but I  wanted to test rc7 and then 24.10.0 came up in my attended sysupgrade - I'll hold my horses and wait for the appropriate announcement.

Thanx

gentrydeng, 2025-02-14:
https://forum.openwrt.org/t/resolved-openwrt-image-builder-errors/224906

> 2025-Feb-14: The issues with the image builder system should now be resolved. If problems persist with Attended Sysupgrade (ASU), OpenWrt Upgrade Tool (OWUT) or the Firmware Selector custom image tool, please report them in [this thread](https://forum.openwrt.org/t/notice-problems-with-custom-firmware-image-creation/224829/4).