openwrt/asu #1193: Attended sysupgrade from OpenWrt 23.05.4 to 23.05.5 on Asus RT-AC51U fails

UAVXP, 2025-02-05:
Asus RT-AC51U
LuCI openwrt-23.05 branch (git-24.212.79335-cdbe903) / OpenWrt 23.05.4 (r24012-d8dd03c46f)
```
Error building the firmware image
Server response: Error: Impossible package selection
```

Request data:
```
{
    "url": "https://sysupgrade.openwrt.org",
    "revision": "r24012-d8dd03c46f",
    "advanced_mode": "0",
    "sha256_unsigned": "",
    "branch": "23.05",
    "efi": null,
    "profile": "asus,rt-ac51u",
    "target": "ramips/mt7620",
    "version": "23.05.5",
    "packages": [
        "base-files",
        "block-mount",
        "busybox",
        "ca-bundle",
        "cgi-io",
        "dnsmasq",
        "dropbear",
        "firewall4",
        "fstools",
        "fwtool",
        "getrandom",
        "hostapd-common",
        "iw",
        "iwinfo",
        "jansson",
        "jshn",
        "jsonfilter",
        "kernel",
        "kmod-cfg80211",
        "kmod-crypto-acompress",
        "kmod-crypto-aead",
        "kmod-crypto-ccm",
        "kmod-crypto-cmac",
        "kmod-crypto-crc32c",
        "kmod-crypto-ctr",
        "kmod-crypto-gcm",
        "kmod-crypto-gf128",
        "kmod-crypto-ghash",
        "kmod-crypto-hash",
        "kmod-crypto-hmac",
        "kmod-crypto-manager",
        "kmod-crypto-null",
        "kmod-crypto-rng",
        "kmod-crypto-seqiv",
        "kmod-crypto-sha512",
        "kmod-fs-ext4",
        "kmod-gpio-button-hotplug",
        "kmod-leds-gpio",
        "kmod-lib-crc-ccitt",
        "kmod-lib-crc32c",
        "kmod-lib-lzo",
        "kmod-mac80211",
        "kmod-mt76x0e",
        "kmod-nf-conntrack",
        "kmod-nf-conntrack6",
        "kmod-nf-flow",
        "kmod-nf-log",
        "kmod-nf-log6",
        "kmod-nf-nat",
        "kmod-nf-reject",
[... 149 lines cut ...]
 * opkg_install_cmd: Cannot install package liblucihttp-ucode.
 * opkg_install_pkg: Checksum or size mismatch for package rpcd-mod-luci. Either the opkg or the package index are corrupt. Try 'opkg update'.
 * opkg_install_cmd: Cannot install package luci.
 * opkg_install_pkg: Checksum or size mismatch for package attendedsysupgrade-common. Either the opkg or the package index are corrupt. Try 'opkg update'.
 * opkg_install_cmd: Cannot install package luci-app-attendedsysupgrade.
 * opkg_install_pkg: Checksum or size mismatch for package hd-idle. Either the opkg or the package index are corrupt. Try 'opkg update'.
 * opkg_install_cmd: Cannot install package luci-app-hd-idle.
 * opkg_install_pkg: Checksum or size mismatch for package liblucihttp0. Either the opkg or the package index are corrupt. Try 'opkg update'.
 * opkg_install_cmd: Cannot install package luci-app-openvpn.
 * opkg_install_pkg: Checksum or size mismatch for package libcap-ng. Either the opkg or the package index are corrupt. Try 'opkg update'.
 * opkg_install_cmd: Cannot install package luci-app-upnp.
 * opkg_install_pkg: Checksum or size mismatch for package openvpn-openssl. Either the opkg or the package index are corrupt. Try 'opkg update'.
 * opkg_install_cmd: Cannot install package openvpn-openssl.
 * opkg_install_pkg: Checksum or size mismatch for package rpcd-mod-luci. Either the opkg or the package index are corrupt. Try 'opkg update'.
 * opkg_install_cmd: Cannot install package rpcd-mod-luci.
 * opkg_install_pkg: Checksum or size mismatch for package rpcd-mod-rrdns. Either the opkg or the package index are corrupt. Try 'opkg update'.
 * opkg_install_cmd: Cannot install package rpcd-mod-rrdns.
 * opkg_install_pkg: Checksum or size mismatch for package ucode-mod-html. Either the opkg or the package index are corrupt. Try 'opkg update'.
 * opkg_install_cmd: Cannot install package ucode-mod-html.
 * opkg_install_pkg: Checksum or size mismatch for package usbutils. Either the opkg or the package index are corrupt. Try 'opkg update'.
 * opkg_install_cmd: Cannot install package usbutils.
make[2]: *** [Makefile:189: package_install] Error 255
make[1]: *** [Makefile:154: _call_manifest] Error 2
make: *** [Makefile:274: manifest] Error 2
```

howels, 2025-02-07:
Same error on x86 using `auc -b 24.10 -B 24.10.0`

UAVXP, 2025-02-07:
Suddenly a reboot fixed that for me

thejustsoul, 2025-02-10:
Similar problem, reboot didn't help. 23.05.5 to 24.10.0 unsuccessfully.