openwrt/asu #1194: CWWK CW-J6-6L Currently running: 24.10.0-rc7 - r28417-daef29c75d Attended Upgrade: Error building the firmware image, Impossible package selection 24.10.0

openwrtnewbie, 2025-02-06:
{
    "url": "https://sysupgrade.openwrt.org",
    "branch": "24.10",
    "revision": "r28417-daef29c75d",
    "efi": {
        "path": "/sys/firmware/efi",
        "type": "directory",
        "size": 0,
        "mode": 16877,
        "atime": 1738806244,
        "mtime": 1738806244,
        "ctime": 1738806244,
        "inode": 7101,
        "uid": 0,
        "gid": 0
    },
    "advanced_mode": "0",
    "request_hash": "22646811f60d962639e059dacea26e9318d36d72dfe01bf3de86b04b0736474d",
    "sha256_unsigned": "",
    "client": "luci/25.035.62793~4f7a183",
    "packages": {
        "luci-proto-wireguard": "25.035.62793~4f7a183",
        "luci-app-nlbwmon": "25.035.62793~4f7a183",
        "mkf2fs": "1.16.0-r3",
        "opkg": "2024.10.16~38eccbb1-r1",
        "libpthread": "1.2.5-r4",
        "kmod-ixgbe": "6.6.73-r1",
        "rpcd": "2024.09.17~9f4b86e7-r1",
        "libiwinfo": "2024.10.20~b94f066e-r1",
        "kmod-nft-fib": "6.6.73-r1",
        "kmod-nfnetlink": "6.6.73-r1",
        "kmod-nf-reject6": "6.6.73-r1",
        "kmod-nf-flow": "6.6.73-r1",
        "kmod-lib-crc-ccitt": "6.6.73-r1",
        "px5g-mbedtls": "11",
        "intel-microcode": "20240531-r1",
        "getrandom": "2024.04.26~85f10530-r1",
        "luci-app-temp-status": "0.4.1-r1",
        "luci-app-irqbalance": "25.035.62793~4f7a183",
        "kmod-pppoe": "6.6.73-r1",
        "kmod-pppox": "6.6.73-r1",
        "collectd-mod-sensors": "5.12.0-r53",
        "kmod-nf-reject": "6.6.73-r1",
        "procd-ujail": "2024.12.22~42d39376-r1",
        "base-files": "1653~daef29c75d",
        "kmod-nf-nat": "6.6.73-r1",
        "kmod-crypto-crc32c": "6.6.73-r1",
        "partx-utils": "2.40.2-r1",
        "libustream-mbedtls": "2024.07.28~99bd3d2b-r1",
        "libf2fs": "1.16.0-r3",
        "libsmartcols": "2.40.2-r1",
        "kmod-r8169": "6.6.73-r1",
        "firewall4": "2024.12.18~18fc0ead-r1",
        "dnsmasq": "2.90-r4",
        "luci-app-banip": "25.035.62793~4f7a183",
        "kmod-crypto-acompress": "6.6.73-r1",
        "rpcd-mod-ucode": "2024.09.17~9f4b86e7-r1",
        "ucode-mod-math": "2024.07.22~b610860d-r3",
        "kmod-lib-crc32c": "6.6.73-r1",
        "kmod-fs-vfat": "6.6.73-r1",
        "kmod-nft-nat": "6.6.73-r1",
        "grub2-efi": "2.12-r1",
        "libmbedtls": "3.6.2-r1",
        "libmnl": "1.0.5-r1",
        "jansson": "2.14-r3",
        "grub2-bios-setup": "2.12-r1",
        "odhcp6c": "2024.09.25~b6ae9ffa-r1",
        "uclient-fetch": "2024.10.22~88ae8f20-r1",
        "uci": "2025.01.20~16ff0bad-r1",
        "grub2": "2.12-r1",
        "luci-ssl": "24.328.43591~9e70c9a",
        "dropbear": "2024.86-r1",
        "libnftnl": "1.2.8-r1",
        "rpcd-mod-file": "2024.09.17~9f4b86e7-r1",
        "mtd": "26",
        "odhcpd-ipv6only": "2024.05.08~a2988231-r1",
        "kmod-nf-log": "6.6.73-r1",
        "urandom-seed": "3",
        "libcomerr": "1.47.0-r2",
        "ppp": "2.5.1-r1",
        "logd": "2024.04.26~85f10530-r1",
        "kmod-nf-log6": "6.6.73-r1",
        "librt": "1.2.5-r4",
        "kmod-tg3": "6.6.73-r1",
        "wireguard-tools": "1.0.20210914-r4",
        "luci-app-nft-qos": "25.035.62793~4f7a183",
        "luci-app-attendedsysupgrade": "25.035.62793~4f7a183",
        "libext2fs": "1.47.0-r2",
        "e2fsprogs": "1.47.0-r2",
        "kmod-e1000": "6.6.73-r1",
        "kmod-nft-offload": "6.6.73-r1",
        "kmod-ppp": "6.6.73-r1",
        "libss": "1.47.0-r2",
        "uhttpd": "2023.06.25~34a8a74d-r4",
        "kmod-nf-conntrack": "6.6.73-r1",
        "ca-bundle": "20240203-r1",
        "kmod-forcedeth": "6.6.73-r1",
        "libuclient": "2024.10.22~88ae8f20-r1",
        "kmod-amd-xgbe": "6.6.73-r1",
        "kmod-dwmac-intel": "6.6.73-r1",
        "luci": "24.297.79519~bcd13b9",
        "kmod-nf-conntrack6": "6.6.73-r1",
        "luci-app-pbr": "1.1.8-r2",
        "kmod-lib-lzo": "6.6.73-r1",
        "kmod-amazon-ena": "6.6.73-r1",
        "kmod-e1000e": "6.6.73-r1",
        "rpcd-mod-iwinfo": "2024.09.17~9f4b86e7-r1",
        "kmod-nft-core": "6.6.73-r1",
        "kmod-button-hotplug": "6.6.73-r3",
        "uhttpd-mod-ubus": "2023.06.25~34a8a74d-r4",
        "kmod-igb": "6.6.73-r1",
        "kmod-igc": "6.6.73-r1",
        "libe2p": "1.47.0-r2",
        "kmod-bnx2": "6.6.73-r1",
        "urngd": "2023.11.01~44365eb1-r1",
        "kmod-slhc": "6.6.73-r1",
        "ppp-mod-pppoe": "2.5.1-r1"
    },
    "profile": "cwwk-cw-j6-6l",
    "target": "x86/64",
    "version": "24.10.0-rc7",
    "diff_packages": true,
    "filesystem": "squashfs"
}
STDERR:
Generate local signing keys...
Generate local certificate...
Package list missing or not up-to-date, generating it.

Building package index...
Downloading http://downloads.openwrt.org/releases/24.10.0/targets/x86/64/packages/Packages.gz
Updated list of available packages in /builder/build_dir/target-x86_64_musl/root-x86/../../../../builder/dl/openwrt_core
Downloading http://downloads.openwrt.org/releases/24.10.0/targets/x86/64/packages/Packages.sig
Signature check passed.
Downloading http://downloads.openwrt.org/releases/24.10.0/packages/x86_64/base/Packages.gz
Updated list of available packages in /builder/build_dir/target-x86_64_musl/root-x86/../../../../builder/dl/openwrt_base
Downloading http://downloads.openwrt.org/releases/24.10.0/packages/x86_64/base/Packages.sig
Signature check passed.
Downloading http://downloads.openwrt.org/releases/24.10.0/targets/x86/64/kmods/6.6.73-1-a21259e4f338051d27a6443a3a7f7f1f/Packages.gz
Updated list of available packages in /builder/build_dir/target-x86_64_musl/root-x86/../../../../builder/dl/openwrt_kmods
Downloading http://downloads.openwrt.org/releases/24.10.0/targets/x86/64/kmods/6.6.73-1-a21259e4f338051d27a6443a3a7f7f1f/Packages.sig
Signature check passed.
Downloading http://downloads.openwrt.org/releases/24.10.0/packages/x86_64/luci/Packages.gz
Updated list of available packages in /builder/build_dir/target-x86_64_musl/root-x86/../../../../builder/dl/openwrt_luci
Downloading http://downloads.openwrt.org/releases/24.10.0/packages/x86_64/luci/Packages.sig
Signature check passed.
Downloading http://downloads.openwrt.org/releases/24.10.0/packages/x86_64/packages/Packages.gz
Updated list of available packages in /builder/build_dir/target-x86_64_musl/root-x86/../../../../builder/dl/openwrt_packages
Downloading http://downloads.openwrt.org/releases/24.10.0/packages/x86_64/packages/Packages.sig
Signature check passed.
Downloading http://downloads.openwrt.org/releases/24.10.0/packages/x86_64/routing/Packages.gz
Updated list of available packages in /builder/build_dir/target-x86_64_musl/root-x86/../../../../builder/dl/openwrt_routing
Downloading http://downloads.openwrt.org/releases/24.10.0/packages/x86_64/routing/Packages.sig
Signature check passed.
Downloading http://downloads.openwrt.org/releases/24.10.0/packages/x86_64/telephony/Packages.gz
Updated list of available packages in /builder/build_dir/target-x86_64_musl/root-x86/../../../../builder/dl/openwrt_telephony
Downloading http://downloads.openwrt.org/releases/24.10.0/packages/x86_64/telephony/Packages.sig
Signature check passed.
Downloading file:packages/Packages
Updated list of available packages in /builder/build_dir/target-x86_64_musl/root-x86/../../../../builder/dl/imagebuilder
Downloading file:packages/Packages.sig
Signature check passed.
Collected errors:
 * opkg_install_cmd: Cannot install package luci-app-temp-status.
make[2]: *** [Makefile:234: package_install] Error 255
make[1]: *** [Makefile:171: _call_manifest] Error 2
make: *** [Makefile:349: manifest] Error 2

Please help. When I do the update without auto assistant I got the system running except for wireguard tools said something like package exists but incorrect architure.

openwrtnewbie, 2025-02-06:
I fixed the issue thanks everyone!

closed 2025-02-06 by openwrtnewbie

teamfighter, 2025-04-21:
> I fixed the issue thanks everyone!

Could you please provide more details?