openwrt/asu #1196: Attended sysupgrade from OpenWrt 23.05.5 to 23.10.0 on Ubiquiti UniFi AC Pro fails

daidan83, 2025-02-06:
Request Data:
`
{
    "url": "https://sysupgrade.openwrt.org",
    "revision": "r24106-10cc5fcd00",
    "advanced_mode": "1",
    "sha256_unsigned": "",
    "branch": "23.05",
    "efi": null,
    "request_hash": "3d578a266fc35257057dddbdbb9d045f44773fd74ceb51655f4c5c2780e358dd",
    "profile": "ubnt,unifiac-pro",
    "target": "ath79/generic",
    "version": "24.10.0",
    "packages": [
        "ath10k-firmware-qca988x-ct",
        "base-files",
        "busybox",
        "ca-bundle",
        "dnsmasq",
        "dropbear",
        "firewall4",
        "fstools",
        "iperf3",
        "kernel",
        "kmod-ath10k-ct",
        "kmod-ath9k",
        "kmod-gpio-button-hotplug",
        "kmod-nft-offload",
        "kmod-usb2",
        "libc",
        "libgcc",
        "libustream-mbedtls",
        "logd",
        "luci",
        "luci-app-attendedsysupgrade",
        "mtd",
        "nano",
        "netifd",
        "nftables-json",
        "odhcp6c",
        "odhcpd-ipv6only",
        "opkg",
        "ppp",
        "ppp-mod-pppoe",
        "procd",
        "procd-seccomp",
        "procd-ujail",
        "screen",
        "swconfig",
        "tcpdump-mini",
        "uboot-envtools",
        "uci",
        "uclient-fetch",
        "urandom-seed",
        "urngd",
        "wpad-mbedtls"
    ],
    "diff_packages": true,
    "filesystem": "squashfs",
    "client": "luci/git-23.339.51123-138595a"
}
`

STDERR:
`
Generate local signing keys...
Generate local certificate...
Package list missing or not up-to-date, generating it.

Building package index...
Downloading http://downloads.openwrt.org/releases/24.10.0/targets/ath79/generic/packages/Packages.gz
Updated list of available packages in /builder/build_dir/target-mips_24kc_musl/root-ath79/../../../../builder/dl/openwrt_core
Downloading http://downloads.openwrt.org/releases/24.10.0/targets/ath79/generic/packages/Packages.sig
Signature check passed.
Downloading http://downloads.openwrt.org/releases/24.10.0/packages/mips_24kc/base/Packages.gz
Updated list of available packages in /builder/build_dir/target-mips_24kc_musl/root-ath79/../../../../builder/dl/openwrt_base
Downloading http://downloads.openwrt.org/releases/24.10.0/packages/mips_24kc/base/Packages.sig
Signature check passed.
Downloading http://downloads.openwrt.org/releases/24.10.0/targets/ath79/generic/kmods/6.6.73-1-c2bc166c1ba350209562f87ffdd33884/Packages.gz
Updated list of available packages in /builder/build_dir/target-mips_24kc_musl/root-ath79/../../../../builder/dl/openwrt_kmods
Downloading http://downloads.openwrt.org/releases/24.10.0/targets/ath79/generic/kmods/6.6.73-1-c2bc166c1ba350209562f87ffdd33884/Packages.sig
Signature check passed.
Downloading http://downloads.openwrt.org/releases/24.10.0/packages/mips_24kc/luci/Packages.gz
Updated list of available packages in /builder/build_dir/target-mips_24kc_musl/root-ath79/../../../../builder/dl/openwrt_luci
Downloading http://downloads.openwrt.org/releases/24.10.0/packages/mips_24kc/luci/Packages.sig
Signature check passed.
Downloading http://downloads.openwrt.org/releases/24.10.0/packages/mips_24kc/packages/Packages.gz
Updated list of available packages in /builder/build_dir/target-mips_24kc_musl/root-ath79/../../../../builder/dl/openwrt_packages
Downloading http://downloads.openwrt.org/releases/24.10.0/packages/mips_24kc/packages/Packages.sig
Signature check failed.
Remove wrong Signature file.
Downloading http://downloads.openwrt.org/releases/24.10.0/packages/mips_24kc/routing/Packages.gz
Updated list of available packages in /builder/build_dir/target-mips_24kc_musl/root-ath79/../../../../builder/dl/openwrt_routing
Downloading http://downloads.openwrt.org/releases/24.10.0/packages/mips_24kc/routing/Packages.sig
Signature check passed.
Downloading http://downloads.openwrt.org/releases/24.10.0/packages/mips_24kc/telephony/Packages.gz
Updated list of available packages in /builder/build_dir/target-mips_24kc_musl/root-ath79/../../../../builder/dl/openwrt_telephony
Downloading http://downloads.openwrt.org/releases/24.10.0/packages/mips_24kc/telephony/Packages.sig
Signature check passed.
Downloading file:packages/Packages
Updated list of available packages in /builder/build_dir/target-mips_24kc_musl/root-ath79/../../../../builder/dl/imagebuilder
Downloading file:packages/Packages.sig
Signature check passed.
Collected errors:
 * opkg_install_cmd: Cannot install package iperf3.
 * pkg_hash_check_unresolved: cannot find dependency cgi-io for luci-base
 * pkg_hash_fetch_best_installation_candidate: Packages for luci-base found, but incompatible with the architectures configured
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for luci:
 * 	cgi-io
 * opkg_install_cmd: Cannot install package luci.
 * pkg_hash_check_unresolved: cannot find dependency attendedsysupgrade-common for luci-app-attendedsysupgrade
 * pkg_hash_check_unresolved: cannot find dependency cgi-io for luci-app-attendedsysupgrade
 * pkg_hash_fetch_best_installation_candidate: Packages for luci-app-attendedsysupgrade found, but incompatible with the architectures configured
 * opkg_install_cmd: Cannot install package luci-app-attendedsysupgrade.
 * opkg_install_cmd: Cannot install package nano.
 * opkg_install_cmd: Cannot install package screen.
make[2]: *** [Makefile:234: package_install] Error 255
make[1]: *** [Makefile:171: _call_manifest] Error 2
make: *** [Makefile:349: manifest] Error 2
`

gentrydeng, 2025-02-14:
https://forum.openwrt.org/t/resolved-openwrt-image-builder-errors/224906

> 2025-Feb-14: The issues with the image builder system should now be resolved. If problems persist with Attended Sysupgrade (ASU), OpenWrt Upgrade Tool (OWUT) or the Firmware Selector custom image tool, please report them in [this thread](https://forum.openwrt.org/t/notice-problems-with-custom-firmware-image-creation/224829/4).

daidan83, 2025-02-16:
The issue is  indeed resolved.

closed 2025-02-16 by daidan83