openwrt/asu #1200: Attended Sysupgrade from OpenWrt 23.05.5 to 24.10 fails on Netgear WNDR3800

richb-hanover, 2025-02-06:
I have two Netgear WNDR3800's that are running 23.05.5. I tried to use LuCI attended sysupgrade to upgrade to 24.10.0.

Both upgrades failed with seemingly similar error messages (below). These devices remain un-upgraded. What other information could I provide? Thanks.

**Netgear WNDR3800 - number 1**

```
Error building the firmware image

Server response: Error: Impossible package selection
[Please report the error message and request](https://github.com/openwrt/asu/issues)

Request Data:

{
    "url": "https://sysupgrade.openwrt.org",
    "revision": "r24106-10cc5fcd00",
    "advanced_mode": "1",
    "sha256_unsigned": "",
    "branch": "23.05",
    "efi": null,
    "profile": "netgear,wndr3800",
    "target": "ath79/generic",
    "version": "24.10.0",
    "packages": [
        "auc",
        "base-files",
        "busybox",
        "ca-bundle",
        "cgi-io",
        "collectd-mod-ethstat",
        "collectd-mod-ipstatistics",
        "collectd-mod-irq",
        "collectd-mod-load",
        "collectd-mod-ping",
        "collectd-mod-powerdns",
        "collectd-mod-sqm",
        "collectd-mod-thermal",
        "collectd-mod-wireless",
        "curl",
        "dnsmasq",
        "dropbear",
        "firewall",
        "fstools",
        "fwtool",
        "getrandom",
        "hostapd-common",
        "ip6tables-nft",
        "iptables-nft",
        "iw",
        "iwinfo",
        "jshn",
        "jsonfilter",
        "kernel",
        "kmod-ath",
        "kmod-ath9k",
        "kmod-ath9k-common",
        "kmod-cfg80211",
        "kmod-gpio-button-hotplug",
        "kmod-ip6tables",
        "kmod-ipt-conntrack",
        "kmod-ipt-core",
        "kmod-ipt-nat",
        "kmod-ipt-offload",
        "kmod-leds-reset",
        "kmod-lib-crc-ccitt",
        "kmod-mac80211",
        "kmod-nf-conntrack",
        "kmod-nf-conntrack6",
        "kmod-nf-flow",
[... 309 lines cut ...]
Updated list of available packages in /builder/build_dir/target-mips_24kc_musl/root-ath79/../../../../builder/dl/openwrt_luci
Downloading http://downloads.openwrt.org/releases/24.10.0/packages/mips_24kc/luci/Packages.sig
Signature check passed.
Downloading http://downloads.openwrt.org/releases/24.10.0/packages/mips_24kc/packages/Packages.gz
Updated list of available packages in /builder/build_dir/target-mips_24kc_musl/root-ath79/../../../../builder/dl/openwrt_packages
Downloading http://downloads.openwrt.org/releases/24.10.0/packages/mips_24kc/packages/Packages.sig
Signature check passed.
Downloading http://downloads.openwrt.org/releases/24.10.0/packages/mips_24kc/routing/Packages.gz
Updated list of available packages in /builder/build_dir/target-mips_24kc_musl/root-ath79/../../../../builder/dl/openwrt_routing
Downloading http://downloads.openwrt.org/releases/24.10.0/packages/mips_24kc/routing/Packages.sig
Signature check passed.
Downloading http://downloads.openwrt.org/releases/24.10.0/packages/mips_24kc/telephony/Packages.gz
Updated list of available packages in /builder/build_dir/target-mips_24kc_musl/root-ath79/../../../../builder/dl/openwrt_telephony
Downloading http://downloads.openwrt.org/releases/24.10.0/packages/mips_24kc/telephony/Packages.sig
Signature check passed.
Downloading file:packages/Packages
Updated list of available packages in /builder/build_dir/target-mips_24kc_musl/root-ath79/../../../../builder/dl/imagebuilder
Downloading file:packages/Packages.sig
Signature check passed.
Collected errors:
 * opkg_install_cmd: Cannot install package auc.
make[2]: *** [Makefile:234: package_install] Error 255
make[1]: *** [Makefile:171: _call_manifest] Error 2
make: *** [Makefile:349: manifest] Error 2
```

patrakov, 2025-02-06:
This is indeed an impossible package selection, as `auc` has been replaced by `owut`. Nothing to fix server-side.

richb-hanover, 2025-02-07:
@patrakov - if I understand you correctly, the workaround is to use Advanced mode and remove `auc` from the list. Correct?

That made things better, but I also got "impossible build" warnings about `luarocks` and `wolfssl*` packages. So I removed those as well.

I am still getting the same message, but this time for `busybox` and `base-files` - I think I need these, don't I?

Thanks

<details>

```
Error building the firmware image

Server response: Error: Impossible package selection
[Please report the error message and request](https://github.com/openwrt/asu/issues)

Request Data:

{
    "url": "https://sysupgrade.openwrt.org",
    "revision": "r24106-10cc5fcd00",
    "advanced_mode": "1",
    "sha256_unsigned": "",
    "branch": "23.05",
    "efi": null,
    "request_hash": "99b8528bf80e334a3ae3ad0c5cd02441b464efb845e709ab6db1ebe7a2a72d4f",
    "profile": "netgear,wndr3800",
    "target": "ath79/generic",
    "version": "24.10.0",
    "packages": [
        "base-files",
        "busybox",
        "ca-bundle",
        "cgi-io",
        "collectd-mod-ethstat",
        "collectd-mod-ipstatistics",
        "collectd-mod-irq",
        "collectd-mod-load",
        "collectd-mod-ping",
        "collectd-mod-powerdns",
        "collectd-mod-sqm",
        "collectd-mod-thermal",
        "collectd-mod-wireless",
        "curl",
        "dnsmasq",
        "dropbear",
        "firewall",
        "fstools",
        "fwtool",
        "getrandom",
        "hostapd-common",
        "ip6tables-nft",
        "iptables-nft",
        "iw",
        "iwinfo",
        "jshn",
        "jsonfilter",
        "kernel",
        "kmod-ath",
        "kmod-ath9k",
        "kmod-ath9k-common",
        "kmod-cfg80211",
        "kmod-gpio-button-hotplug",
        "kmod-ip6tables",
        "kmod-ipt-conntrack",
        "kmod-ipt-core",
        "kmod-ipt-nat",
        "kmod-ipt-offload",
        "kmod-leds-reset",
        "kmod-lib-crc-ccitt",
[... 121 lines cut ...]
Downloading http://downloads.openwrt.org/releases/24.10.0/packages/mips_24kc/packages/Packages.sig
Signature check passed.
Downloading http://downloads.openwrt.org/releases/24.10.0/packages/mips_24kc/routing/Packages.gz
Updated list of available packages in /builder/build_dir/target-mips_24kc_musl/root-ath79/../../../../builder/dl/openwrt_routing
Downloading http://downloads.openwrt.org/releases/24.10.0/packages/mips_24kc/routing/Packages.sig
Signature check passed.
Downloading http://downloads.openwrt.org/releases/24.10.0/packages/mips_24kc/telephony/Packages.gz
Updated list of available packages in /builder/build_dir/target-mips_24kc_musl/root-ath79/../../../../builder/dl/openwrt_telephony
Downloading http://downloads.openwrt.org/releases/24.10.0/packages/mips_24kc/telephony/Packages.sig
Signature check passed.
Downloading file:packages/Packages
Updated list of available packages in /builder/build_dir/target-mips_24kc_musl/root-ath79/../../../../builder/dl/imagebuilder
Downloading file:packages/Packages.sig
Signature check passed.
Collected errors:
 * opkg_install_pkg: Checksum or size mismatch for package busybox. Either the opkg or the package index are corrupt. Try 'opkg update'.
 * opkg_install_cmd: Cannot install package base-files.
 * opkg_install_pkg: Checksum or size mismatch for package busybox. Either the opkg or the package index are corrupt. Try 'opkg update'.
 * opkg_install_cmd: Cannot install package busybox.
make[2]: *** [Makefile:234: package_install] Error 255
make[1]: *** [Makefile:171: _call_manifest] Error 2
make: *** [Makefile:349: manifest] Error 2
```

</details>

richb-hanover, 2025-02-08:
**Update:** Abot 12 hours later, I did remove the `auc` and `wolfssl*` packages from the Advanced list and the upgrade proceeded on both WNDR3800's without incident. Closing this..

closed 2025-02-08 by richb-hanover

aparcar, 2025-02-08:
@richb-hanover there was a recent update on the LuCI app to honor package_changes, a long overdue fix to handle this type of situations. Could you give it a try?

richb-hanover, 2025-02-08:
@aparcar I don't think I can test this now. I upgraded both WNDR3800 routers that had this problem.

For the future: How can I test the "package_changes" fix? Thanks

aparcar, 2025-02-08:
An example is an upgrade from any 23.05 device to 24.10 having `auc` installed. In those cases, `auc` asks you to replace `auc` via `owut`. In the LuCI app, it's not yet verbosely shown, however hopefully in the near future.

richb-hanover, 2025-02-08:
Here's a (maybe) crazy idea. The next time that ASU is faced with a known issue like `auc`-> `owut`, perhaps we could build an empty `auc` package that does nothing but act as a placeholder.

Then luci-app-attendedsysupgrade could request it (no special cases necessary); the ASU server adds the (empty) package, and all is well. (And the `owut` package might even be primed to remove `auc` after it's installed and running...)

Could this work? Or is there a better path in the works? Thanks again