openwrt/asu #1201: Attended Sysupgrade from OpenWrt 23.05.5 to 24.10 fails on GL.iNet Flint 2 (MT6000)

richb-hanover, 2025-02-06:
I got an "impossible package selection" message when upgrading a Flint 2 router to 24.10.0

```
Error building the firmware image

Server response: Error: Impossible package selection
[Please report the error message and request](https://github.com/openwrt/asu/issues)

Request Data:

{
    "url": "https://sysupgrade.openwrt.org",
    "revision": "r24106-10cc5fcd00",
    "advanced_mode": "1",
    "sha256_unsigned": "",
    "branch": "23.05",
    "efi": null,
    "profile": "glinet,gl-mt6000",
    "target": "mediatek/filogic",
    "version": "24.10.0",
    "packages": [
        "base-files",
        "busybox",
        "ca-bundle",
        "cgi-io",
        "dnsmasq",
        "dropbear",
        "e2fsprogs",
        "eip197-mini-firmware",
        "f2fsck",
        "firewall4",
        "fstools",
        "fwtool",
        "getrandom",
        "hostapd-common",
        "iw",
        "iwinfo",
        "jansson",
        "jshn",
        "jsonfilter",
        "kernel",
        "kmod-cfg80211",
        "kmod-crypto-acompress",
        "kmod-crypto-aead",
        "kmod-crypto-authenc",
        "kmod-crypto-ccm",
        "kmod-crypto-cmac",
        "kmod-crypto-crc32c",
        "kmod-crypto-ctr",
        "kmod-crypto-des",
        "kmod-crypto-gcm",
        "kmod-crypto-gf128",
        "kmod-crypto-ghash",
        "kmod-crypto-hash",
        "kmod-crypto-hmac",
        "kmod-crypto-hw-safexcel",
        "kmod-crypto-manager",
        "kmod-crypto-md5",
        "kmod-crypto-null",
        "kmod-crypto-rng",
        "kmod-crypto-seqiv",
        "kmod-crypto-sha1",
        "kmod-crypto-sha256",
        "kmod-crypto-sha512",
        "kmod-gpio-button-hotplug",
        "kmod-hwmon-core",
        "kmod-leds-gpio",
        "kmod-lib-crc-ccitt",
        "kmod-lib-crc32c",
        "kmod-lib-lzo",
[... 150 lines cut ...]
Updated list of available packages in /builder/build_dir/target-aarch64_cortex-a53_musl/root-mediatek/../../../../builder/dl/imagebuilder
Downloading file:packages/Packages.sig
Signature check passed.
Collected errors:
 * opkg_install_cmd: Cannot install package cgi-io.
 * pkg_hash_check_unresolved: cannot find dependency cgi-io for luci-base
 * pkg_hash_fetch_best_installation_candidate: Packages for luci-base found, but incompatible with the architectures configured
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for luci:
 * 	cgi-io
 * opkg_install_cmd: Cannot install package luci.
 * pkg_hash_check_unresolved: cannot find dependency attendedsysupgrade-common for luci-app-attendedsysupgrade
 * pkg_hash_check_unresolved: cannot find dependency cgi-io for luci-app-attendedsysupgrade
 * pkg_hash_fetch_best_installation_candidate: Packages for luci-app-attendedsysupgrade found, but incompatible with the architectures configured
 * opkg_install_cmd: Cannot install package luci-app-attendedsysupgrade.
 * pkg_hash_check_unresolved: cannot find dependency sqm-scripts for luci-app-sqm
 * pkg_hash_fetch_best_installation_candidate: Packages for luci-app-sqm found, but incompatible with the architectures configured
 * opkg_install_cmd: Cannot install package luci-app-sqm.
 * pkg_hash_check_unresolved: cannot find dependency travelmate for luci-app-travelmate
 * pkg_hash_fetch_best_installation_candidate: Packages for luci-app-travelmate found, but incompatible with the architectures configured
 * opkg_install_cmd: Cannot install package luci-app-travelmate.
 * opkg_install_cmd: Cannot install package travelmate.
make[2]: *** [Makefile:234: package_install] Error 255
make[1]: *** [Makefile:171: _call_manifest] Error 2
make: *** [Makefile:349: manifest] Error 2
```

PanosChtz, 2025-02-07:
I was getting the same error an hour ago but works now.

rasa, 2025-02-11:
Using the Attended Sysupgrade luci app, I updated from 23.05.5 to 24.10.0 without issue on two different Flint 2s. Perhaps the error was intermittent?

gentrydeng, 2025-02-14:
https://forum.openwrt.org/t/resolved-openwrt-image-builder-errors/224906

> 2025-Feb-14: The issues with the image builder system should now be resolved. If problems persist with Attended Sysupgrade (ASU), OpenWrt Upgrade Tool (OWUT) or the Firmware Selector custom image tool, please report them in [this thread](https://forum.openwrt.org/t/notice-problems-with-custom-firmware-image-creation/224829/4).

richb-hanover, 2025-02-14:
Just now (14Feb2025), on my Flint 2 router that I had manually upgraded to 24.10.0, I was able to use the Advanced config of the Attended Sysupgrade plugin to request a new build. It was successful - no error messages. I did not use that generated build to flash the router a second time.

I don't know if this counts as "SOLVED" but I think I am content.