openwrt/asu #1202: Attended Sysupgrade - "Simple" doesn't see 24.10.0 as an option

richb-hanover, 2025-02-06:
I wanted to upgrade several routers from 23.05.5 to 24.10.0. I used the LuCI web interface and chose "Search for firmware upgrade"

If the "simple" (non-Advanced) mode is used, I see:

> **No upgrade available**
>
> The device runs the latest firmware version 23.05.5 - r24106-10cc5fcd00

Going to Configuration tab, and choosing Advanced Mode, saving settings, then going back to "Search for firmware upgrade" gives the choice of 24.10.0. (There are follow-on failures, but I detailed these separately in this repo)

This happens on WNDR3800 and GL.iNet MT6000 (Flint 2)

What other diagnostic information could I provide? Thanks

lewi3069, 2025-02-07:
I also had this issue on my MT6000

maximilianovermeyer, 2025-02-07:
Same here for a KVM with "Generic EFI Boot (armsr/armv8)".

pety3bi, 2025-02-10:
The same on BPI-R3, Archer AX-23, Archer C7, TL-WDR4300

gentrydeng, 2025-02-14:
https://forum.openwrt.org/t/resolved-openwrt-image-builder-errors/224906

> 2025-Feb-14: The issues with the image builder system should now be resolved. If problems persist with Attended Sysupgrade (ASU), OpenWrt Upgrade Tool (OWUT) or the Firmware Selector custom image tool, please report them in [this thread](https://forum.openwrt.org/t/notice-problems-with-custom-firmware-image-creation/224829/4).

richb-hanover, 2025-02-14:
Because I _was_ able to update to 24.10.0 on the routers I reported earlier, I can no longer test whether the "Simple" mode works (I get a "this router is up to date with 24.10.0"message, which is true).

Can anyone who still is on 23.05.5 test and see if 24.10.0 is suggested in Simple mode? Thanks

aparcar, 2025-02-15:
Motivation was that an upgrade between major releases is more likely to break the device, meaning it's somewhat advanced. I'm happy to always show the latest stable version, someone just needs to make a LuCI PR, @efahl maybe?

efahl, 2025-02-15:
Cross-release version upgrades from 23.05 should be a lot more robust after https://github.com/openwrt/luci/pull/7621 gets merged (the package-changes code is already in 24.10 and main).  So, I'd say it's a good time to expose them in simple mode as soon as that's in.

richb-hanover, 2025-02-15:
> Motivation was that an upgrade between major releases is more likely to break the device..

Ahah! I now understand why it was "hidden" in Advanced mode. I was confused because the message - "No upgrade available..." - is clearly incorrect. And that made me think something was broken.

For reference, here's the message I saw:

> **No upgrade available**
>
>    The device runs the latest firmware version 23.05.5 - r24106-10cc5fcd00

So the trick is to decide what to tell the user in this situation. 

* When a minor upgrade (say 23.05.4 to 23.05.5) is available, the GUI works properly, and the operation is extremely safe
* When a major upgrade (say, 23.05.4 to 24.10.0) is also available, it seems we need to convey several pieces of information:
    * A minor upgrade is available (or not)
    * A major upgrade (to 24.10.0) is available
    * (Maybe a link to warnings...)
    * A way to "Do it" that initiates the upgrade to the new version

Perhaps we should simply change the wording in the message to say this:

> **No minor upgrade is available**
> The device runs version 23.05.5 - r24106-10cc5fcd00
> 
> **Major upgrade is available**
> Version 24.10.0 is available. To install it, set "Advanced Mode" in the Configuration tab.

No change to the code, but the message tells people what to do, and would have solved my problem. Thanks for listening.

efahl, 2025-02-15:
@richb-hanover , take a look at this, too: https://github.com/openwrt/luci/pull/7597  Maybe add some comments to get it rolling...

richb-hanover, 2025-02-15:
Thanks for pointing that out. I left comments over there. (TL;DR we should test for opt-in/opt-out and if permitted, check for new versions at login). 

This `attended-sysupgrade` message needs attention as well, especially if automatic version check are authorized.

I am completely content with the suggested wording above (to use Advanced mode if you want to make a major version update).

efahl, 2025-03-09:
Fixed in https://github.com/openwrt/luci/pull/7672

closed 2025-05-02 by ldir-EDB0