openwrt/asu #1205: Internal server error for custom builds

laszlo-t, 2025-02-06:
I get an Internal server when I'd like to create custom build even without chaining the default package list.

`curl 'https://sysupgrade.openwrt.org/api/v1/build' -X POST -H 'User-Agent: Mozilla/5.0 (X11; Linux x86_64; rv:135.0) Gecko/20100101 Firefox/135.0' -H 'Accept: */*' -H 'Accept-Language: hu,en-US;q=0.7,en;q=0.3' -H 'Accept-Encoding: gzip, deflate, br, zstd' -H 'Referer: https://firmware-selector.openwrt.org/' -H 'Content-Type: application/json' -H 'Origin: https://firmware-selector.openwrt.org' -H 'Connection: keep-alive' -H 'Sec-Fetch-Dest: empty' -H 'Sec-Fetch-Mode: cors' -H 'Sec-Fetch-Site: same-site' -H 'Priority: u=4' -H 'Cache-Control: max-age=0' -H 'TE: trailers' --data-raw '{"profile":"generic","target":"x86/64","packages":["base-files","ca-bundle","dnsmasq","dropbear","e2fsprogs","firewall4","fstools","grub2-bios-setup","kmod-button-hotplug","kmod-nft-offload","libc","libgcc","libustream-mbedtls","logd","mkf2fs","mtd","netifd","nftables","odhcp6c","odhcpd-ipv6only","opkg","partx-utils","ppp","ppp-mod-pppoe","procd-ujail","uci","uclient-fetch","urandom-seed","urngd","kmod-amazon-ena","kmod-amd-xgbe","kmod-bnx2","kmod-dwmac-intel","kmod-e1000e","kmod-e1000","kmod-forcedeth","kmod-fs-vfat","kmod-igb","kmod-igc","kmod-ixgbe","kmod-r8169","kmod-tg3","kmod-drm-i915","luci","kmod-alx","openssh-sftp-server","pciutils","usbutils","mc","kmod-mt7921e","kmod-mt76x0u","block-mount","nfs-kernel-server","luci-app-ddns","intel-microcode","bash","kmod-hwmon-it87","lm-sensors","luci-app-hd-idle","hdparm","smartmontools","htop","wpad-basic-mbedtls","luci-app-transmission","transmission-web"],"defaults":"","version_code":"r28427-6df0e3d02a","version":"24.10.0","diff_packages":true,"client":"ofs/v5.0.0-2-g4336d760"}'`

[image]

Avendor7, 2025-02-06:
Also seeing this on an x86 system

laszlo-t, 2025-02-08:
it is working now, so I'm closing the issue

closed 2025-02-08 by laszlo-t