openwrt/asu #1207: Error building firmware Impossible package

brent2489, 2025-02-06:
[Untitled document.pdf](https://github.com/user-attachments/files/18698744/Untitled.document.pdf)

gentrydeng, 2025-02-14:
https://forum.openwrt.org/t/resolved-openwrt-image-builder-errors/224906

> 2025-Feb-14: The issues with the image builder system should now be resolved. If problems persist with Attended Sysupgrade (ASU), OpenWrt Upgrade Tool (OWUT) or the Firmware Selector custom image tool, please report them in [this thread](https://forum.openwrt.org/t/notice-problems-with-custom-firmware-image-creation/224829/4).

brent2489, 2025-02-22:
Downloaded the sysupgrade Tuf-AX6000 24.10.0 image from https://openwrt.org/toh/asus/tuf-ax6000.  Installed perfectly and working great!!

closed 2025-02-22 by brent2489