openwrt/asu #1208: Error building the firmware image Impossible package selection

Lesbos888, 2025-02-07:
Lots of signature failures for target rockchip/armv8:

Error building the firmware image
Server response: Error: Impossible package selection

[Please report the error message and request](https://github.com/openwrt/asu/issues)
Request Data:

{
    "url": "https://sysupgrade.openwrt.org",
    "revision": "r24106-10cc5fcd00",
    "advanced_mode": "1",
    "sha256_unsigned": "",
    "branch": "23.05",
    "efi": null,
    "profile": "friendlyarm,nanopi-r4s",
    "target": "rockchip/armv8",
    "version": "24.10.0",
    "packages": [
        "attendedsysupgrade-common",
        "base-files",
        "busybox",
        "ca-bundle",
        "cgi-io",
        "collectd",
        "collectd-mod-cpu",
        "collectd-mod-interface",
        "collectd-mod-iwinfo",
        "collectd-mod-load",
        "collectd-mod-memory",
        "collectd-mod-network",
        "collectd-mod-rrdtool",
        "collectd-mod-thermal",
        "curl",
        "ddns-scripts",
        "ddns-scripts-cloudflare",
        "ddns-scripts-services",
        "dnsmasq-full",
        "drill",
        "dropbear",
        "e2fsprogs",
        "fdisk",
        "firewall4",
        "fstools",
        "fwtool",
        "getrandom",
        "htop",
        "https-dns-proxy",
        "ip6tables-nft",
        "ipset",
        "jansson",
        "jq",
        "jshn",
        "jsonfilter",
        "kernel",
        "kmod-crypto-crc32c",
        "kmod-crypto-hash",
        "kmod-gpio-button-hotplug",
        "kmod-lib-crc-ccitt",
        "kmod-lib-crc32c",
        "kmod-nf-conntrack",
        "kmod-nf-conntrack6",
        "kmod-nf-flow",
        "kmod-nf-log",
        "kmod-nf-log6",
        "kmod-nf-nat",
        "kmod-nf-reject",
        "kmod-nf-reject6",
        "kmod-nfnetlink",
        "kmod-nft-core",
[... 193 lines cut ...]
 * opkg_install_cmd: Cannot install package luci-app-sqm.
 * opkg_install_cmd: Cannot install package luci-app-statistics.
 * opkg_install_cmd: Cannot install package luci-base.
 * opkg_install_cmd: Cannot install package luci-compat.
 * opkg_install_cmd: Cannot install package luci-lib-base.
 * opkg_install_cmd: Cannot install package luci-lib-ip.
 * opkg_install_cmd: Cannot install package luci-lib-jsonc.
 * opkg_install_cmd: Cannot install package luci-lib-nixio.
 * opkg_install_cmd: Cannot install package luci-lua-runtime.
 * opkg_install_cmd: Cannot install package luci-mod-admin-full.
 * opkg_install_cmd: Cannot install package luci-mod-network.
 * opkg_install_cmd: Cannot install package luci-mod-status.
 * opkg_install_cmd: Cannot install package luci-mod-system.
 * opkg_install_cmd: Cannot install package luci-proto-ipv6.
 * opkg_install_cmd: Cannot install package luci-proto-ppp.
 * opkg_install_cmd: Cannot install package luci-ssl.
 * opkg_install_cmd: Cannot install package luci-theme-bootstrap.
 * opkg_install_cmd: Cannot install package nano.
 * opkg_install_cmd: Cannot install package openvpn-openssl.
 * opkg_install_cmd: Cannot install package pbr.
 * opkg_install_cmd: Cannot install package rpcd-mod-luci.
 * opkg_install_cmd: Cannot install package rpcd-mod-rrdns.
make[2]: *** [Makefile:234: package_install] Error 255
make[1]: *** [Makefile:171: _call_manifest] Error 2
make: *** [Makefile:349: manifest] Error 2

Lesbos888, 2025-02-07:
Ended up unchecking all the mentioned packages, which worked, but left me without LUCI or the ability for opkg to use TLS.

Had to edit distfeeds.conf to use http and reinstall everything that way.

gentrydeng, 2025-02-14:
https://forum.openwrt.org/t/resolved-openwrt-image-builder-errors/224906

> 2025-Feb-14: The issues with the image builder system should now be resolved. If problems persist with Attended Sysupgrade (ASU), OpenWrt Upgrade Tool (OWUT) or the Firmware Selector custom image tool, please report them in [this thread](https://forum.openwrt.org/t/notice-problems-with-custom-firmware-image-creation/224829/4).

closed 2025-02-14 by Lesbos888