openwrt/asu #1209: Error building the firmware image:  Impossible package selection

andriej, 2025-02-07:
```
{
    "url": "https://sysupgrade.openwrt.org",
    "branch": "24.10",
    "revision": "r28304-6dacba30a7",
    "efi": null,
    "advanced_mode": "1",
    "request_hash": "",
    "sha256_unsigned": "",
    "client": "luci/25.037.68331~42f464c",
    "packages": {
        "kmod-usb-storage": "6.6.69-r1",
        "luci-proto-wireguard": "25.005.74981~65c86ed",
        "libopenssl": "3.0.15-r1",
        "openssh-sftp-server": "9.9_p1-r1",
        "iputils-arping": "20240905-r1",
        "libc": "1.2.5-r4",
        "mkf2fs": "1.16.0-r3",
        "opkg": "2024.10.16~38eccbb1-r1",
        "kmod-usb-core": "6.6.69-r1",
        "luci-app-vnstat2": "25.005.74981~65c86ed",
        "diffutils": "3.10-r1",
        "kmod-tun": "6.6.69-r1",
        "openssh-server": "9.9_p1-r1",
        "kmod-rtw88-8822ce": "6.6.69.6.12.6-r1",
        "procd-ujail": "2024.12.22~42d39376-r1",
        "base-files": "1649~6dacba30a7",
        "mosquitto-client-ssl": "2.0.18-r3",
        "partx-utils": "2.40.2-r1",
        "libustream-mbedtls": "2024.07.28~99bd3d2b-r1",
        "6in4": "29",
        "kmod-r8169": "6.6.69-r1",
        "firewall4": "2024.12.18~18fc0ead-r1",
        "uboot-envtools": "2024.07-r1",
        "dnsmasq": "2.90-r3",
        "usbutils": "017-r1",
        "kmod-usb-net-dm9601-ether": "6.6.69-r1",
        "exfat-fsck": "1.2.5-r1",
        "mdns-repeater": "2020.05.03~921d8850-r2",
        "etherwake": "1.09-r5",
        "usbids": "0.387-r1",
        "block-mount": "2024.07.14~408c2cc4-r1",
        "kmod-fs-vfat": "6.6.69-r1",
        "kmod-usb2": "6.6.69-r1",
        "kmod-usb-serial-option": "6.6.69-r1",
        "kmod-fs-exfat": "6.6.69-r1",
        "cJSON": "1.7.17-r1",
        "tcpdump": "4.99.5-r1",
        "odhcp6c": "2024.09.25~b6ae9ffa-r1",
        "iptables-mod-ipopt": "1.8.10-r1",
        "uci": "2023.08.10~5781664d-r1",
        "nmap-full": "7.95-r1",
        "luci-ssl": "25.005.74981~65c86ed",
        "dropbear": "2024.86-r1",
        "tailscale": "1.78.1-r1",
        "curl": "8.10.1-r1",
        "mtd": "26",
        "odhcpd-ipv6only": "2024.05.08~a2988231-r1",
        "smstools3": "3.1.21-r4",
        "luci-app-sqm": "25.005.74981~65c86ed",
        "urandom-seed": "3",
        "ppp": "2.5.1-r1",
        "arp-scan-database": "1.10.0-r2",
        "bcp38": "5-r11",
        "kmod-gpio-button-hotplug": "6.6.69-r5",
        "logd": "2024.04.26~85f10530-r1",
        "kmod-usb-net": "6.6.69-r1",
        "luci-app-attendedsysupgrade": "25.037.68331~42f464c",
        "e2fsprogs": "1.47.0-r2",
        "htop": "3.3.0-r2",
        "iptraf-ng": "1.2.1-r1",
        "ca-bundle": "20240203-r1",
        "luci-app-tinyproxy": "25.023.69389~2e7e104",
        "fping": "5.2-r1",
        "luci-app-bcp38": "25.005.74981~65c86ed",
        "luci-app-nextdns": "25.005.74981~65c86ed",
        "luci": "25.005.74981~65c86ed",
        "owut": "2024.12.10~e38844ae-r1",
        "kernel": "6.6.69~7d76954c8713c92c9e7202dd1f9c1f2b-r1",
        "blkid": "2.40.2-r1",
        "kmod-ifb": "6.6.69-r1",
        "mtr-json": "0.95-r3",
        "iperf3": "3.17.1-r4",
        "urngd": "2023.11.01~44365eb1-r1",
        "ppp-mod-pppoe": "2.5.1-r1",
        "arp-scan": "1.10.0-r2"
    },
    "profile": "friendlyarm,nanopi-r5c",
    "target": "rockchip/armv8",
    "version": "24.10.0-rc5",
    "diff_packages": true,
    "filesystem": "squashfs"
}
```

STDERR:
```
Generate local signing keys...
Generate local certificate...
Package list missing or not up-to-date, generating it.

Building package index...
Downloading http://downloads.openwrt.org/releases/24.10.0/targets/rockchip/armv8/packages/Packages.gz
Updated list of available packages in /builder/build_dir/target-aarch64_generic_musl/root-rockchip/../../../../builder/dl/openwrt_core
Downloading http://downloads.openwrt.org/releases/24.10.0/targets/rockchip/armv8/packages/Packages.sig
Signature check passed.
Downloading http://downloads.openwrt.org/releases/24.10.0/packages/aarch64_generic/base/Packages.gz
Updated list of available packages in /builder/build_dir/target-aarch64_generic_musl/root-rockchip/../../../../builder/dl/openwrt_base
Downloading http://downloads.openwrt.org/releases/24.10.0/packages/aarch64_generic/base/Packages.sig
Signature check passed.
Downloading http://downloads.openwrt.org/releases/24.10.0/targets/rockchip/armv8/kmods/6.6.73-1-f35e93bc2c89b98d107e57cdea041972/Packages.gz
Updated list of available packages in /builder/build_dir/target-aarch64_generic_musl/root-rockchip/../../../../builder/dl/openwrt_kmods
Downloading http://downloads.openwrt.org/releases/24.10.0/targets/rockchip/armv8/kmods/6.6.73-1-f35e93bc2c89b98d107e57cdea041972/Packages.sig
Signature check passed.
Downloading http://downloads.openwrt.org/releases/24.10.0/packages/aarch64_generic/luci/Packages.gz
Updated list of available packages in /builder/build_dir/target-aarch64_generic_musl/root-rockchip/../../../../builder/dl/openwrt_luci
Downloading http://downloads.openwrt.org/releases/24.10.0/packages/aarch64_generic/luci/Packages.sig
Signature check passed.
Downloading http://downloads.openwrt.org/releases/24.10.0/packages/aarch64_generic/packages/Packages.gz
Updated list of available packages in /builder/build_dir/target-aarch64_generic_musl/root-rockchip/../../../../builder/dl/openwrt_packages
Downloading http://downloads.openwrt.org/releases/24.10.0/packages/aarch64_generic/packages/Packages.sig
Signature check passed.
Downloading http://downloads.openwrt.org/releases/24.10.0/packages/aarch64_generic/routing/Packages.gz
Updated list of available packages in /builder/build_dir/target-aarch64_generic_musl/root-rockchip/../../../../builder/dl/openwrt_routing
Downloading http://downloads.openwrt.org/releases/24.10.0/packages/aarch64_generic/routing/Packages.sig
Signature check passed.
Downloading http://downloads.openwrt.org/releases/24.10.0/packages/aarch64_generic/telephony/Packages.gz
Updated list of available packages in /builder/build_dir/target-aarch64_generic_musl/root-rockchip/../../../../builder/dl/openwrt_telephony
Downloading http://downloads.openwrt.org/releases/24.10.0/packages/aarch64_generic/telephony/Packages.sig
Signature check passed.
Downloading file:packages/Packages
Updated list of available packages in /builder/build_dir/target-aarch64_generic_musl/root-rockchip/../../../../builder/dl/imagebuilder
Downloading file:packages/Packages.sig
Signature check passed.
Collected errors:
 * opkg_install_pkg: Checksum or size mismatch for package rpcd-mod-luci. Either the opkg or the package index are corrupt. Try 'opkg update'.
 * opkg_install_cmd: Cannot install package luci.
make[2]: *** [Makefile:234: package_install] Error 255
make[1]: *** [Makefile:171: _call_manifest] Error 2
make: *** [Makefile:349: manifest] Error 2
```

walmartshopper, 2025-02-07:
Getting the same error on x86_64.  Owut list:

luci-proto-wireguard luci-app-adblock nano intel-microcode luci-app-banip luci-app-ddns tcpdump curl luci-app-sqm dnscrypt-proxy2 luci-app-attendedsysupgrade lm-sensors rng-tools smartmontools htop luci owut iperf3 -kmod-drm-i915 -nftables

gentrydeng, 2025-02-14:
https://forum.openwrt.org/t/resolved-openwrt-image-builder-errors/224906

> 2025-Feb-14: The issues with the image builder system should now be resolved. If problems persist with Attended Sysupgrade (ASU), OpenWrt Upgrade Tool (OWUT) or the Firmware Selector custom image tool, please report them in [this thread](https://forum.openwrt.org/t/notice-problems-with-custom-firmware-image-creation/224829/4).

andriej, 2025-02-15:
Confirmed, issue is solved and image is built without issue.

closed 2025-02-15 by andriej